arjenhiemstra/ithowifi #91: Sometimes unavailable

tiestofan9, 2022-05-15:
Hi,  my WiFi module switches from available to unavailable almost every 5 minutes. Any idea how to fix this?

Thanks in advance!

Diagnostics:

{
  "home_assistant": {
    "installation_type": "Home Assistant Supervised",
    "version": "2022.5.4",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.9.9",
    "docker": true,
    "arch": "armv7l",
    "timezone": "Europe/Amsterdam",
    "os_name": "Linux",
    "os_version": "5.10.17-v7l+",
    "supervisor": "2022.05.1",
    "host_os": "Raspbian GNU/Linux 10 (buster)",
    "docker_version": "20.10.7",
    "chassis": "",
    "run_as_root": true
  },
  "custom_components": {
    "eufy_security": {
      "version": "0.3.0",
      "requirements": [
        "python-eufy-security==0.3.0"
      ]
    },
    "libratone_zipp": {
      "version": "3.0.3",
      "requirements": [
        "python_libratone_zipp==3.0.0"
      ]
    },
    "govee": {
      "version": "0.2.2",
      "requirements": [
        "govee-api-laggat==0.2.2",
        "dacite==1.6.0"
      ]
    },
    "tgtg": {
      "version": "4.0.1",
      "requirements": [
        "tgtg>=0.11.1"
      ]
    },
    "adaptive_lighting": {
      "version": "1.0.14",
      "requirements": []
    },
    "hacs": {
      "version": "1.24.5",
      "requirements": [
        "aiogithubapi>=21.11.0"
      ]
    },
    "xiaomi_cloud_map_extractor": {
      "version": "v2.1.5",
      "requirements": [
        "pillow",
        "pybase64",
        "python-miio",
        "requests",
        "pycryptodome"
      ]
[... 523 lines cut ...]
              "device": {
                "identifiers": "nrg-itho-e138",
                "manufacturer": "Arjen Hiemstra",
                "model": "ITHO Wifi Add-on",
                "name": "ITHO-WIFI(nrg-itho-e138)",
                "sw_version": "HW: v2, FW: 2.3.5"
              },
              "availability_topic": "itho/lwt",
              "device_class": "temperature",
              "unique_id": "nrg-itho-e138_temperature",
              "name": "nrg-itho-e138_temperature",
              "state_topic": "itho/ithostatus",
              "state_class": "measurement",
              "value_template": "{{ value_json.temp }}",
              "unit_of_measurement": "\u00b0C",
              "platform": "mqtt"
            }
          },
          "transmitted": []
        }
      ],
      "triggers": []
    }
  }
}

jasperslits, 2022-05-15:
You should check and include the add-on logs under Debug. The dump from home assistant is only the result of a problem elsewhere.

tiestofan9, 2022-05-15:
> You should check and include the add-on logs under Debug. The dump from home assistant is only the result of a problem elsewhere.

Ah! Like this? 

1652642149: Client <unknown> closed its connection.
1652642195: Saving in-memory database to /data//mosquitto.db.
1652642269: New connection from 172.30.32.2:47898 on port 1883.
1652642269: Client <unknown> closed its connection.
1652642389: New connection from 172.30.32.2:47966 on port 1883.
1652642389: Client <unknown> closed its connection.
1652642411: New connection from 192.168.1.198:55520 on port 1883.
1652642411: New client connected from 192.168.1.198:55520 as nrg-itho-e138 (p2, c1, k15, u'mqtt-user').
1652642419: New connection from 192.168.1.198:55599 on port 1883.
1652642419: Client nrg-itho-e138 already connected, closing old connection.
1652642419: New client connected from 192.168.1.198:55599 as nrg-itho-e138 (p2, c1, k15, u'mqtt-user').
1652642476: Client nrg-itho-e138 has exceeded timeout, disconnecting.
1652642509: New connection from 172.30.32.2:48054 on port 1883.
1652642509: Client <unknown> closed its connection.

arjenhiemstra, 2022-05-15:
Not really :) go to the webpage of the add-on, click on 'Debug' in the menu on the left en copy the content of the debug log.

closed 2022-07-21 by stale[bot]