arjenhiemstra/ithowifi #93: Beginners problems, I suppose

sciurius, 2022-06-28:
Hi,

I received my add-on and connected it to my Itho CVE-S PAK. Up and running in a whimp.

I do, however, encounter some problems that are probably due to my misunderstanding of concepts and documentation.

As said, the add-on is connected to the WiFi and MQTT broker. The RF add-on is enabled, paired, the remote control has been unpaired and paired again. MQTT Explorer shows the messages flowing. So far, so good. I'm not using HA and Domoticz MQTT. (Well I do use HA MQTT, but all manually.)

When I publish `{"speed":"120"}` to `itho/cmd` it is correctly received and understood. `itho/lastcmd` shows 

    {"source":"MQTT API","command":"speed: 120","timestamp":1656428880}

**Describe the bug**

When I publish `{"vremote":"low"}` to `itho/cmd, I get

    {"source":"MQTT API","command":"low","timestamp":1656428992}

and nothing happens. The same happens with the HTML API:

    $ curl http://nrgitho.squirrel.nl/api.html\?vremote=low

results in

    {"source":"HTML API","command":"low","timestamp":1656429048}

Note that in both cases the command reads `"low"` and not (what I would expect) `"vremote":"low"`.

Is there documentation on the meaning of the miscellaneous attributes? I'm a bit puzzled by attributes like `Speed status`, `FanInfo`, `ExFanSpeed` and `Fan speed`...

**Device information**
- Firmware version [e.g. 2.2.1] 2.3.5
- Hardware revision (e.g. non-cve 1.1) 2 CVE
- Type/model number Itho (on sticker outside the box) CVE-S PAK
- CC1101 RF module enabled [ yes, no ] yes

**Debug logging**

[logfile.txt](https://github.com/arjenhiemstra/ithowifi/files/9002444/logfile.txt)

arjenhiemstra, 2022-06-28:
I really appreciate the effort you took to make a very clear description of your issue! Thanks!

The vremote API command is a software emulation of physical remote. It adheres to the same logic and restrictions as physical remote. As a result, before a vremote can control the itho, it needs to be joined to the itho as remote.
In the latest alpha version I've changed the API description to (hopefully) be more clear about what works in what situation and what is needed to get it to work. (vremotecmd == vremote in this case)

Screenshot from alpha7:
[image]

About the status items; there is not a lot documentation about it. The labels give an indication what it is. The information is presented in the itho servicetool without further explanation. Every itho device and even between firmware version these labels differ making it kind of difficult to document and understand what they mean.

From what I know from my own CVE:
Speed status == actual running % motor speed
Ventilation setpoint (%) == requested % motor speed

Some odd high values (in the >60000 range) are most often error codes

sciurius, 2022-06-28:
Hi Arjen,

Thanks for your elaborate response.

In the "Virtual remote settings" I've configured "Send join command" on "every power on", so I assume that the virtual remote is correctly paired. The real remote shows up on the RF Remote Setup page so I assume that's okay too. Nevertheless, pressing any of the Low/Medium/High buttons on the front page changes the slider position, but does not change the actual fan speed. Pressing one of the real remote buttons changes the slider and also the speed.

So it seems that I cannot change anything via HTML API, MQTT and web UI. Apparently I'm missing something crucial. Any ideas?

arjenhiemstra, 2022-06-28:
I'm afraid that "send join command" setting is nog working properly in 2.3.5

Looking through my release notes, I fixed something in alpha3: fix: restore send join command on power-on function

Can't completely remember what the culprit was.

You can do 2 things: manually push the "join" button on the debug page within 2 minutes after poweron. Upgrade to the latest alpha7 release and then the "Send join command" options works as expected.

sciurius, 2022-06-28:
Would that also explain why non-remote commands (e.g. just a "speed:200") do not work?

arjenhiemstra, 2022-06-28:
That is usually a sign that your itho is not in medium mode. If a power cycle of the itho unit makes the speed:200 possible again you can be sure that it was the issue

sciurius, 2022-06-28:
Ok, did a manual JOIN after power on and now certain things work.
How exactly do I control medium mode? I have "force medium mode" on.

arjenhiemstra, 2022-06-28:
That setting should handle a fall out of medium mode whenever it is needed for certain commands to work (basically before every speed:x command). If your remote is just joined to the add-on and not directly to the itho (and you also do not use a 3 way switch) this should not make any difference because then there is no way your itho can fall out of medium mode then.

 If it makes a difference, your remote is most probably still joined to the itho as well as the add-on, which results in undefined behavior of the itho cve.

sciurius, 2022-06-28:
Yes, I double checked the pairings. The Itoh does not respond at all to the remote so I'm pretty confident it is not paired to the Itoh.

FanInfo is "auto" so that looks ok.

1. Slider on the main page and API "speed" calls seem to work.
2. N/A
3. Map RF remotes to virtual remote is already off
4. I2C virtual remote commands do not work. Re-paired the vremote
5. This is what happens. E.g. after pressing [High] the FanInfo is "high" and no commands work. "Force medium" is on. Power cycle...
Now it is "auto" again. Pressing the [low] or [high] on the main page works and FanInfo stays "auto". However, the low level i2C commands on de debug page change the speed **and** FanInfo so that's "einde oefening". Also the speed slider doesn't work anymore.
Interesting: [Low] on the main page drops Speed status to around 30. [Low] on the debug page drops to around 15.
Power cycle...
6. Physical remote seems to work as well.

So the bottom line is: do not press the low level I2C buttons. Also, it seems that the one that would be harmless, low level [Medium], does nothing.

BTW, I would assume the add-on can control the Itho without involving virtual or real remotes. IOW, things like changing the fan speed should always work.

arjenhiemstra, 2022-06-28:
with "after pressing [High]" you mean pressing the high button on the debug page?

If this pressing high works and changes the FanInfo status, the vremote is paired correctly. Pressing medium on the debug page should change the FanInfo back to "auto". If not, then we entered an interesting area...

Then I wonder what your setup is:
1. what kind of itho model do you have, can you post a picture of the label on the fan, if possible also a picture the PCB inside? 
2. What kind of physical remote do you have? A picture would be helpful as wel.

If either it is an unknown version additional support might have to be added. Upgrading to a later firmware version and using the RF debug function to capture some RF commands might also be a next step.

arjenhiemstra, 2022-06-29:
What is the Itho device type & Itho fw version on the itho settings page?

sciurius, 2022-06-29:
> What is the Itho device type & Itho fw version on the itho settings page?

    Itho device type: CVE-Silent
    Itho fw version: 27

> The latest alpha version has the possibility to change the virtual remote type. In this case you should select "RFT AUTO". Unfortunately you will have to unpair / pair the virtual remote again after changing the type.

Yes, I already did that some time ago.

[image]

> in the picture I see a connector that usually has a perilex plug on the other end. Do you have a 3-way switch in use? If so, just to be sure, is the 3-way switch in the mid position?

Yes, it has a periliex plug. Originally the Itho was wired to a 2-way switch in the bathroom. This gave two options: "auto" and "high". I did not find that acceptible (power consumption!) so it was rewired to "auto" and "off" (no power). When I switch it on with the bathroom switch it is always "auto".

sciurius, 2022-06-29:
Current setup: add-on paired as a virtual RFT AUTO remote. Real remote unpaired from the Itho and paired as an RF remote to the add-on.
When I press the [Auto] button on the remote, according to lastcmd the add-on sends "speed:120" to the Itho, 

    {"source":"badkamer","command":"speed:120","timestamp":1656502301}

Likewise, [Low] sends "speed:20". This is the same as the buttons from the main page, e.g. [Medium]:

    {"source":"web interface","command":"speed:120","timestamp":1656502569}

So far, so good.

When I press [Medium] on the debug page, strange data is sent:

````
{"source":"�Cb interface-vrem������?���?���?s\\�C�C�?|3�?","command":"medium","timestamp":1656502692}
````
There is no response on the Itho. (I increased fan speed before pressing the button so there would be a noticeable 
change in speed if the command did work.)

When I press [Low] on the debug page:

````
{"source":"�Cb interface-vrem������?���?���?s\\�C�C�?|3�?","command":"low","timestamp":1656502924}
````
The Itho responds by switching the fan speed and going into "low" mode. No way to get it in auto mode again (except for a power flip). It should revert to auto after 24 hours if I recall the manual correctly. I'll try that some day.
Playing with the Itoh add-on is good for your health. Can't count the number of times I went upstairs to flip the power ☺.

Summarizing: I can do what I want as long as I don't issue vremote commands, either via the API or by pressing the buttons on the debug page.

Two issues remain:
1. Why doesn't the [Medium] button on the debug page work?
2. The [Low] command reduces speed to 13% (450RPM). With speed commands I cannot get lower than 30% (905RPM). Is that blocked in the add-in software? E.g. speed 254 (99%) is OK, but 255 (100%) is NOK.
Lower values down to 90 (920RPM) all work, 89 is also 920RPM, 88 is 905RPM, and anything lower than 88 is stuck at 905RPM.

arjenhiemstra, 2022-06-29:
>When I press [Medium] on the debug page, strange data is sent:

See my earlier remark: "with changing the remote type the buttons on the debug page are not functional correctly anymore but you can use the buttons from your image"

>When I press [Low] on the debug page:

Same fir this one. It is an alpha version, the debug buttons will be removed in future versions. After changing to a different remote type those debug page button can point to non existing remote commands (hence the garbage strange data you see).
Bottom line: use the vremote through Virtual Remote page or API but do not use the debug buttons in the alpha7 version.

>Playing with the Itoh add-on is good for your health. Can't count the number of times I went upstairs to flip the power ☺.

Story of my life ;-)

>Why doesn't the [Medium] button on the debug page work?

Same as my earlier notion

>The [Low] command reduces speed to 13% (450RPM). With speed commands I cannot get lower than 30% (905RPM). Is that blocked in the add-in software? E.g. speed 254 (99%) is OK, but 255 (100%) is NOK.
Lower values down to 90 (920RPM) all work, 89 is also 920RPM, 88 is 905RPM, and anything lower than 88 is stuck at 905RPM.

This is due to some itho settings out of the factory. They can be changed. The 30% can be lowered by changing this setting to a lower value: "Minimum ventilation level in Auto (%)"
That setting is probably on 30 now.

Furthermore it is possible to lower the low setting further by tweaking the "setting of potentiometer" settings to a lower value. Lastly, if you disable the humidity sensor (can be done on this model by settings setting "Additional sensor support" to "on") you can even stop the motor completely. Possible downside is that the fan won't react to humidity increases anymore but with the info on the API you can handle that yourself through you home domotica system.

sciurius, 2022-06-29:
> Bottom line: use the vremote through Virtual Remote page or API but do not use the debug buttons in the alpha7 version.

No problem. API `vremote=low` switches out of "auto" mode, but `vremote=auto` brings it back. `vremote=medium` does nothing.

> Minimum ventilation level in Auto (%)

Indeed. Setting[26] was 30, I've lowered it to 10 but it will not go lower thanapprox. 687RPM, even though the lowest permissible setpoint is 450RPM.

Well, okay for now. Maybe someday I'll find out how to get it at its knees at 450RPM.
Thanks a lot for your patient assistance and the nice piece of hard- and software!

arjenhiemstra, 2022-06-29:
>No problem. API vremote=low switches out of "auto" mode, but vremote=auto brings it back. vremote=medium does nothing.

That seems just how it should be.

>Indeed. Setting[26] was 30, I've lowered it to 10 but it will not go lower thanapprox. 687RPM, even though the lowest permissible setpoint is 450RPM.

If the lowest setting results in approx 1,5-2,0% Speed status (slider on 9 or 10), that is the lowest possible setting without switching the fan off. I register a Fan speed (rpm) of 684 at that setting, seems comparable. 
Below this (if you have switched on additional sensor support) the motor will turn off.

>Thanks a lot for your patient assistance and the nice piece of hard- and software!

No problem, your welcome! Thanks for following through :)

closed 2022-06-30 by sciurius