arjenhiemstra/ithowifi #96: Option to let remotes adjust MQTT itho/state

blaauboer, 2022-08-01:
**Describe the bug**
Remotes connected to the Itho (not the add-on) don't adjust mqtt topic itho/state (probably by design).

**Expected behaviour**
Is it possible to create a option that lets remotes connected to the Itho-unit (not the add-on) adjust MQTT topic itho/state based on the given input on the remote? That way remote commands will be visible through MQTT, while maintaining all standard functionality of the RFT CO2 230V being connected to the Itho-unit.

**Device information**
- Current firmware version: 2.4.0-alpha8
- Hardware revision: 2
- CC1101 RF module enabled: yes

arjenhiemstra, 2022-09-06:
FYI; MQTT commands are already visible on MQTT through the remotesinfo topic
Would you want to have values like 'low', 'timer1' etc. on the itho/state topic while the remote in question is still connected to the itho (and with that not controlling the itho through the add-on)?

blaauboer, 2022-09-06:
Yes, I would like one topic (like itho/state) to reflect the state commanded via MQTT and commanded by a rtf remote that is still connected to the Itho. Right now I command by setting itho/cmd and get the state by getting itho/state, consequently I don't get the correct state via MQTT when I change the rtf remote on the wall. I tried using an other status topics in itho/ithostatus, but those are delayed and/or not usable. I hope this explains the issue a bit better.

closed 2022-11-14 by stale[bot]