arjenhiemstra/ithowifi #97: Information HRU250 and HRU300 devices

dennisvbussel, 2022-08-02:
Hi, 

Since [Version 2.4.0-alpha8](https://github.com/arjenhiemstra/ithowifi/releases/tag/Version-2.4.0-alpha8) support for HRU250 and HRU 300 devices has been added. What are the details for this?

In my home setup I have a HRU ECO 300 installed and I'm interested in integrating this into Home Assistant. If needed I can help with testing.

EdwinGH, 2022-08-18:
I filtered the Tweakers posts in the forum [Itho Daalderop ventilatie - wifi control add-on module](https://gathering.tweakers.net/forum/list_messages/1976492) on author [Onl1ne1373](https://tweakers.net/gallery/865609):  https://gathering.tweakers.net/forum/find/poster/865609/messages, which by reading from bottom to top seems quite instructive.
I just ordered the Itho HRU/DemandFlow/QualityFlow wifi module with the CC1101 to receive the CO2 meter commands, so ready to start when I get them.

dennisvbussel, 2022-09-04:
Thanks for pointing me to information @EdwinGH, 

I bought the same pci-express extension cable, soldered a network cable to it and it works perfectly.
[image]

I made the following panel in Home Assistant:
[image]

EdwinGH, 2022-09-05:
Looks nice!
I did a full sensor replacement (under warranty, old sensors were very poorly designed) yesterday, so will start the servicetool today to see if my issues are gone. Then I will also start my project, hopefully it will look as clean as yours :-)

bokkoman, 2022-11-11:
I am new to this, so pardon me.
But is it working on HRU300 now? Since i have that one. Together with a C02 sensor remote in the living and a 4-button (timer) remote in the bathroom.
But I kind of want to implement this in Home Assistant to automate fan speed with humidity sensors.

arjenhiemstra, 2022-11-11:
No problem, so officially it is not supported because the connector on the add-on is not compatible but unofficially; the interface is electrically compatible and the add-on has support for the HRU300 in its firmware.
There might be a special version in the future for the HRU300, for now you can use the cable example mentioned earlier in this issue.

bokkoman, 2022-11-11:
> No problem, so officially it is not supported because the connector on the add-on is not compatible but unofficially; the interface is electrically compatible and the add-on has support for the [...]

What exactly does that mean?
When i buy the non cve board, and such a pci cable, it more or less works?

arjenhiemstra, 2022-11-11:
well, it is just as functional as any other itho device supported by the add-on. But you will have to make the cable yourself indeed.

arjenhiemstra, 2022-11-11:
@dennisvbussel 
If the add-on would contain the connector on the left, it would work with a standard pce-i power splitter cable like on this image correct?
[image]

bokkoman, 2022-11-11:
It would be nice if I would have to cut any cables. But just put another cable in between the cake coming out HRU and the little screen.

bokkoman, 2022-11-17:
I noticed that there is a service-port on the front of the screen. Can't this be used instead?
There are also 2 connection ports (rj11?) on the side, have no idea what these are used for.

Onl1ne1373, 2022-11-22:
@EdwinGH, For the status string I made this:
```
- name: Itho WTW Status String
  state_topic: "itho-wtw/ithostatus"
  value_template: >
    {% set st = value_json.Status| int %}
    {% if st == 4 %}
    Warmte terugwinning
    {% elif st == 8 %}
    Koelte terugwinning
    {% elif st == 9 %}
    Zomer bypass
    {% endif %}
  availability_template: >
    {% set st = value_json.Status| int %}
    {% if st in [4,8,9] %}
    True
    {% endif %}
  unique_id: "itho_wpu_status_string"
```

@arjenhiemstra I think you are correct on the cable.

@bokkoman The front micro usb service port can be used with the Itho ServiceTool software. I tried to connect a Raspberry Pi, but the i2c driver of the Pi kernel did not support the right addressing mode.

EdwinGH, 2022-11-22:
@Onl1ne1373 Thank you for trying to parse the status value; so far I have not looked into this.
I will take a look in my logs to see if your script would generate the correct string. I think I remember my (HRU300) status is mostly 5, so I hope it does not depend on model or firmware...

@bokkoman I used the micro USB on the screen for connecting a Windows laptop running the Itho service tool (to read out system values before I used Itho WiFi, and upgrading firmware).

Onl1ne1373, 2022-11-22:
@EdwinGH I got these string status values by using the ServiceTool to check. I’m interested in other values. I haven’t seen 5 yet.

EdwinGH, 2022-11-25:
Hi @Onl1ne1373, in the ServiceTool I only see status _numbers_ myself... I see some examples online where text strings like 'Cold Recovery' are shown, but not with my device. I also see the online screenshots show HRU300 firmware version 9; mine only updated up to version 6 (though IthoWiFi reports 7), so I may have an older hardware revision.
In the ServiceTool screenshots I took I see status numbers of 4, 8, and 9, but since I log with the IthoWiFi software I only encountered status 5, 8, and 9.
Full Itho status system value list of my HRU300 with IthoWiFi firmware version: 2.4.3 / Hardware revision NON-CVE 1:
[image]

desertspider, 2022-12-11:
> Looks nice! I did a full sensor replacement (under warranty, old sensors were very poorly designed) yesterday, so will start the servicetool today to see if my issues are gone. Then I will also start my project, hopefully it will look as clean as yours :-)

Maybe een little off-topic, but what kind of issues your where having that needed a sensor replacement? I have one of the first units. I want to make sure I won't get problems in the future.

EdwinGH, 2022-12-11:
Hi @desertspider , my first observation was that the ventilation unit never scaled up the fan (I have it connected to the RTF CO2 sensor). Then I opened the unit and noticed the poor status of the temperature sensors, and decided to connect the service tool to see if any values we off. And indeed, most of the temperature sensors gave high negative values. Then I contacted Itho Daalderop that I thought my unit was not working correctly (did not tell I opened the unit nor that I connected the service tool) and they told me they were aware of the problem and could send new sensors with wires to my installer. They did and we installed the new sensors together. Now sensors are fine, but the fan has become much louder over time, and I have regular freezes of the HMI unit (which I can monitor with Itho WiFi and do power cycles until it works)...

desertspider, 2022-12-13:
Thank you for your reply. I will receive the Wifi board today, and will test if my HRU300 is ok or not.

Nefkens, 2022-12-24:
The non-CVE wifi module works good with the HRU300. Only i dont get the virtual remote to work. It wont join, so i cannot control the module. I my assumption correct that for the vitrualremote to work, i dont need the CC1101 RF Module since it will communicate by I2C?

@EdwinGH lucky my temperature readings seems ok. 

> Hi @Onl1ne1373, in the ServiceTool I only see status _numbers_ myself... I see some examples online where text strings like 'Cold Recovery' are shown, but not with my device. I also see the online [...]

With a "Absolute speed of the fan" of  52% i get a "The mass flow of the air leaving the house (kg/h)" of 150. How do you get 362 with 3.5% fan speed?

dennisvbussel, 2022-12-28:
> The non-CVE wifi module works good with the HRU300. Only i dont get the virtual remote to work. It wont join, so i cannot control the module. I my assumption correct that for the vitrualremote to [...]

What software version do you run? 

Sometimes my HRU seems to lock-up due to the fact that in this configuration the I2C bus has two masters. When the HRU is locked up, i'm unable to change the ventilation mode (even by the official Itho remotes). This is fixed after a power on reset by unplugging the HRU for a couple of seconds.

desertspider, 2022-12-28:
> What software version do you run?

Itho device type: HRU 250-300
Itho fw version: 8

Arrrowman, 2023-01-05:
@dennisvbussel 
Do you have specs for the utp cable you soldered to de pci cable. I have hru eco 300 myself and cannot connect the non- cve now. As i understand now i can buy the splitter pci cable and connect one to the hmi female cable. The other should go to an UTP tot connect to the module. Is there  a schematic you used so i know which cables tot connect to the utp or are the female pci to utp cables tot can buy?

I am new to this tinkering myself so want to prevent making mistakes you already encountered in your learning curve

Arrrowman, 2023-01-05:
> 

I have the same issues i have 2x co2 sensors and 5x spiders connected to the wtw unit. I had tot reset so often that i added a Shelly plug to reset the wtw by app tot prevent running to the attic every time i could not change the ventilation mode. Must say its going well for some weeks now...

dennisvbussel, 2023-01-05:
> @dennisvbussel Do you have specs for the utp cable you soldered to de pci cable. I have hru eco 300 myself and cannot connect the non- cve now. As i understand now i can buy the splitter pci cable [...]

Let's put it this way. If I would re-do my wiring I would go for the PCI extension/splitter cable to expose the HRU I2C wiring. Then I would cut of the female PCI connector and a ethernet cable with some wires attached so I can solder them together. 

I will make a quick drawing in the weekend.

EdwinGH, 2023-01-06:
I documented for myself the following; maybe that will help:
PCIe connector pinout (looking into male plug):
[image]
So according to this official pinout:
- PCIe PIN 1 (Square shape, Yellow wire)  - Itho SCL
- PCIe PIN 2 (Half-Diamond shape, Yellow wire)  - Itho 12V
- PCIe PIN 3 (Half-Diamond shape, Yellow wire)  - (not used)
- PCIe PIN 4 (Half-Diamond shape, Black wire)  - Itho SDA
- PCIe PIN 5 (Half-Diamond shape, Black wire)  - Itho GND
- PCIe PIN 6 (Square shape, Black wire)  - (not used).

Official UTP network cable pinout:
[image]
So according to this pinout :
- GND is blue (wire 4 from top)
- SCL is orange ( wire 2 from top)
- SDA is half brown (wire 7 from top)
- 12V is half blue ( wire 3 from top)

Hope this helps!

desertspider, 2023-01-06:
> I documented for myself the following; maybe that will help: PCIe connector pinout (looking into male plug, ignore the color codes and use on the right side): [image] So according to this official pinout:
> 
>     * PCIe PIN 1 (Square shape, Yellow wire)  - Itho SCL
> 
>     * PCIe PIN 2 (Half-Diamond shape, Yellow wire)  - Itho 12V
> 
>     * PCIe PIN 3 (Half-Diamond shape, Yellow wire)  - (not used)
> 
>     * PCIe PIN 4 (Half-Diamond shape, Black wire)  - Itho SDA
> 
>     * PCIe PIN 5 (Half-Diamond shape, Black wire)  - Itho GND
> 
>     * PCIe PIN 6 (Square shape, Black wire)  - (not used).
> 
> 
> Official UTP network cable pinout: [image] So according to this pinout :
> 
>     * GND is blue (wire 4)
> 
>     * SCL is orange ( wire 2)
> 
>     * SDA is half brown (wire 7)
> 
>     * 12V is half blue ( wire 3)
> 
> 
> Hope this helps!

Thank you for the pinout.
A important note: the pinout is true when looking at the back of the Male connector, not the front. I made this mistake myself when using this pinout in the Tweakers topic. I had to resolder two wires :). Also the text in the pictures does not match the true pinout.

Arrrowman, 2023-01-06:
Thanks guys for the quick responses. By reading the answer i got confused. 12V is it on half blue or on wire 3 which is half green? Wanna make sure i don't connect that one wrong...

EdwinGH, 2023-01-06:
Oops my mistake: It is position 3 so half green! I corrected the original post.

desertspider, 2023-01-07:
Strange, I used the colors from [This](https://gathering.tweakers.net/forum/list_message/72088312#72088312) post and it works. I am 100% sure I used the half blue wire for 12v.

EdwinGH, 2023-01-07:
Let me check my actual cable before making any other update. Sorry for the confusion!

Arrrowman, 2023-01-07:
Thanks i got these from Arjan before maybe it can help sort this out.
[PastedGraphic-2.pdf](https://github.com/arjenhiemstra/ithowifi/files/10365755/PastedGraphic-2.pdf)
[PastedGraphic-1.pdf](https://github.com/arjenhiemstra/ithowifi/files/10365756/PastedGraphic-1.pdf)

thijsdejong, 2023-01-08:
I have made the RJ45 to pci express cable. I have verified the connections with a multimeter. The board boots up just fine, I'm still facing a blocking issue:

`Itho I2C connection status: init failed - please power cycle the itho unit -`

Querying the device type in the Debug tab also does not seem to work. Rebooting the HRU300 also does not seem to work. Does anyone have any ideas on how I could debug this?

bokkoman, 2023-02-15:
I managed to install the noncve board today.
Hooked it up to my HRU 300 and turns on and all.

Now my question is, how do I join the board to the unit?
I have the c1101 addon on my board, installed by Arjen himself.

bokkoman, 2023-02-16:
> Strange, I used the colors from [This](https://gathering.tweakers.net/forum/list_message/72088312#72088312) post and it works. I am 100% sure I used the half blue wire for 12v.

Did you by chance get a virtual remote to work with your HRU300?

EdwinGH, 2023-02-16:
I have not been able to get the virtual remote to work, neither any fan settings from the API or webpages. I  am building a basic version to monitor I2C traffic to see what commands are on the HRU300 bus, but this is only progressing slowly from my side...

bokkoman, 2023-02-17:
> I have not been able to get the virtual remote to work, neither any fan settings from the API or webpages. I am building a basic version to monitor I2C traffic to see what commands are on the HRU300 bus, but this is only progressing slowly from my side...

Hmm that sux, i really hoped i could control the unit.
Is there anything i can help with?

bokkoman, 2023-02-17:
@arjenhiemstra is there anything we can do to be able to control the unit?

Or maybe it is possible to send commands with the RF module? Instead of sending them through i2c?

bokkoman, 2023-02-18:
I found this post about someone who reverse engineered the RF signal.

https://www.progz.nl/blog/2015/05/19/reverse-engineering-remote-itho-cve-eco-rft-part-6/

His GitHub:
https://github.com/xs4free/Itho-library

Not sure if it's useful, but perhaps this way we can control the HRU 200/300.

arjenhiemstra, 2023-02-19:
> I found this post about someone who reverse engineered the RF signal.
> 
> https://www.progz.nl/blog/2015/05/19/reverse-engineering-remote-itho-cve-eco-rft-part-6/

Have a look at the last comment on that page ;-)

arjenhiemstra, 2023-02-19:
> @arjenhiemstra is there anything we can do to be able to control the unit?
> 
> Or maybe it is possible to send commands with the RF module? Instead of sending them through i2c?

That last part is certainly possible. All the hardware is there, the sending part as well but currently limited to only one type of remote. Rework is needed to implement it in a proper way. Somebody was working on this back in december, need to check what to current status of that work is.

1ntroduc3, 2023-02-21:
> That last part is certainly possible. All the hardware is there, the sending part as well but currently limited to only one type of remote. Rework is needed to implement it in a proper way. Somebody [...]

That would be great. I would like to be able to monitor and manage my fan speeds in house.

Tom5990, 2024-09-20:
I've been struggling to get the virtualremote to connect as the webinterface buttons seem to do nothing. The following steps worked for me:
1. Setup virtual remote using the Copy ID function from an existing physical remote.
2. Use api commands for RF signals, e.g. http://192.168.4.1/api.html?rfremotecmd=high
or
http://nrg-itho-A1B2.local/api.html?rfremotecmd=high

Although I couldn't pair the remote it still broadcasted the right signal and because it's a copy of the physical remote the unit did respond. I'm using the wifi addon with the Itho HRU ECO 300

arjenhiemstra, 2024-09-20:
There is a important difference between the Virtual Remote (this uses the I2C bus, not supported by the HRU300) and the RF devices section. The latter uses the CC1101 module and this is what you need. 
All remote actions for the HRU300 should be done under the menu "RF devices" (or through the API, also there be sure to use the RF API commands. See menu API, section "Commands only for devices with CC1101 module" 

Hope this helps!

fmurodov, 2025-12-10:
Can't disconnect PCI-E connector from my HRU as it's stuck and I can't pull it without damaging.
Does anyone know connector type and pinout on front panel side so I could use it for splitting
[image]
[image]