arjenhiemstra/ithowifi #98: No MQTT connection

Klabatzie, 2022-08-07:
Since a wile now I'm unable to get an MQTT connection.

In the MQTT setup page I get the status connection timeout.

Log file from MQTT broker is the following: 

Client connection from 192.168.178.35 failed: error:1408F10B:SSL routines:ssl3_get_record:wrong version number.

Logfile:
2022-07-28 03:27:46 N: Mem free: 32104, Mem low: 30480, Mem block: 31968
2022-07-28 09:27:54 N: Mem free: 32104, Mem low: 30480, Mem block: 31096
2022-07-28 15:28:03 N: Mem free: 32272, Mem low: 30480, Mem block: 32088
2022-07-28 21:28:10 N: Mem free: 32488, Mem low: 30480, Mem block: 31760
2022-07-29 03:28:20 N: Mem free: 32360, Mem low: 30480, Mem block: 31160
2022-07-29 09:28:31 N: Mem free: 32272, Mem low: 30480, Mem block: 31968
2022-07-29 15:28:44 N: Mem free: 32544, Mem low: 30480, Mem block: 31648
2022-07-29 21:28:56 N: Mem free: 32512, Mem low: 30480, Mem block: 32152
2022-07-30 03:29:04 N: Mem free: 32104, Mem low: 30480, Mem block: 32296
2022-07-30 09:29:17 N: Mem free: 32432, Mem low: 30480, Mem block: 32336
2022-07-30 15:29:28 N: Mem free: 32432, Mem low: 30480, Mem block: 31696
2022-07-30 21:29:42 N: Mem free: 32104, Mem low: 30480, Mem block: 31776
2022-07-31 03:29:51 N: Mem free: 32104, Mem low: 30480, Mem block: 31096
2022-07-31 09:30:06 N: Mem free: 32104, Mem low: 30480, Mem block: 31096
2022-07-31 15:30:13 N: Mem free: 32272, Mem low: 30480, Mem block: 31696
2022-07-31 21:30:13 N: Mem free: 33632, Mem low: 30480, Mem block: 31888
2022-08-01 03:30:13 N: Mem free: 33632, Mem low: 30480, Mem block: 31888
2022-08-01 09:30:13 N: Mem free: 33632, Mem low: 30480, Mem block: 31888
2022-08-01 15:30:13 N: Mem free: 33656, Mem low: 30480, Mem block: 31888
2022-08-01 21:30:13 N: Mem free: 33632, Mem low: 30480, Mem block: 31888
2022-08-02 03:30:13 N: Mem free: 33632, Mem low: 30480, Mem block: 31888
2022-08-02 09:30:14 N: Mem free: 33632, Mem low: 30480, Mem block: 31888
2022-08-02 15:30:14 N: Mem free: 33656, Mem low: 30480, Mem block: 31888
2022-08-02 21:30:14 N: Mem free: 33632, Mem low: 30480, Mem block: 31888
2022-08-03 03:30:14 N: Mem free: 33632, Mem low: 30480, Mem block: 31888
2022-08-03 09:30:14 N: Mem free: 33632, Mem low: 30480, Mem block: 31888
2022-08-03 15:30:14 N: Mem free: 33656, Mem low: 30480, Mem block: 31888
2022-08-03 21:30:15 N: Mem free: 33632, Mem low: 30480, Mem block: 31888
2022-08-04 03:30:15 N: Mem free: 33632, Mem low: 30480, Mem block: 31888
2022-08-04 09:30:15 N: Mem free: 33632, Mem low: 30480, Mem block: 31888
2022-08-04 15:30:15 N: Mem free: 33656, Mem low: 30480, Mem block: 31888
2022-08-04 21:30:15 N: Mem free: 33632, Mem low: 30480, Mem block: 31888
2022-08-05 03:30:15 N: Mem free: 33632, Mem low: 30480, Mem block: 31888
2022-08-05 09:30:16 N: Mem free: 33632, Mem low: 30480, Mem block: 31888
2022-08-05 15:30:16 N: Mem free: 33656, Mem low: 30480, Mem block: 31888
       222 N: System boot, last reset reason: External System
       394 N: HW rev: 1, FW ver.: 2.2.2
1970-01-01 00:00:04 N: WiFi: connection successful
1970-01-01 00:00:04 N: WiFi info:
1970-01-01 00:00:04 N: Mode:STA
1970-01-01 00:00:04 N: PHY mode:N
1970-01-01 00:00:04 N: Channel:1
1970-01-01 00:00:05 N: AP id:0
1970-01-01 00:00:05 N: Status:5
1970-01-01 00:00:05 N: Auto connect:0
1970-01-01 00:00:05 N: SSID (10):(Correct SSID)
1970-01-01 00:00:05 N: IP:192.168.178.35
1970-01-01 00:00:06 N: Setup: Virtual remote ID: 227,110,21
1970-01-01 00:00:21 N: MQTT: connection failed, System config: 1
1970-01-01 00:00:21 N: Webserver: started
1970-01-01 00:00:21 N: mDNS: started
1970-01-01 00:00:21 N: Hostname: nrg-itho-6e15
1970-01-01 00:00:22 N: Setup: done
2022-08-06 01:44:25 N: Mem free: 32680, Mem low: 25704, Mem block: 32376
2022-08-06 07:44:35 N: Mem free: 32680, Mem low: 25704, Mem block: 31928
2022-08-06 13:44:39 N: Mem free: 32512, Mem low: 25704, Mem block: 32872
2022-08-06 19:44:46 N: Mem free: 32512, Mem low: 25704, Mem block: 31288
2022-08-07 01:44:59 N: Mem free: 32512, Mem low: 25704, Mem block: 31888
2022-08-07 07:45:07 N: Mem free: 32736, Mem low: 25704, Mem block: 31456
2022-08-07 13:45:08 N: Mem free: 31656, Mem low: 25704, Mem block: 30080

arjenhiemstra, 2022-09-06:
Do you have any security setup for MQTT? Have you recently upgraded MQTT to a newer version maybe?

My advise to get to the root cause of the issue would be to start with the most basic MQTT config possible (ie. no login, password or other settings) and see if the issue still persists. If not, build out from that config.

Klabatzie, 2022-09-19:
You can close this issue the problem is probably on my side.
The ssl certificate is wrong. Now i need to figure out how to correct it.

 Client connection from 192.168.178.35 failed: error:1408F10B:SSL routines:ssl3_get_record:wrong version number.
2022-09-19 20:04:39: Client connection from 192.168.178.35 failed: error:1408F10B:SSL routines:ssl3_get_record:wrong version number.

closed 2022-09-19 by Klabatzie