arjenhiemstra/ithowifi #99: Spontaneous reboots

janbredenbeek, 2022-08-18:
Every once in a while my non-CVE module reboots spontaneously. This appears to happen more frequently since I upgraded from 2.4.0-alpha6 to 2.4.0-alpha8.
This is a grep from the current debug log (I upgraded on 2022-08-17 at 10:12):

2022-08-16 23:13:27 I: System boot, last reset reason: SDIO_RESET
2022-08-17 10:12:25 I: Reboot requested
2022-08-17 10:10:08 I: System boot, last reset reason: SDIO_RESET
2022-08-17 10:10:17 I: System boot, last reset reason: SW_RESET
2022-08-17 17:15:39 I: System boot, last reset reason: RTCWDT_SYS_RESET
2022-08-17 17:31:12 I: Reboot requested
2022-08-17 17:31:13 I: System boot, last reset reason: SDIO_RESET
2022-08-18 05:17:40 I: System boot, last reset reason: RTCWDT_SYS_RESET
2022-08-18 05:33:13 I: Reboot requested
2022-08-18 05:33:15 I: System boot, last reset reason: SDIO_RESET
2022-08-18 05:45:38 I: System boot, last reset reason: RTCWDT_SYS_RESET
2022-08-18 06:01:11 I: Reboot requested
2022-08-18 06:01:13 I: System boot, last reset reason: SDIO_RESET

Interestingly, only the first 'Reboot requested' was actually initiated by me.
From the previous logfile (with 2.4.0-alpha6):

2022-08-06 22:24:06 I: System boot, last reset reason: SW_RESET
2022-08-06 22:44:28 I: System boot, last reset reason: RTCWDT_SYS_RESET
2022-08-09 22:52:47 I: System boot, last reset reason: SW_RESET
2022-08-09 22:58:42 I: System boot, last reset reason: OWDT_RESET
2022-08-10 04:58:45 I: System boot, last reset reason: RTCWDT_SYS_RESET
2022-08-16 09:55:30 I: System boot, last reset reason: SW_RESET

The wifi connection to my access point is good and shows no packet loss.
Any ideas?

arjenhiemstra, 2022-09-06:
How do you power your non-CVE module?

janbredenbeek, 2022-09-07:
> How do you power your non-CVE module?

From the HRU-350 unit.

closed 2022-10-19 by janbredenbeek