arjenhiemstra/ithowifi #101: Reboot redirects to incorrect address

sciurius, 2022-09-02:
**Describe the bug**
When rebooting using the web interface, the page is refreshed to "nrgito.local" even if that is not the host name of the unit.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to Restart/Restore device page
2. Click on Restart device
3. Confirm the pop-up dialog. Note that the dialog displays the actual host name.
4. After a couple of seconds the page refreshes to "IWe can’t connect to the server at nrgitho.local."

**Expected behaviour**
The page shoudl refresh to the actual host name.

**Device information**
- Firmware version [e.g. 2.2.1] 2.4.0-alpha8
- Hardware revision (e.g. non-cve 1.1) CVE-S hw 2

**Desktop (please complete the following information):**
 - OS: [e.g. iOS] Fedora 35
 - Browser [e.g. chrome, safari] Firefox
 - Version [e.g. 22] 103.0.2

arjenhiemstra, 2022-09-06:
Do you mean using the menu Reset and then clicking the button "Restart device"
I seem not to be able to reproduce it somehow...

arjenhiemstra, 2022-09-07:
I see, you are using a fully qualified domain name (FQDN) as hostname setting. That will not work indeed, you should only use "nrgitho" and leave out the ".squirrel.nl" part. The add-on uses mDNS to be able to be reachable on you local network, mDNS uses the .local "TLD".

sciurius, 2022-09-07:
Initially, there was just "nrgitho" and it didn't work. So I tried "nrgitho.squirrel.nl" instead (without success).
I'd expect the add-on to not redirect to a `.local` address if there is a FQDN.

arjenhiemstra, 2022-09-07:
You cannot use a FQDN as hostname, DNS does not allow it. You could, if you have a DNS server running, make a CNAME record but that will reside on your DNS server, not on a specific host.
The add-on is such a host and uses mDNS with .local (which is standard I believe), this a way to enable the use of an URL without having to have a fully configured DNS server. Is you want to use s FQDN you should setup DNS, for example with an A record, to point to the IP address of the add-on.

sciurius, 2022-09-07:
My DNS has an A record for the add-on so that's not the problem.
But as I said, the issue is minor.
(BTW, all Tasmota things correctly redirect in case of reboot.)

vandenberghev, 2022-10-04:
I've experienced the same 'issue'. The add-on just assumes you're using the mDNS name and redirects to there, instead of using the actual URI the page was requested from (be it an IP, an FQDN, ...). So if you use `http://192.168.x.x` and the add-on has to reboot, you'll be redirected to the mDNS name. This could cause timeouts if mDNS does not work properly on your network, or if the add-on is only reachable through an rProxy, or...

sanderkob, 2022-10-11:
Or when mDNS not works, as in the older android versions (such as mine).

closed 2022-12-17 by stale[bot]