arjenhiemstra/ithowifi #102: Error: Task SysControl timed out!

SupportKraatje, 2022-09-03:
Used to work fine, just stopped......

2022-09-03 09:03:11 N: System boot, last reset reason: SDIO_RESET
2022-09-03 09:03:11 N: HW rev: 2, FW ver.: 2.3.5
2022-09-03 09:03:15 N: WiFi: connection successful
2022-09-03 09:03:15 N: WiFi info:
2022-09-03 09:03:15 N: Mode:STA
2022-09-03 09:03:15 N: Status:3
2022-09-03 09:03:16 N: IP:192.168.1.135
2022-09-03 09:03:16 N: Setup: Virtual remote ID: 205,194,36
2022-09-03 09:03:16 N: Webserver: started
2022-09-03 09:03:16 N: mDNS: started
2022-09-03 09:03:16 N: Hostname: nrg-itho-c224
2022-09-03 09:03:16 N: Setup: done
2022-09-03 09:03:51 N: Error: Task SysControl timed out!
2022-09-03 09:03:59 N: System boot, last reset reason: SDIO_RESET
2022-09-03 09:03:59 N: HW rev: 2, FW ver.: 2.3.5
2022-09-03 09:04:03 N: WiFi: connection successful
2022-09-03 09:04:03 N: WiFi info:
2022-09-03 09:04:03 N: Mode:STA
2022-09-03 09:04:03 N: Status:3
2022-09-03 09:04:03 N: IP:192.168.1.135
2022-09-03 09:04:04 N: Setup: Virtual remote ID: 205,194,36
2022-09-03 09:04:04 N: Webserver: started
2022-09-03 09:04:04 N: mDNS: started
2022-09-03 09:04:04 N: Hostname: nrg-itho-c224
2022-09-03 09:04:04 N: Setup: done

DohmenICT, 2022-09-05:
I have the same issue. 

Wifi works. MQTT works

Itho Settings show: (Retrieve settings doesn't work)

```
Configuration of the Itho unit

Itho device type: Unkown device type
Itho fw version: 0
```

Itho status shows no Labels/Values.

I had the 2.4 Beta 8 firmware on this when it happened. Since then I've went back to 2.3.5 but that didn't help.

This is the config.json:

`
{"sys_username":"admin","sys_password":"admin","syssec_web":0,"syssec_api":0,"syssec_edit":0,"api_normalize":0,"syssht30":0,"itho_rf_support":0,"rfInitOK":false,"itho_fallback":20,"itho_low":20,"itho_medium":120,"itho_high":220,"itho_timer1":10,"itho_timer2":20,"itho_timer3":30,"itho_updatefreq":5,"itho_sendjoin":0,"itho_forcemedium":0,"itho_vremoteapi":0,"nonQ_cmd_clearsQ":1,"mqtt_active":0,"mqtt_serverName":"192.168.1.123","mqtt_username":"","mqtt_password":"","mqtt_port":1883,"mqtt_version":1,"mqtt_state_topic":"itho/state","mqtt_ithostatus_topic":"itho/ithostatus","mqtt_remotesinfo_topic":"itho/remotesinfo","mqtt_lastcmd_topic":"itho/lastcmd","mqtt_state_retain":"yes","mqtt_cmd_topic":"itho/cmd","mqtt_lwt_topic":"itho/lwt","mqtt_domoticz_active":0,"mqtt_ha_active":0,"mqtt_ha_topic":"homeassistant","mqtt_idx":1,"sensor_idx":1,"version_of_program":"005"}
`

arjenhiemstra, 2022-09-06:
I suspect the firmware of your itho devices crashed. It seems that some itho devices access the I2C interface without checking if there is activity on the interface leading to a crash. I'm looking into it.
For now a powercylce of your itho should 'fix' the issue. Also lowering the update speed from the standard 5 sec to a more moderate 10 sec. will also help.

TD-er, 2022-09-06:
> I suspect the firmware of your itho devices crashed. It seems that some itho devices access the I2C interface without checking if there is activity on the interface leading to a crash. I'm looking into it.

Or.... the I2C bus is stuck.
This may happen more often than you'd think.
What I do in ESPEasy is this:
- Set timeout on I2C object
- On timeout or when starting a cycle, detect if either SDA or SCL is pulled low for quite a long time
- reverse SDA and SCL and send something (doesn't really matter what) until SDA and SCL are high again (of course take timeout into account)
- revert SDA and SCL pins again and restart the I2C bus.

arjenhiemstra, 2022-09-06:
Looking into this indeed. Thing is with this particular situation, this is a multi master setup where (at least from the itho side) there seems (for some firmware versions?) no checking of I2C bus status before a transmit is started crashing itself. 

Luckily there is a regular pattern, at least for CVE devices, where the itho firmware sends something on the I2C bus. The idea now is to implement some control code to check for this pattern and only send something on the bus after x amount of time until y amount of time before the next I2C communication is to be expected. But it needs to be tested for other itho devices as well.

TD-er, 2022-09-06:
Do you also have devices on the bus which may use clock stretching?

SupportKraatje, 2022-09-06:
No, only login with the webbrower (now standalone, will connect it later to domoticz)

arjenhiemstra, 2022-09-06:
> Do you also have devices on the bus which may use clock stretching?

I expect that indeed. It has been al long time ago but I remember during my initial reverse engineering efforts I needed to adjust some things around clock stretching to get it to work reliable.

arjenhiemstra, 2022-09-06:
btw @SupportKraatje could you please provide the details of your itho device after a restart? 
This info can be found on the itho settings page.
ie.: 
[image]

TD-er, 2022-09-06:
OK, so then it might not be useful to use the statistical approach you suggested.
It is better to wait to see if both lines are "up" before you initiate an I2C transfer.
If one of them is "down" for prolonged amount of time, you can assume one of the devices that may try to clock stretch  is either waiting for something or actually hanging.
Most often either the master or the sensor missed 1 pulse somewhere.

By temporary swapping SDA and SCL and simply initiating something like a single I2C scan call, or making some call to a non-existing device, you will "push" that hanging device out of its waiting state.

There have to be some pulses on the data and clock to get it to work and depending on where things get stuck, you might have to send various sets of data with different amount of pulses on the data pin.

Luckily there is no need for the I2C clock to be at a fixed frequency, so that's why this may work to unlock the I2C bus when stuck.

SupportKraatje, 2022-09-07:
> btw @SupportKraatje could you please provide the details of your itho device after a power cycle? This info can be found on the itho settings page. ie.: [image]

Itho device type: CVE-Silent
Itho fw version: 27

SupportKraatje, 2022-09-11:
ok the 10 seconds interval did not help. Hangs again, no fan readings or control (always 0)

sciurius, 2022-09-27:
FWIW, I seem to be running into the same issue with 2.4.0-alpha8.
````
2275 I: System boot, last reset reason: POWERON_RESET
2482 I: HW rev: 2, FW ver.: 2.4.0-alpha8
6247 I: WiFi: connection successful
6543 I: WiFi info:
6640 I: Mode:STA
6750 I: Status:3
6874 I: IP:192.168.1.115
7010 I: Setup: Virtual remotes, start ID: 224,139,139
7205 I: Setup: remotes configfile loaded
7377 I: Setup: init of CC1101 RF module successful
7805 I: Setup: remotes configfile loaded
7982 I: MQTT: connected, System config: 1
8358 I: Webserver: started
8536 I: mDNS: started
8679 I: Hostname: nrgitho
8829 I: Setup: done
````

arjenhiemstra, 2022-09-27:
> FWIW, I seem to be running into the same issue with 2.4.0-alpha8.
>
> ```
> 2275 I: System boot, last reset reason: POWERON_RESET
> 2482 I: HW rev: 2, FW ver.: 2.4.0-alpha8
> 6247 I: WiFi: connection [...]

What is your itho model and firmware version?

sciurius, 2022-09-27:
Same: CVS-E fw 27.

tjoen, 2022-10-19:
I also still encounter this issue. 
I'm already running the latest firmware (27).

Indeed, I already figured out that powercycling the unit makes it work again.
My Itho device type is CVE-Silent

I also noticed last time when it hung that the SysControl task memory had 0 bytes free.
Might be useful in finding a more permanent solution?

arjenhiemstra, 2022-10-21:
>I also noticed last time when it hung that the SysControl task memory had 0 bytes free.

That's useful info, thanks!

Please try https://github.com/arjenhiemstra/ithowifi/releases/tag/Version-2.4.3-beta2
I made some changes to the bus check and added logging to get an idea how many times a collision happens.

tjoen, 2022-10-22:
Brilliant, I will install it now Arjen.

arjenhiemstra, 2022-10-27:
I just released version 2.4.3 which addresses this further. In some situations the add-on could still access the I2C bus while another master was active. The latest code changes in 2.4.3 should prevent this or at least further lower the change of this happening.

sciurius, 2022-12-26:
I still get it occasionally

````
2022-12-22 16:25:40 I: Mem free: 103032, Mem low: 65624, Mem block: 44020
2022-12-22 21:46:59 I: Warning: Task SysControl timed out!
2022-12-22 22:54:32 I: System boot, last reset reason: SDIO_RESET
2022-12-22 22:54:32 I: HW rev: 2, FW ver.: 2.4.3
````

The reboot was manual (power flip) since the unit did not respond.

sciurius, 2023-01-05:
Up and running. Enabled the I2C safeguard. Stay tuned...

arjenhiemstra, 2023-02-17:
2.5.0 has been released as new stable release, including fixes for the described issue.

https://github.com/arjenhiemstra/ithowifi/releases/tag/2.5.0

arjenhiemstra, 2023-02-21:
> 2.5.0 has been released as new stable release, including fixes for the described issue.
> 
> https://github.com/arjenhiemstra/ithowifi/releases/tag/2.5.0

I've found a serious issue with the fail save firmware update, please do not install 2.5.0 and if you have already, wait with further updates until 2.5.1 comes out.

sciurius, 2023-04-22:
Is it?

closed 2023-04-29 by stale[bot]