arjenhiemstra/ithowifi #103: Reset Highest received CO2 value

BlueCase, 2022-09-18:
Hi,
I have a HRU350 and currently try to connect it with HomeAssistant.
By this I found few topic and hope someone can answer my questions.

1.  Is it possible to reset the Highest received CO2 value?
2. I tried the reset Filter function. Not working (Know a alpha feature but perhaps the information help)
3. What Label Status mean? Currently I have the value 5. 

Thanks in advance...

arjenhiemstra, 2022-09-23:
As far as I know there is no active way to reset the "Highest received CO2 value".

About the filter reset function; I expect it to work only if:
1- the virtual remote type is set to RFT AUTO (that is the only remote that has this function as far as I know)
2- you have sent the filter reset quick enough, I just looked at this (https://www.youtube.com/watch?v=peKzWCsLyrg) clip and I think it is quite impossible to achieve with the add-on in the current setup. Probable it needs a function like "reset filter on next boot"

>What Label Status mean? Currently I have the value 5.

Which status label do you mean? The meaning of labels not only depends on your itho model but also differs between firmware version.

BlueCase, 2022-10-23:
Thanks for your answer.

I mean the label "status" in the "itho status" view

arjenhiemstra, 2022-10-23:
Could you post a screenshot?

BlueCase, 2022-11-14:
[image]

arjenhiemstra, 2022-11-14:
||||
|-|-|-|
|1|normaal|WTW bedrijf
|2|vorstklep|Vorstklep verstellen
|3|aanvoerfan|Aanvoer ventilator aftoeren
|4|afvoerfan|Afvoer ventilator optoeren
|5|aan uit|Aanvoer ventilator stil

This is what I get from the itho tool

closed 2023-01-20 by stale[bot]

BlueCase, 2023-06-10:
@arjenhiemstra Sorry but I had not time to investigate further.
What status 5 mean. I translated it with google but the result make no sense. :D
I think the it mean  "on off -  Supply Fan off". 
But make no sense because all Fan's running. Checked that.
Wrong meassage? Normal? No idea..

Also a additional Question: Can I force open the Bypass anyway. Would only check if its work because I currently not understand exactly when the Fan open the bypass.
[image]

BlueCase, 2023-06-11:
Ok self healing... 
Seams Status 5 is always the default after a Power unplug and switch to 0 after a while.

Also the bypass need a while to work probably after a power on/off.

Thanks for your help