arjenhiemstra/ithowifi #104: WiFi setup: NTP server is not saved

vandenberghev, 2022-09-21:
On the `WiFi Setup` page, when I enter another NTP server and hit save, the setting is not retained but the original `pool.ntp.org` seems to be restored. Editing the json on the file system directly does work.

**Device information**
- Firmware version 2.4.0-beta
- Hardware revision CVE rev. 2
- CC1101 RF module enabled [ yes ]

arjenhiemstra, 2022-09-23:
Thanks for spotting this!
https://github.com/arjenhiemstra/ithowifi/blob/884c97b463dfa2cd9fd5835b3d8408b6efdd5ecc/software/NRG_itho_wifi/src/webroot_source/controls_js_sources/javascript.js#L266
nptserver != ntpserver ;-) Will be corrected in the next version.

arjenhiemstra, 2022-09-26:
Fixed with https://github.com/arjenhiemstra/ithowifi/releases/tag/Version-2.4.0

closed 2022-09-26 by arjenhiemstra