arjenhiemstra/ithowifi #106: Control via add-on stops working

RPJacobs, 2022-09-23:
**Describe the bug**
I installed two add-ons (with CC1101) on two CVE-Silent

Itho device type: CVE-Silent
Itho fw version: 11

They both work initially, enabled MQTT -> Homebridge -> Apple Homekit, removed remote from unit and joined the add-on. (works great, thanks Arjen!)

But after a tmie the fan on the ground floor stops controlling the Itho Fan. I can login to the web interface and retrieve the settings, run all debug  queries but not controle the fanspeed! I2C not working? Status page indicates connected... After a power cycle the unit works as accepted then stops again.

Can your please advice?

**To Reproduce**
not sure what is going wrong...

**Expected behaviour**
controle the fan by MQTT or the web interface.

**Screenshots**

**Device information**
Configuration of the Itho unit:
Itho device type: CVE-Silent
Itho fw version: 11

- Hardware revision (e.g. non-cve 1.1)

Current firmware version: 2.3.5
Hardware revision: 2

- Type/model number Itho (on sticker outside the box)
[image]

- CC1101 RF module enabled [ yes, no ] YES
- In case of wifi issues, wifi access point make and model: No issue

**Debug logging**
Please provide debug logging from the debug page if possible:
[logfile0.current.log](https://github.com/arjenhiemstra/ithowifi/files/9632597/logfile0.current.log)

arjenhiemstra, 2022-09-26:
If debug queries still give a valid result than 9 out of 10 times the fan is not in the correct mode for (PWM2I2C) control. 
If the situation occurs again, please go to the Itho status page off the add on and post the value of the label "FanInfo"

On the wiki there is some more info about this:
https://github.com/arjenhiemstra/ithowifi/wiki/Controlling-the-speed-of-a-fan

RPJacobs, 2022-09-26:
Thank for the reply! Dump question but there is no FanInfo on the status page:

[image]

But I did find:

[image]

RF Mode it is....

RPJacobs, 2022-09-26:
It’s in the initial post:

Itho device type: CVE-Silent 
Itho fw version: 11

arjenhiemstra, 2022-09-26:
> It’s in the initial post:
> 
> Itho device type: CVE-Silent Itho fw version: 11

Doh, sorry! Overlooked that..... grrr

RPJacobs, 2022-09-26:
I joined the remote to the fan and removed it again. Then joined the addon again. Now it does not stop working! Also updated to 2.4.0

RPJacobs, 2022-09-26:
If you need more info about the fan contact me directly.

closed 2022-09-26 by RPJacobs

arjenhiemstra, 2022-09-26:
Ah now I see, "Selection" is comparable with "FanInfo" on newer itho models.
This is the table:
0 Off
1 Holiday
2 Low
3 Nominal / auto
4 High
5 Timer

Your fan on the low setting and should be on 2 or Nominal/auto. Probably if you powercyle the unit it will be on 3. Something pushed you itho out of mode 3. Most probably that is a remote (either physical of virtual)

arjenhiemstra, 2022-09-26:
> If you need more info about the fan contact me directly.

Thanks! I'll update the wiki with this info, very helpful! Thanks for creating an issue!