arjenhiemstra/ithowifi #111: init failed - please power cycle the itho unit - (Warning: I2C timeout)

MartVisser, 2022-10-07:
**Describe the bug**
After some time, usually a day after a power cycle, the connection between the module and the unit drops out with the following error message in the logs 'Warning: I2C timeout' 

**To Reproduce**
After a power cycle, wait a day (sometimes two days)

**Expected behaviour**
N/A

**Screenshots**
N/A

**Device information**
- Firmware version: 2.4.2
- Hardware revision: 2
- Type/model number Itho: CVE-S ECO SP
- CC1101 RF module enabled: yes, but no remote configured
- In case of wifi issues, wifi access point make and model: Ubiquiti APs but no WiFi issues. 

```
Itho device type: CVE-Silent
Itho fw version: 27
```

**Debug logging**

`Itho I2C connection status: init failed - please power cycle the itho unit -`

```
[...]
2022-10-07 08:58:19 I: Warning: I2C timeout
2022-10-07 08:58:31 I: Warning: I2C timeout
2022-10-07 08:58:42 I: Warning: I2C timeout
2022-10-07 08:58:53 I: Warning: I2C timeout
[...]
```

**Desktop (please complete the following information):**
N/A

**Smartphone (please complete the following information):**
N/A

PatrickV2, 2022-10-08:
I have similar behavior with the CVE-S ECO SP.

Configuration of the Itho unit
Itho device type: CVE-Silent
Itho fw version: 27

2022-10-08 07:50:43 I: HW rev: 2, FW ver.: 2.4.2
2022-10-08 07:45:24 I: I2C init: QueryDevicetype - fw:27 hw:27

arjenhiemstra, 2022-10-08:
This is most probably a duplicate of issue #102 
It seems that there is a bug (?) in itho firmware version 27 that causes the i2c bus to crash. This is the first issue I will pick up next. (I had to put some time in wifi issues first)

TD-er, 2022-10-08:
I2C can get stuck for many reasons.
In ESPEasy I try to detect this and then temporary "swap" the SDA and SCL pins. Then try to send some data and swap them back.
This way you "reset" the bus by letting some device "finish" its communication.

PatrickV2, 2022-10-09:
> I2C can get stuck for many reasons. In ESPEasy I try to detect this and then temporary "swap" the SDA and SCL pins. Then try to send some data and swap them back. This way you "reset" the bus by letting some device "finish" its communication.

@arjenhiemstra, would this be an idea when you detect the "I2C timeout" or are you already doing this?

arjenhiemstra, 2022-10-09:
It's exactly what @TD-er suggests that I would like to try and see if this can release the I2C bus. I just haven't had time to work on it

MartVisser, 2022-10-12:
@arjenhiemstra  I installed [2.4.3-beta1](https://github.com/arjenhiemstra/ithowifi/commit/274c0c37d6f6ab5c68fb86b44d9ed2be690a3ebe), but after a few hours it failed again with the following messages in the log:
```
2022-10-12 19:15:37 I: I2C bus could not clear: SCL clock line held low
2022-10-12 19:15:48 I: Warning: I2C timeout, trying I2C bus reset...
2022-10-12 19:15:51 I: I2C bus could not clear: SCL clock line held low
2022-10-12 19:16:03 I: Warning: I2C timeout, trying I2C bus reset...
2022-10-12 19:16:05 I: I2C bus could not clear: SCL clock line held low
2022-10-12 19:16:16 I: Warning: I2C timeout, trying I2C bus reset...
```

PatrickV2, 2022-10-12:
I have downgraded to version 2.3.5 and that version is running stable the past 3 days.

arjenhiemstra, 2022-10-13:
Thanks for testing @MartVisser ! That was the worst possible outcome but at least we have that info now. I'll work further on a different approach

didericko, 2022-10-14:
I am having the same problems. Just installed the beta firmware and will post results in a few days.

didericko, 2022-10-15:
I just checked and the unit stopped responding this morning. Log file:
2022-10-15 08:02:56 I: Warning: I2C timeout, trying I2C bus reset...
2022-10-15 08:02:58 I: I2C bus could not clear: SCL clock line held low
2022-10-15 08:02:59 I: I2C init: QueryDevicetype - failed

PatrickV2, 2022-10-16:
@arjenhiemstra, just a quick update
I have been running version 2.3.5 stable for the past week so apparently something has changed in the code between version 2.3.5 and 2.4.2 which is causing this issue. Hopefully it gives you some pointers in what may be causing this.

Itho device type: CVE-Silent
Itho fw version: 27

didericko, 2022-10-18:
I downgraded to 2.4.2-beta3 and it has been running stable since Sunday morning.

didericko, 2022-10-19:
This morning unit stopped responding. I will try to downgrade to 2.3.5 today

arjenhiemstra, 2022-10-21:
Please try https://github.com/arjenhiemstra/ithowifi/releases/tag/Version-2.4.3-beta2

MartVisser, 2022-10-21:
So far so good, just some information messages;
```
2022-10-21 20:08:45 I: Info: I2C bus busy
2022-10-21 20:09:26 I: Info: I2C bus busy
2022-10-21 20:16:07 I: Info: I2C bus busy
2022-10-21 20:28:49 I: Info: I2C bus busy
2022-10-21 20:29:29 I: Info: I2C bus busy
```

arjenhiemstra, 2022-10-21:
Those info messages are good, a sign of proper working code. I will remove it once we get to a stable state.
As long as there are no new "Warning: I2C timeout, trying I2C bus reset..." in the log, I'm satisfied.

didericko, 2022-10-22:
Back in 2 weeks, will try then, thx!

MartVisser, 2022-10-27:
Sadly, it failed again after 4-5 days on 2.4.3-beta2;
```
[...]
2022-10-27 07:15:36 I: Info: I2C bus busy
2022-10-27 07:15:46 I: Warning: I2C timeout, trying I2C bus reset...
2022-10-27 07:15:49 I: I2C bus could not clear: SCL clock line held low
2022-10-27 07:15:50 I: Info: I2C bus busy
2022-10-27 07:16:00 I: Warning: I2C timeout, trying I2C bus reset...
2022-10-27 07:16:03 I: I2C bus could not clear: SCL clock line held low
2022-10-27 07:16:04 I: Info: I2C bus busy
2022-10-27 07:16:14 I: Warning: I2C timeout, trying I2C bus reset...
2022-10-27 07:16:17 I: I2C bus could not clear: SCL clock line held low
[...]
```

arjenhiemstra, 2022-10-27:
I just released version 2.4.3 which addresses this further. In some situations the add-on could still access the I2C bus while another master was active. The latest code changes in 2.4.3 should prevent this or at least further lower the change of this happening.

didericko, 2022-11-10:
The unit still stops responding after a few days.

MartVisser, 2022-11-10:
Yes here it also keeps failing on the latest version. Downgraded to 2.3.5 and has been running stably for 7 days now.

didericko, 2022-11-11:
2022-11-10 02:09:34 I: Mem free: 110144, Mem low: 72988, Mem block: 44020
2022-11-10 08:09:34 I: Mem free: 109576, Mem low: 59468, Mem block: 44020
2022-11-10 14:09:35 I: Mem free: 109564, Mem low: 59468, Mem block: 44020
2022-11-10 20:09:35 I: Mem free: 100992, Mem low: 59468, Mem block: 44020
2022-11-11 02:09:35 I: Mem free: 100588, Mem low: 59468, Mem block: 44020
2022-11-11 07:30:00 I: System boot, last reset reason: OWDT_RESET
2022-11-11 07:30:00 I: HW rev: 2, FW ver.: 2.4.3
2022-11-11 07:30:06 I: WiFi: connection successful
2022-11-11 07:30:06 I: WiFi info:
2022-11-11 07:30:06 I: SSID:TROPEN | BSSID[74:AC:B9:49:3D:74]
2022-11-11 07:30:06 I: RSSI:-55dBm
2022-11-11 07:30:06 I: Mode:STA
2022-11-11 07:30:07 I: Status:WL_CONNECTED
2022-11-11 07:30:07 I: IP:192.168.1.189
2022-11-11 07:30:07 I: Setup: Virtual remotes, start ID: 32,D1,B4 - No.: 1
2022-11-11 07:30:07 I: Setup: remotes configfile loaded
2022-11-11 07:30:07 I: Setup: init of CC1101 RF module successful
2022-11-11 07:30:07 I: MQTT: connected, System config: 1
2022-11-11 07:30:08 I: Setup: remotes configfile loaded
2022-11-11 07:30:08 I: Webserver: started
2022-11-11 07:30:08 I: mDNS: started
2022-11-11 07:30:08 I: Hostname: nrg-itho-d1b4
2022-11-11 07:30:09 I: Setup: done
2022-11-11 07:30:14 I: I2C init: QueryDevicetype - failed

didericko, 2022-11-11:
System rebooted itself this morning after being inresponsive for several days. During this period, log file showed nothing wrong, just the memory log lines. Last line in the log shows the I2C Init failure.

What is the last really stable version? I only need MQTT for HA and nothing else. Version 2.3.5?

MartVisser, 2022-11-11:
@didericko I have had the same experiences with version 2.4.3, unresponsive and nothing in the logs. Downgraded to 2.3.5 last week and have been running "stable" ever since. Sometimes it resets to "Low" mode, but I have it set to my desired value, probably a bug in 2.3.5. I also use it via MQTT in Home Assistant.

arjenhiemstra, 2022-11-11:
2.3.5 seems to be stable for users that have i2c issues. A lot of extra logging was removed from 2.4.3 because it seemed to be fixed. I'm working nog on a test version that keeps al log of the i2c transactions, in case of a failure I hope this points to the culprit.

>Sometimes it resets to "Low" mode (or "Start/fallback speed" mode, not sure), but I have it set to my desired value, probably a bug in 2.3.5. 

As far as I know this is not a known bug in 2.3.5. On MQTT there is a last command topic that might point to the reason for switching to "Low" mode.

MartVisser, 2022-11-11:
Just installed 2.4.4-beta1. I don't have to enable anything right? Just wait for the "crash" and check the debug page for the log.

arjenhiemstra, 2022-11-11:
[image]

This setting needs to be switched on

MartVisser, 2022-11-11:
Ah right, I was looking at the debug page for the setting, just enabled it at the system settings.

didericko, 2022-11-12:
OK, I'll install it tomorrow and will also report the log later

HAuser2019, 2022-11-17:
Experienced bug #111 and/or #115
Just tried 2.4.4-beta1. Initally it works.
-The slider in the webinterface does not move when using low, medium, high. Also not when sending commands via HA.
-The RFT AUTO does virtual remote does not work. Using the first RFT.
-I enable the I2C-bug checker.
-I will try to help in a limited way, the wife if allready upset.... what would be the safest firmware to revert to? (ITHO CVE-S FW27  HW2 with CC1101 RF)

bhulsken, 2022-11-17:
Having the same issue, installed 2.4.4-beta1. I managed to "unstick" the I2C using the "Reset Temp/Hum Sensor" button on the I2C debug page.

Here is the requested info, hope it helps!

`origin	start	duration	error	final
QUERY_31DA	150187814	5626	I2C_ERROR_SCL_LOW	I2C_NOK
QUERY_STATUS	150186813	1001		I2C_NOK
QUERY_31D9	150182156	2		I2C_OK
QUERY_31DA	150181973	2		I2C_OK
QUERY_STATUS	150181795	2		I2C_OK
QUERY_31D9	150177145	2		I2C_OK
QUERY_31DA	150176963	2		I2C_OK
QUERY_STATUS	150176782	2		I2C_OK
QUERY_31D9	150172133	2		I2C_OK
QUERY_31DA	150171949	2		I2C_OK
QUERY_STATUS	150171771	2		I2C_OK
QUERY_31D9	150167118	2		I2C_OK
QUERY_31DA	150166935	2		I2C_OK
QUERY_STATUS	150166755	2		I2C_OK
QUERY_31D9	150162105	2		I2C_OK`

and after the reset (after which everything functions as normal again):
`origin	start	duration	error	final
QUERY_31D9	209777899	2		I2C_OK
QUERY_31DA	209777713	2		I2C_OK
QUERY_STATUS	209777530	2		I2C_OK
QUERY_STATUS	209774463	2		I2C_OK
QUERY_STATUS_FORMAT	209774110	4		I2C_OK
QUERY_DEVICE_TYPE	209773752	2		I2C_OK
QUERY_31DA	150187814	5626	I2C_ERROR_SCL_LOW	I2C_NOK
QUERY_STATUS	150186813	1001		I2C_NOK
QUERY_31D9	150182156	2		I2C_OK
QUERY_31DA	150181973	2		I2C_OK
QUERY_STATUS	150181795	2		I2C_OK
QUERY_31D9	150177145	2		I2C_OK
QUERY_31DA	150176963	2		I2C_OK
QUERY_STATUS	150176782	2		I2C_OK
QUERY_31D9	150172133	2		I2C_OK
`

MartVisser, 2022-11-18:
After 6 days;
[image]

After pressing  the "Reset Temp/Hum Sensor" button;
[image]

After a power cycle everything is normal again as usual.

arjenhiemstra, 2022-11-18:
Thanks for the report! For now, you can move back to 2.3.5. 
I'll be including a I2C sniffer that will be able to log commands originating from the itho firmware as well. Hopefully that gives a bit more information what happend not only from the add-on side but also from the itho side.

tomkooij, 2022-11-21:
I just update to 2.4.4-beta1
non-CVE on WPU 5G  
Current firmware version: 2.4.4-beta1
Hardware revision: NON-CVE 1
I2C QueryDevicetype - fw:37 hw:13

```
Last I2C commands:

origin	start	duration	error	final
QUERY_STATUS	83062232	5685	I2C_ERROR_SCL_LOW	I2C_NOK
QUERY_31D9	83057529	3	I2C_ERROR_CLEARED_OK	I2C_NOK
QUERY_31DA	83057427	2		I2C_OK
QUERY_STATUS	83057228	2		I2C_OK
QUERY_31D9	83052527	1		I2C_OK
QUERY_31DA	83052425	2		I2C_OK
QUERY_STATUS	83052218	2		I2C_OK
QUERY_31D9	83047517	1		I2C_OK
QUERY_31DA	83047415	2		I2C_OK
QUERY_STATUS	83047196	2		I2C_OK
QUERY_31D9	83042495	1		I2C_OK
QUERY_31DA	83042393	2		I2C_OK
QUERY_STATUS	83042191	2		I2C_OK
QUERY_31D9	83037440	1		I2C_OK
QUERY_31DA	83037338	2		I2C_OK
```

Will move back to 2.3.5

Dvm08, 2022-11-27:
Had the same I2C issues last weeks. Tried to get back on 2.3.5 which also give me instant I2C error. Now tried to upgrade to newest software to try and now I can' acces the Itho anymore. Any tips to get back to a sort of factory settings?

HAuser2019, 2022-11-27:
2.3.5 is somewhat most of the time stable of my end. I disabled the humidiy internal sensor with the Addon webinterface so I control it with HA. But Sometimes I set the fan to Max with homeassistant, and then after 10min the ITHO/Addon seems to take it over, I can't go back to medium anymore. Have to take the powerplug out to reboot. But not alway's can't put my finger on the situation yet.

roelerps, 2022-12-07:
Here the same problem ( Current firmware version: 2.4.3 Hardware revision: 2) :
Itho I2C connection status: init failed - please power cycle the itho unit -

Only to be resolved by a hard power cycle. Too bad the unit is tucked away and power cycle is additional work.
Any lead on a solution?

arjenhiemstra, 2022-12-07:
> Here the same problem ( Current firmware version: 2.4.3 Hardware revision: 2) : Itho I2C connection status: init failed - please power cycle the itho unit -
> 
> Only to be resolved by a hard power cycle. Too bad the unit is tucked away and power cycle is additional work. Any lead on a solution?

For now, best to stay on 2.3.5 until a fix is available.

tomkooij, 2022-12-11:
@arjenhiemstra 2.4.4-beta3 showed this error yesterday. (connected to WPU 5G). No new data was reported to MQTT and the status page was stuck with the "i2c connection failed - please power cycle the itho unit"

However, I just rebooted the non-CVE module and it works again. No need to powercycle the actual heatpump. In fact, In an earlier crash this was the same.

This behaviour seems different from what is described above, as I it seems the Heatpump/CVE needs to be powercycled, pointing to a stuck i2c bus on the Itho heatpump/cve.  In my case the non-CVE module seemed to be stuck. 

Is this something worth investigating?

arjenhiemstra, 2022-12-11:
Because of reporting purposes with these beta's the i2c bus deliberately hangs forever on an error (in order to be able to capture the i2c log). Apparently sometimes I2C errors occur that are not related and normally would resolve itself. Not with this beta. I will change the code to log the latest log to flash in case of an error but continue normal operation if possible.

arjenhiemstra, 2023-01-05:
for those with a CVE and humidity sensor, please try 2.4.4-beta4 and enable i2c safe guard function if available on your add-on hardware revision. It seems to run smoothly on my devices but I'm more than interested how it runs on those devices that have issues.

clousberg, 2023-01-12:
hi, I have a CVE-S CO2 - Optima inside, and also had this issue on 3.4.2.
Now I installed 2.4.4-beta4 and enabled i2c safe guard.
After 3 days the I2C connection is still connected! :)

bhulsken, 2023-01-21:
> for those with a CVE and humidity sensor, please try 2.4.4-beta4 and enable i2c safe guard function if available on your add-on hardware revision. It seems to run smoothly on my devices but I'm more than interested how it runs on those devices that have issues.

I have a CVE with built in humidity and CO2 sensor (CVE-S OPTIMA IN, I2C init: QueryDevicetype - fw:27 hw:27). I've had I2C stability problems before, but with firmware 2.4.4-beta6 and "I2C init: Safe guard enabled" I have been running without an I2C failure for 7 days now (and counting). It's never been without a hang for that long before, so it definitely seems your changes help to at least reduce the occurrence of this issue.

Thanks!

arjenhiemstra, 2023-01-22:
> I have a CVE with built in humidity and CO2 sensor (CVE-S OPTIMA IN, I2C init: QueryDevicetype - fw:27 hw:27). I've had I2C stability problems before, but with firmware 2.4.4-beta6 and "I2C init: [...]

Thanks for the feedback and I'm glad to hear this! Hopefully it stays like this! I would appreciate if you could check in again after lets say a week or so to see if it is still working a ok.

bhulsken, 2023-01-29:
> Thanks for the feedback and I'm glad to hear this! Hopefully it stays like this! I would appreciate if you could check in again after lets say a week or so to see if it is still working a ok.

Still going strong, 15 days uptime, without a single I2C failure (or other failure), most stable firmware so far for me!

ezwep, 2023-01-29:
Hi,

Since yesterday I connected a new CVE-S ECO SP. Unfortunately, the module does not work properly. I get this message:
init failed - please power cycle the itho unit -
I tested nrgitho-hw2-v2.4.4-beta6 and nrgitho-hw2-v2.4.4-beta4.

arjenhiemstra, 2023-01-29:
Doesn’t it work altogether or just after some period? 
have you power cycled the unit and have you switched the “i2c safe guard” function to “on” in the latest beta6?

ezwep, 2023-01-29:
> Doesn’t it work altogether or just after some period? have you power cycled the unit and have you switched the “i2c safe guard” function to “on” in the latest beta6?

I'll try to be a little more specific. ;-)
If I run on firmware 2.4.3, the control will stop working at some point. I have not (yet) saved a specific notification of this. So I tested the hw2-v2.4.4-beta4 and the hw2-v2.4.4-beta6 and these and in both cases I get the message
init failed - please power cycle the itho unit -
I also enabled the i2c safe guard option. But that has no effect.

I am now running on 2.4.3 again. If I have to provide specific logs or something, I am happy to do so. Or if I can test something with the beta versions as well.

arjenhiemstra, 2023-01-29:
Ok, please follow the following steps:
1. update to the latest beta release (beta6)
2. switch on the i2c safe guard function. 
3. Power cycle  the unit and report back what happened

ezwep, 2023-01-29:
I followed the steps.
Itho device type: Unknown device type
Itho I2C connection status: init failed - please power cycle the itho unit -
I can operate it, but I no longer see the temperature or humidity in the web interface.

PS.
When I restart the device it goes to maximum. Can you edit this somewhere? I have set the setting to Start/fallback speed 20

Thanks for your quick responses!

arjenhiemstra, 2023-01-29:
>I can operate it, but I no longer see the temperature or humidity in the web interface.

But you did see the temp/hum info on the web interface before?

Did you take the power of your itho off, waited for a bit and then turned the itho on again?

>When I restart the device it goes to maximum. Can you edit this somewhere? I have set the setting to Start/fallback speed 20

What settings did you set around the virtual remote? Can you share all the settings you made?

ezwep, 2023-01-29:
> > I can operate it, but I no longer see the temperature or humidity in the web interface.
> 
> But you did see the temp/hum info on the web interface before?
> Yes, in 2.4.3 I see these values

> Did you take the power of your itho off, waited for a bit and then turned the itho on again?
> Yes, I wait for a minute or so
> > When I restart the device it goes to maximum. Can you edit this somewhere? I have set the setting to Start/fallback speed 20
> 
> What settings did you set around the virtual remote? Can you share all the settings you made?
> I have 1 virtual remote: remtype: RRT CVE
I did some testing and it seems to be due to my MQTT settings, I will look into this further. But after I turn off MQTT it works fine after a reboot that the value is at 20.

arjenhiemstra, 2023-01-29:
> Still going strong, 15 days uptime, without a single I2C failure (or other failure), most stable firmware so far for me!

Very nice! Thanks again! I'm working on some last details, after that I'll release it as the next stable version.

ezwep, 2023-02-03:
>

Hi Arjen,

I was wondering if you can still help me with the problems I'm having? With firmware 2.4.3 I have problems (state hangs at 49, and the unit no longer responds to commands) and 2.4.4-beta6 does not work for me at all.

arjenhiemstra, 2023-02-03:
> Hi Arjen,
>
> I was wondering if you can still help me with the problems I'm having? With firmware 2.4.3 I have problems (state hangs at 49, and the unit no longer responds to commands) and 2.4.4-beta6 [...]

As I think this is not related to this described issue, could you create a separate issue for this?
And also, please provide the details request of this earlier question:
"Can you share all the settings you made?"

This can be as screenshot or config file which can be downloaded through the file editor on the debug page

PatrickV2, 2023-02-03:
@arjenhiemstra I have been running the latest beta for two weeks now and it has been stable sofar. In the previous releases it would fail within 1 or 2 days.

ezwep, 2023-02-03:
> As I think this is not related to this described issue, could you create a separate issue for this? And also, please provide the details request of this earlier question: "Can you share all the [...]

Ok, here I made a new request: https://github.com/arjenhiemstra/ithowifi/issues/150

Thanks

GelsemaP, 2023-02-07:
Just wanted to let you know that I experience the same error. After a couple of days I cannot control the itho fan anymore and I2C is disconnected. Powering on and off the fan completely fixes it. I've now updated the firmware to latest beta and enabled safeguard mode and restarted the whole device. Let's see if this also fixes my issue.

some details of my setup:
     16103 I: I2C init: Safe guard enabled
     15761 I: I2C init: QueryStatus
     15452 I: I2C init: QueryStatusFormat - items:12
     15280 I: I2C init: QueryDevicetype - fw:27 hw:27
     12359 I: Hostname: nrg-itho-741c
     11944 I: mDNS: started
     11730 I: Webserver: started
     11527 I: MQTT: connected, System config: 1
     11341 I: Setup: init of CC1101 RF module successful
     10604 I: WiFi: connection successful
      2757 E: Unable to set wifi disconnect
      1652 I: I2C sniffer capable hardware: yes
      1498 I: HW rev: 2, FW ver.: 2.4.4-beta6
      1257 I: System boot, last reset reason: POWERON_RESET

I have a Itho CVE-S ECO SE connected to Home Assistant.

mvaneijken, 2023-02-08:
Same issue here, running version 2.4.4-beta6: Itho I2C connection status: init failed - please power cycle the itho unit -

The issue occurred the first time on version 2.4.3, and did not work on any update since. Power cycles are not helping in this case.

2023-02-08 19:35:46 E: I2C init: QueryDevicetype - failed
2023-02-08 19:35:42 I: SHT3x sensor reset: executed
2023-02-08 11:29:52 E: Error: I2C bus could not be cleared!
2023-02-08 11:29:50 W: Warning: I2C timeout, trying I2C bus reset...
2023-02-04 16:52:06 E: I2C init: QueryDevicetype - failed
2023-02-04 16:52:05 I: SHT3x sensor reset: executed
     15213 E: I2C init: QueryDevicetype - failed
      9960 I: Hostname: nrg-itho-6804
      9789 I: mDNS: started
      9630 I: Webserver: started
      9419 I: MQTT: connected, System config: 1
      8266 I: WiFi: connection successful
      2773 E: Unable to set wifi disconnect
      1858 I: I2C sniffer capable hardware: no
      1634 I: HW rev: 2, FW ver.: 2.4.4-beta6
      1333 I: System boot, last reset reason: TG0WDT_SYS_RESET
2023-01-29 19:55:29 E: I2C init: QueryDevicetype - failed
2023-01-29 19:55:27 E: Error: I2C bus could not be cleared!
2023-01-29 19:55:24 W: Warning: I2C timeout, trying I2C bus reset...
2023-01-29 19:55:24 I: Hostname: nrg-itho-6804
2023-01-29 19:55:24 I: mDNS: started
2023-01-29 19:55:23 I: Webserver: started
2023-01-29 19:55:23 I: MQTT: connected, System config: 1
2023-01-29 19:55:22 I: WiFi: connection successful
2023-01-29 19:55:17 E: Unable to set wifi disconnect
2023-01-29 19:55:16 I: I2C sniffer capable hardware: no
2023-01-29 19:55:15 I: HW rev: NON-CVE 2, FW ver.: 2.4.4-beta6
2023-01-29 19:55:15 I: System boot, last reset reason: SDIO_RESET
2023-01-29 21:12:53 I: Reboot requested
2023-01-19 08:14:30 E: Error: I2C bus could not be cleared!
2023-01-19 08:14:27 W: Warning: I2C timeout, trying I2C bus reset...
     15204 E: I2C init: QueryDevicetype - failed
     10180 I: Hostname: nrg-itho-6804
      9772 I: mDNS: started
      9553 I: Webserver: started
      9347 I: MQTT: connected, System config: 1
      8580 I: WiFi: connection successful
      2770 E: Unable to set wifi disconnect
      1695 I: I2C sniffer capable hardware: no
      1507 I: HW rev: 2, FW ver.: 2.4.4-beta6
      1272 I: System boot, last reset reason: TG0WDT_SYS_RESET
2023-01-13 19:17:48 E: I2C init: QueryDevicetype - failed
2023-01-13 19:17:44 I: SHT3x sensor reset: executed
2023-01-13 20:22:30 E: I2C init: QueryDevicetype - failed
2023-01-13 20:22:25 I: Hostname: nrg-itho-6804
2023-01-13 20:22:25 I: mDNS: started
2023-01-13 20:22:24 I: Webserver: started
2023-01-13 20:22:24 I: MQTT: connected, System config: 1
2023-01-13 20:22:23 I: WiFi: connection successful
2023-01-13 20:22:18 E: Unable to set wifi disconnect
2023-01-13 20:22:17 I: I2C sniffer capable hardware: no
2023-01-13 20:22:17 I: HW rev: 2, FW ver.: 2.4.4-beta6
2023-01-13 20:22:16 I: System boot, last reset reason: SDIO_RESET
2023-01-13 19:16:10 I: Reboot requested
2023-01-13 17:52:10 E: I2C init: QueryDevicetype - failed
2023-01-13 17:52:04 I: Hostname: nrg-itho-6804
2023-01-13 17:52:04 I: mDNS: started
2023-01-13 17:52:04 I: Webserver: started
2023-01-13 17:52:04 I: MQTT: connected, System config: 1
2023-01-13 17:52:03 I: WiFi: connection successful
2023-01-13 17:51:57 E: Unable to set wifi disconnect
2023-01-13 17:51:56 I: I2C sniffer capable hardware: no
2023-01-13 17:51:56 I: HW rev: 2, FW ver.: 2.4.4-beta6
2023-01-13 17:51:56 I: System boot, last reset reason: SDIO_RESET
2023-01-05 13:25:28 E: I2C init: QueryDevicetype - failed
2023-01-05 13:25:26 I: SHT3x sensor reset: executed
[... 75 lines cut ...]

From the I2C debug menu:
<html>
<body>

origin | start | duration | error | final
-- | -- | -- | -- | --
QUERY_DEVICE_TYPE | 345200832 | 4528 | I2C_ERROR_SDA_LOW | I2C_NOK
QUERY_DEVICE_TYPE | 345195829 | 3 | I2C_ERROR_CLEARED_OK | I2C_NOK
QUERY_DEVICE_TYPE | 345190826 | 2 |   | I2C_OK
QUERY_DEVICE_TYPE | 345185823 | 2 |   | I2C_OK
QUERY_DEVICE_TYPE | 345180820 | 2 |   | I2C_OK
QUERY_DEVICE_TYPE | 345175817 | 2 |   | I2C_OK
QUERY_DEVICE_TYPE | 345170939 | 2 |   | I2C_OK
QUERY_DEVICE_TYPE | 345170811 | 2 |   | I2C_OK
QUERY_DEVICE_TYPE | 345165807 | 2 |   | I2C_OK
QUERY_DEVICE_TYPE | 345160804 | 2 |   | I2C_OK
QUERY_DEVICE_TYPE | 345155801 | 2 |   | I2C_OK
QUERY_DEVICE_TYPE | 345150779 | 2 |   | I2C_OK
QUERY_DEVICE_TYPE | 345145776 | 2 |   | I2C_OK
QUERY_DEVICE_TYPE | 345140773 | 2 |   | I2C_OK
QUERY_DEVICE_TYPE | 345135770 | 2 |   | I2C_OK

</body>
</html>

arjenhiemstra, 2023-02-08:
@mvaneijken 

Also for you, please follow the instruction posted here earlier:

```
1. update to the latest beta release (beta6)
2. switch on the i2c safe guard function.
3. Power cycle the unit and monitor what happens
```

[10 more replies]

closed 2023-04-29 by stale[bot]