arjenhiemstra/ithowifi #112: Ventilation setpoint (%) stays at 99%

sciurius, 2022-10-08:
**Describe the bug**
Speed setting accepts values between 0 and 254.
When set to 254, Ventilation setpoint (%) stays at 99%, not 100%.

Could this be an indication that acceptable values should be between 0 and **255**?

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to main screen
2. Slide the slider to max
3. Use MQTT inspector
4. `state` is 254, `Ventilation setpoint (%)` is 99%.

**Expected behaviour**
I'd expect the ventilation setpoint to be 100%.

**Device information**
- Firmware version [e.g. 2.2.1] 2.4.2
- Hardware revision (e.g. non-cve 1.1) cve 2
- Type/model number Itho (on sticker outside the box) CVE-S fw:27 hw:27
- CC1101 RF module enabled [ yes, no ] yes

arjenhiemstra, 2022-10-08:
I need to go back and check the first reverse engineering logs for this but I believe to remember that 255 causes a 'rollover' to 0 on the itho. So 99% seems to be the highest value reachable.

sciurius, 2022-10-09:
FYI, when on 'high', or 'auto' + humidity, the ventilation setpoint is 100%.

sciurius, 2022-10-09:
Overigens probeer ik de software te bouwen (met pio) maar ik krijg compileerfouten (op elke build):
````
Compiling .pio/build/beta_cve/src/Config.cpp.o
src/Config.cpp: In function 'bool loadWifiConfig()':
src/Config.cpp:5:8: error: 'ACTIVE_FS' was not declared in this scope
   if (!ACTIVE_FS.exists("/wifi.json"))
        ^~~~~~~~~
src/Config.cpp:7:5: error: 'D_LOG' was not declared in this scope
     D_LOG("Setup: writing initial wifi config\n");
     ^~~~~
...
````
Mis ik nog iets?

arjenhiemstra, 2022-10-10:
Lijkt erop dat je ergens een include mist of iets in platformio.ini niet helemaal lekker staat. 
In hardware.h wordt ACTIVE_FS gedefinieerd, D_LOG in Dbglog.h.

sciurius, 2022-10-10:
Het lijkt erop dat ik `sys/epoll.h` mis. Waar zit die bij jou?

arjenhiemstra, 2022-10-10:
Die is onderdeel van libc geloof ik. Ik denk daarmee dat de c-compiler niet goed geinstalleerd is.

sciurius, 2022-10-10:
Ik verkeer in de veronderstelling dat pio dat allemaal keurig regelt, ik heb met esp8266 projekten nooit problemen gehad.
Ik heb ~/.platformio etc compleet weggehaald en alles opnieuw geïnstaleerd en opgezet. Precies hetzelfde probleem. 
Er is geen epoll.h in `!/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/sys`.

arjenhiemstra, 2022-10-10:
Dat veronderstel ik eigenlijk ook :) 
Geen idee nog wat het kan zijn, ik heb overigens ook geen epoll.h in ~/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/sys
Ik zal eens een clean vm opzetten en vanaf scratch proberen platformio op te zetten en kijken of ik hetzelfde krijg.

Welk OS gebruik je btw?

sciurius, 2022-10-10:
Interessant. Ik gebruk Fedora 35.

arjenhiemstra, 2022-10-12:
> 

Gevonden. Er stond op twee plekken #include "config.h" ipv #include "Config.h"
Met de laatste commit https://github.com/arjenhiemstra/ithowifi/commit/57929f09bc7895eccbd6aa7009aff43c927756bd zou het moeten werken.

sciurius, 2022-10-12:
Confirmed. Dank.

closed 2022-11-11 by arjenhiemstra