arjenhiemstra/ithowifi #113: Lost WiFi

blokl, 2022-10-19:
Hallo Arjen,

Even ter info.
Ik wilde een vast ip adres toekennen. Dit heb ik ook gedaan.
Na een reboot/reset via het menu, kom ik niet meer op de webpage.
Ik gebruik volgens mij versie 2.4 
De enige manier is waarschijnlijk om hardware matig  te resetten? Is er nog een andere manier? Ik kom er niet zo makkelijk bij.

Groet Leo Blok

arjenhiemstra, 2022-10-19:
Check even of de access point niet nog actief is. Als er iets mis gaat met het opzetten van de WiFi verbinding start het access point weer op.

blokl, 2022-10-19:
Ik zie ook geen access point. Ik heb al verschillende keren opnieuw gestart

TD-er, 2022-10-19:
Misschien een typo gemaakt in het IP-adres?

blokl, 2022-10-19:
> Vreemd, dan zou er een verbinding met je WiFi netwerk moeten zijn. Ik zou een netwerk scanner gebruiken om te kijken of ik het apparaat kan vinden. Anders inderdaad een config reset helaas

Er is een verbinding met mijn WiFi  netwerk . Ik zie hem in mijn router monitor, alleen zonder ip adres.

Name |iIp | mac addon | besturingssysteem | Ssid | Mac router | Channel | Type | role | ipv6 | signal | speed

-- | 0.0.0.0 | c0:49:ef:32:d1:10 | NOFP | Guard-Home | 34:fc:b9:cf:df:10 | 7 | G | Guard-Home | -- | 45 | 24
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --

arjenhiemstra, 2022-10-19:
Typo in het IP adres, security instellingen van je netwerk zijn de dingen die ik al eens voorbij heb zien komen, Iets anders zou ik zo even niet weten.
Welke firmware versie staat op de add-on?

blokl, 2022-10-19:
De laatste firmware. Ik ga hem wel resetten dan zal het wel goed komen. 
bedankt voor je ondersteuning

arjenhiemstra, 2022-10-21:
Is het nog gelukt?

blokl, 2022-10-21:
> Is het nog gelukt?

Hoi Arjen,

Ja, vanmiddag een reset gegeven. En opnieuw verbonden.
Ik had bij de WIFI setup alleen de dhcp uitgezet en het laatste ip gewijzigd. Van 192.168.2.80 naar 192.168.2.82.
Ik nam aan dat hij de rest zou onthouden.
Daarna via het menu Reset en dan restart device. Ik bedoel misschien moet je alle IP gegevens wel opnieuw handmatig invoeren? 

Bedankt voor je hulp,

Leo

closed 2022-10-31 by arjenhiemstra