arjenhiemstra/ithowifi #114: RFE: ability to set itho settings via API

rplevka, 2022-10-19:
**Describe the bug**
It would be useful to be able to set the itho settings (Manual control, bypass valve position, target rpm, reset filter day counter, ...) using API.

these values are already published for reading, but in order to set any of them, I still need to manually set them via UI - this prevents me from using this in my automation, which makes the use of the addon quite limiting.

**Expected behaviour**
since some might consider these settings - "advanced", there could be some opt-in setting (with an appropriate warning message) to enable this api endpoint

**Device information**
- Firmware version [e.g. 2.4.2]
- Hardware revision (e.g. cve 2)
- Type/model number Itho (ECO HRU350)
- CC1101 RF module enabled [ yes ]

arjenhiemstra, 2022-11-08:
I understand the request and you are not the first to ask it. There is a certain danger and difficulty with this.
first: every itho device and even every firmware has a different set of settings
second: errors are easily made and difficult to revert, even with the current web based solution I sometimes get support requests to help users to restore settings...

So an opt-in would be minimum, but I'm still not sure if this at all is a wise move

rplevka, 2022-11-10:
perhaps a flag in the source code that one would have to uncomment and build the firmware themselves?

DutchMikeDutch, 2022-11-16:
I fully understand people can wrongly configure the devices nevertheless to overrule the device in certain situation. I rather have the ability to set settings via an automation on Home Assistant then manually going to the Website and setting it to Manual mode and execute. building a mouseclick script is a bit overkill no?

YorickPeterse, 2023-01-12:
I'm currently evaluating the WiFi add-on, though for a somewhat particular reason: I have an HRU 350 with the Demandflow/Controlflow system attached. This seems to result in it being impossible to reset the filter status, as also discussed on https://gathering.tweakers.net/forum/list_messages/2005642.

The workaround is to buy the ridiculously expensive service interface (€300), then use the Itho service tool to reset it (which in turn seems to only work on Windows. Unfortunately, it seems this interface isn't sold directly to consumers as far as I've been able to find, meaning you need to buy it through your local installation company; assuming they even want to do that.

If the WiFi add-on were to support this, it would prove a much more affordable (and probably easier to use) alternative to work around this problem :smiley:

arjenhiemstra, 2023-01-12:
I have an official service tool for my R&D work. Also the filter reset code is already present in the add-on firmware (although I have not been able to test it because I do not have access to a HRU350)
How you could test it:
first have a recent firmware version where you can select the virtual remote type (ie. latest beta fw version will do)
1- connect the add-on to the HRU
2- configure the first virtual remote to be of the type: RFT AUTO
3- unplug the power from the HRU, wait a bit and power it again
4- within 2 mins. on the virtual remotes page click the join button of the first virtual remote
5- confirm you are able to control the speed of the HRU by clicking ie. high and low buttons on the same page
6- go to the debug page and click the button "filter reset", that should do the trick

YorickPeterse, 2023-01-12:
@arjenhiemstra Thanks for the details. I'll likely buy the add-on this week then, and report back once I've tested it :smiley:

gns1l, 2023-01-13:
Nov last year I tried to reset the filter warning with my remote without success. Then I tried the filter reset command on the Itho-wifi debug page. Also without success. In the end I just changed the filter counter to zero hours. That worked.
vremote is functional but I'm not sure which type.

rplevka, 2023-01-13:
I can confirm i managed to successfully reset the filter "alarm" on my HRU350 using the addon. resetting via the Remote control didn't work for me either

YorickPeterse, 2023-01-14:
I received the module today and managed to get it up and running. As I have a Demandflow/Qualityflow system, the use of an RFT AUTO virtual remote didn't seem to result in it being able to control anything; instead I had to set up the virtual remote as a RFT DF/QF remote. I paired it with the DF/QF control panel, as it seems the HRU ECO 350 doesn't respond to pairing when the DF/QF control panel is hooked up.

In this setup the "Filter reset" button doesn't seem to do anything. Oddly enough, the only filter related setting on the Itho settings page is "Filter factor", set to a value of 5. The reported Itho firmware version is 21.

Should I in this case set up an extra virtual remote and pair that directly with the HRU?

YorickPeterse, 2023-01-14:
I just tried to hook the module up to the HRU's COM port. This port doesn't provide any power so I also had to plug in a USB cable. In this scenario the module is seemingly unable to detect what it's connected to, and thus isn't able to retrieve any settings.

If I had to guess, I suspect the DF/QF control panel simply doesn't read/write the filter related settings, hence the official remotes don't support resetting the filter either.

YorickPeterse, 2023-01-14:
Updating to the beta firmware doesn't seem to have any impact on the filter status being reported. Is there anything I can do to debug this (or any information I can provide to help with that)?

YorickPeterse, 2023-01-14:
After much fiddling I managed to pair a bunch of RFT Auto virtual remotes. It seems the official documentation isn't entirely correct, or maybe my setup was weird: I had to pair my physical DF/QF remote 2-3 times, then press the bottom left/right (previous and next) button once, then all the lights blink and you can start assigning rooms. From that point on you follow the manual, selecting rooms and pairing the next virtual remote. Getting this right took several tries, but now it works.

Unfortunately, the "Reset filter" button still doesn't do anything in this case, even though remote 0 is an RFT Auto one. This further suggests the DF/QF control panel just doesn't forward the signal to the HRU.

rplevka, 2023-01-15:
@YorickPeterse try navigating to `Itho Settings` -> `Retrieve Settings`, then wait till all the values are read and populated.
then try to set Option no. 50:
`Filter counter (Hours)` to `0`. Good luck!

YorickPeterse, 2023-01-15:
@rplevka Unfortunately, I don't have such a setting, and option 50 is "Sample measurement interval (sec)" in my case :confused:

gns1l, 2023-01-15:
@YorickPeterse: Is the module connected to the HRU 350?

YorickPeterse, 2023-01-15:
@gns1l As mentioned in https://github.com/arjenhiemstra/ithowifi/issues/114#issuecomment-1382746884 and https://github.com/arjenhiemstra/ithowifi/issues/114#issuecomment-1382752902, my WiFi module is hooked up to the DemandFlow/QualityFlow control unit, not the HRU directly. Hooking it up directly doesn't seem to work either as no I2C connection is detected, most likely because the DF/QF unit takes over control.

rplevka, 2023-01-15:
I have 0 experience with DF/QF, but would it be a bad idea to temporarily disconnect it from the main unit?
Btw, what do the `device type` and `fw version` at the `Itho Settings` say?
Mine are:
```
Itho device type: HRU 350
Itho fw version: 3
```

gns1l, 2023-01-15:
How are the DF/QF unit and the HRU 350 connected? 
Is the HRU 350 a special version?

arjenhiemstra, 2023-01-15:
> @gns1l As mentioned in [#114 (comment)](https://github.com/arjenhiemstra/ithowifi/issues/114#issuecomment-1382746884) and [#114 (comment)](https://github.com/arjenhiemstra/ithowifi/issues/114#issuecomment-1382752902), my WiFi module is hooked up to the DemandFlow/QualityFlow control unit, not the HRU directly. Hooking it up directly doesn't seem to work either as no I2C connection is detected, most likely because the DF/QF unit takes over control.

The filter can only be reset by connecting the add-on to the HRU directly.
The HRU should provide power to the add-on, it doesn’t matter whether there is a DF/QF connected through RF.
If there is no power to the add-on from the HRU, something is wrong with your HRU I’m afraid.

YorickPeterse, 2023-01-15:
@arjenhiemstra 

> The filter can only be reset by connecting the add-on to the HRU directly.

I figured as much, but it's fine: I've just set a calendar reminder instead. Instead I'll be using the module to better control the WiFi (i.e. by controlling rooms better using the virtual remotes). Now I just need to settle on a Raspberry Pi or some other small computer for running my own software part of this :smile: 

@rplevka The DF/QF unit is pretty crucial for the ventilation system we have, as without it there's no way to, well, ventilate the rooms. Since only the filter reset option is missing I'm just going to leave it as-is.

@gns1l The HRU ECO 350 is just a WTW ventilation unit. The DF/QF control unit is connected to it using a wireless signal, probably the same frequency/protocol as the remotes. You can find some more information on it [here](https://www.ithodaalderop.nl/nl-NL/consument/product/536-0405).

arjenhiemstra, 2023-01-15:
> I figured as much, but it's fine: I've just set a calendar reminder instead. Instead I'll be using the module to better control the WiFi (i.e. by controlling rooms better using the virtual remotes). [...]

Still strange that the HRU does not power the add-on, could it be that the daughter board that houses the COM port isn't properly connected internally?
Are you using the virtual remote control option on the DF/QF for per room control?

YorickPeterse, 2023-01-15:
@arjenhiemstra 

> Still strange that the HRU does not power the add-on, could it be that the daughter board that houses the COM port isn't properly connected internally?

Honestly it wouldn't surprise me, because the setup was a bit of a mess. The previous owners of this house were the first owners, and based on the state I found the ventilation in they hadn't changed anything (i.e. cables in the plenum were a mess, ventilation was way too aggressive/loud, etc).

> Are you using the virtual remote control option on the DF/QF for per room control?

That's my plan. I currently have a virtual remote for each room set up, and already started playing around with it to control e.g. the laundry room. I'm still trying to work out some details, such as how I'm going to turn the fans on automatically when a bathroom/toilet is used without forking out more money for PIR sensors and SCBs to hook them up to. But at least so far it already saved me €1000+ by not having to buy the physical remotes :smiley:

arjenhiemstra, 2023-01-15:
>I currently have a virtual remote for each room set up, and already started playing around with it to control e.g. the laundry room.

Is that working as expected? I specifically created the multiple virtual remote option and selectable remote type function for the situation you describe but until now never received user feedback on it :)

YorickPeterse, 2023-01-15:
@arjenhiemstra It works perfectly :smiley:

YorickPeterse, 2023-01-16:
Over night I left my laptop running to measure the sensor details in our house. The goal here was to see how accurate the measurements are, and what impact our current ventilation settings have. 

Rather than messing with the Eco and Comfort settings, I'm simply using the Eco settings as _the_ settings. This way you don't have to remember what to set the system back to. The default of 1200 ppm is far too high. Based on some threads on tweakers.net, the DF/QF system isn't the most accurate in measuring CO2 values. With that in mind I have the thresholds set as follows:

All rooms have a threshold of 900 ppm, but our bedroom has a threshold of 800 ppm. The idea here is to compensate for potential measurement errors, and to ensure the bedroom is ventilated a bit faster. I also had a few rooms set up with a minimal flow of 5 m3/h.

Measuring this overnight, I made the following observations:

- Ventilation seems to kick in at a low setting when a threshold is reached. This is different from a Climarad system I had in a previous house, which seemed to kick in earlier and switch to a higher setting when reaching the CO2 threshold.
- There's no time based adjustments or anything. This means that if the CO2 concentration stays at the threshold for X hours, the ventilation keeps running at the same speed for X hours; instead of saying "OK that's enough, let's crank it up a bit"
- As a result the CO2 concentrations in the bedroom kept hovering between 800 and 900 ppm. I have since reduced the threshold to 700 ppm and will measure this again over night
- Weirdly enough it seems the system _is_ more aggressive in our office rooms, kicking in seemingly when necessary and shutting down rather quickly again. This in spite of the bedroom having a greater max flow value.
- In our kitchen we have two exhausts, one for the kitchen and one for the extractor. The extractor had a minimum flow of 5 m3/h, but the kitchen had this turned off. For some reason the CO2 concentration in the kitchen was consistently reported to be around 200 ppm _higher_ compared to the extractor, even though they're more or less right next to each other. I have since given the kitchen a minimum flow of 5 m3/h to see if that has any impact
- Even with the system not being particularly accurate, you can tell when somebody moves across rooms: my partner woke up around 07:00 and spent about 30 minutes in the bathroom, and this is reflected in the CO2 readings for our bathroom and bedroom: the bedroom CO2 levels went down by about 20%, while increasing by about 20% in the bathroom.
- In our laundry room the CO2 levels are a bit spiky over night, even though nobody was there, i.e. at 04:00 there was a 100 ppm spike. Perhaps it's haunted.
- RH/humidity values are consistently reported to be 15-20% higher than the actual values (i.e. the system reports 70% instead of 55%). This seems to be reflected in the settings, as the various RH thresholds are around 90%, which matches the usual "real world" thresholds of 70-80% (i.e. my old Climarad bathroom fans would kick in around 65-70% RH).

Which brings me to the following:

Being able to change the settings through the API (at least the flow related settings) would offer greater control over the ventilation. For example, I was thinking of doing the following:

1. When CO2 levels in the bathroom and toilet increase by >=25%, start ventilating at the "Low" setting. Higher levels in these rooms likely indicate the presence of a person about to stink up the place, so ventilating earlier is better than ventilating too late. 
2. Adjust the minimum flow settings twice a day: in the morning the minimum flow is turned off for rooms only used at night, while it's turned on for rooms used during the day. In the evening this is inverted. The idea is that there's no point in ventilating unused rooms while other rooms on the same floor (i.e. our offices) _are_ used. Simply setting minimal ventilation for all rooms may result in fresh air not reaching the ground floor as well. 
4. Scheduled ventilation, i.e. ventilating the bedroom at 21:00 and the office rooms at 09:00, or the kitchen between 18:00 and 20:00. Not sure yet if this is beneficial, but it's something I want to experiment with.

For this I'd at least need the ability to change at least the following settings through the API/MQTT:

- Minimum flow
- Maximum flow
- RH related values
- CO2 concentration values

YorickPeterse, 2023-01-18:
I've been toying a bit more with the settings, as the DF/QF system is frankly rather dumb. For example, last night I went to the toilet for a few minutes, and the system decided to ventilate it for the next hour at a high speed, because somehow it derived the CO2 to be at 1500 ppm, even though the CO2 in the plenum was only around 700 ppm. And earlier today it decided to ventilate the bathroom at 100% speed, even though nobody has been there for hours.

Based on the CO2 levels reported back (e.g. I've seen values such as -600) and the odd behaviour of the system, I suspect the Demandflow system is either just really stupid, or there are firmware bugs I'm running into.

Fortunately I found a workaround: using the web interface I can flip the system into manual mode. When done it closes all the valves and doesn't do anything automatically from that point on. Controlling the valves is then done using the settings such as "Flap position bathroom 1 with manual control (step)". These settings have values in the range of 0 to 5000, and specify how far the valves should be opened. These values don't do anything unless the system is in manual mode.

By changing these settings you get fine-grained control over how much air is exhausted from the rooms into the plenum.

The setting "Ventilation requirement for exhaust air fan in manual mode (%)" is then used to control the speed of the HRU ECO 350 attached to the DF system.

With the two combined you should be able to replicate the DF system's behaviour, but implement it to better suit your needs (i.e. having it ventilate the right rooms, instead of a completely unrelated one).

Air intake is done automatically with a speed relative to the exhaust speed. While there exists a setting "Required supply air fan with manual control (%)", changing its value does nothing, and there's never a speed reported for it. This is fine though, as the intake speed is already tuned in my case (took me long enough).

One thing to note is that in manual mode the DF system doesn't respond to any of the RFT Auto remotes, though I have yet to test if it responds to my physical DF/QF remote.

Power cycling the DF system results in manual mode being reset, and the flap settings are reset to 0. I'm guessing these aren't persisted in any EEPROMs, so changing them frequently shouldn't pose a problem. 

Seeing as how the web interface uses web sockets to change settings, I _think_ I may have a temporary solution, but having these settings be available in the API/via MQTT would be really nice :smiley:

rplevka, 2023-01-20:
The manual function is what i'm actually after as I'm looking to controlling HRU350's fan speeds and a bypass/anti-freeze valve manually (sometimes).
In the future, i was considering to craft/purchase some servo-driven duct valves to introduce "zoning".
Since demand-flow seems to do just that, and you are already experimenting with it - there is one thing i'm curious about whil operating in manual mode - does the system allow you to close all the valves at the same time? if so, that might be quite dangerous as it might (probably) damage the motor.
Would you mind trying that out? :rat:

YorickPeterse, 2023-01-20:
@rplevka 

I think you can't close all the valves at exactly the same time, as the DemandFlow system doesn't write back to the I2C slave until a valve reaches its desired value (at least that's what I think is happening). You can also see this in practise: when you flip the system into manual mode, it closes valves one by one instead of all at the same time.

I don't think you'd want this anyway, as this would cut off all ventilation. You'd also have to make sure the exhaust fan isn't running, or it will try to exhaust a more or less airtight plenum, and I'm not sure what the impact of that is.

philipkocanda, 2023-01-23:
Manual control is something that I'd also like to use on my HRU ECO (firmware: v12, hardware: v3) as the built-in algorithm for opening/closing the bypass is just not that great (a good attempt by the Itho folks though given the limited information it has to work with). And I like being able to turn off the supply fan when the neighbors are having a nice BBQ party (🔥), so we still have some exhaust ventilation in the bathroom when needed.

My approach to this used to be a running `rustyx/itho-esp` on a custom ESP32, and passing it messages generated by my own [custom Python library](https://github.com/philipkocanda/ithopy) over MQTT. But I prefer using the ithowifi hardware/software much more for its community and the reliability of having a dedicated piece of hardware.

I'm thinking of just having an option of passing raw messages over MQTT, which given the built-in checksum validation should be a high enough threshold to stop those who don't know what they're doing from clicking some random buttons and frying their expensive machinery.

arjenhiemstra, 2023-01-24:
> Manual control is something that I'd also like to use on my HRU ECO (firmware: v12, hardware: v3) as the built-in algorithm for opening/closing the bypass is just not that great (a good attempt by the Itho folks though given the limited information it has to work with). And I like being able to turn off the supply fan when the neighbors are having a nice BBQ party (🔥), so we still have some exhaust ventilation in the bathroom when needed.
> 
> My approach to this used to be a running `rustyx/itho-esp` on a custom ESP32, and passing it messages generated by my own [custom Python library](https://github.com/philipkocanda/ithopy) over MQTT. But I prefer using the ithowifi hardware/software much more for its community and the reliability of having a single integrated piece of hardware rather than a bunch of dangling wires in a box.
> 
> I'm thinking of just having an option of passing raw messages over MQTT, which given the built-in checksum validation should be a high enough threshold to stop those who don't know what they're doing from clicking some random buttons and frying their expensive machinery.

Just FYI, rustyx' tho-esp firmware runs without changes on the ithowifi add-on (non-cve variant), no need for dangling wires :)

yorickpeterse, 2023-03-25:
Re the above bot: This PR is very much still active, I just haven't had the time yet to add the setting to enable/disable the API.

yorickpeterse, 2023-05-24:
Issue is very much still active, just pending review on https://github.com/arjenhiemstra/ithowifi/pull/144.

commit 241d96d3 (2023-10-31): Add API support for managing settings

This adds the ability to get and set settings using the API. Settings
are obtained using the "getsetting" parameter, while updating settings
is done using "setsetting" (for the index) and "value" (for the value).

All settings that are supported can be changed through the API, instead
of only allowing a limited subset. Due to the different settings
supported across devices and firmware versions maintaining an allow/deny
list isn't practical. It's also not clear what settings should or
shouldn't be allowed in the first place.

This fixes #114
closed 2023-10-31 by arjenhiemstra