arjenhiemstra/ithowifi #115: Unable to control fan speed from main screen/mqtt

Jasperwolsing, 2022-11-02:
**Describe the bug**
I added the RAF module and connected it as virtual remote. On the virtual remote page I can set fanspeed an the ITHO Will charge speed.

when I use the buttons at the main screen, ontging happens.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to 'main screen '
2. Click on 'high’
3. Nottingham happens4. 

**Expected behaviour**
Fan Goes to 100

**Device information**
- Firmware version 2.4.3
- Hardware revision non-cve 1.1
- Type/model number Itho ECO HRU 350
- CC1101 RF module enabled Yes

**Debug logging**
Please provide debug logging from the debug page if possible:
Notting appears with level 3 debug

**Desktop (please complete the following information):**
 - OS: ios, Windows and mqtt

arjenhiemstra, 2022-11-02:
Thanks for spotting this. I also already know why this is. The ability to change the type of virtual remote is new since version 2.4. The buttons on the virtual remote page are linked to the type correctly but the ones on the main page are not. These should also reflect the defined remote type on the virtual remote page.
I will update this in a future release.

Jasperwolsing, 2022-11-03:
Thanks, am I correct that there is one virtual remote type that should work for now?

Jasperwolsing, 2022-11-06:
I  also tried downgrading to 2.3 and 2.3.5 but unable to get it working in any way.

arjenhiemstra, 2022-11-06:
The RFT CVE remote type should work with the front end buttons.
Switching remote type unfortunately needs the remote to be repaired to the itho.
Best to install 2.4.3, set the virtual remote to rft cve and follow the leave/join proces of the itho manual

HAuser2019, 2022-11-10:
Same probleem hier. 2.4.3 Home Assistant stuurt een speed change naar MQTT van ITHO naar 100%, er gebeurd niets. Als je in de webinterface van de ITHO kijkt staat deze wel op 100% high maar hij gaat niet daadwerkelijk naar hoog. Als ik de NRG webinterface toggled van medium naar High gaat het ding wel naar high. Klik ik weer op medium gaat hij naar medium maar de bar blijft 100% max aan geven. Misschien omdat home assistant MAX blijf vragen aan de API? Er lijkt iets niet te werken in 2.4.3 firmware. (Ben blij dat Home Assistant en MQTT iniedergeval lijken te praten, had me rot gezocht en re-installs gedaan etc.) Anyway issue dus dat de boel niet (meer) reageert op MQTT commando's eigenlijk, de webinterface wel mee gaat, maar de ITHO niet. Het gekke was dat met de eerste keer configuren alles werkte, het probleem onstond pas na een tijdje en ik krijg het er nu ook niet meer uit met reboots enzo.

HAuser2019, 2022-11-11:
A meh, communication went out again last night after my previous update.
Reboot again this morning to get communication with HA.

arjenhiemstra, 2022-11-11:
> Thanks, am I correct that there is one virtual remote type that should work for now?

Please try https://github.com/arjenhiemstra/ithowifi/releases/tag/Version-2.4.4-beta1
the buttons should now change with the set vremote type.

arjenhiemstra, 2022-11-11:
> A meh, communication went out again last night after my previous update. Reboot again this morning to get communication with HA.

Seems not to be related to this issue. It could well be this one: https://github.com/arjenhiemstra/ithowifi/issues/111

HAuser2019, 2022-11-17:
Experienced bug #111 and/or #115
Just tried 2.4.4-beta1. Initally it works.
-The slider in the webinterface does not move when using low, medium, high. Also not when sending commands via HA.
-The RFT AUTO does virtual remote does not work. Using the first RFT.
-I enable the I2C-bug checker.
-I will try to help in a limited way, the wife if allready upset.... what would be the safest firmware to revert to? (ITHO CVE-S FW27  HW2 with CC1101 RF)

Jasperwolsing, 2022-11-17:
Hi Arjen,

I did install the beta after your post and buttons on the home page works. I wast able to control trough HA, but I did not had the time to figure out if that’s something in my own setup. I will figure that out.

HAuser2019, 2022-11-17:
Something happened so posting here. I do NOT have an I2C timeout/error. 
I was wondering why my HA automation didn't lower the fanspeed after 10min done showering. So I pushed the 2 button on my RFT Spider base (which does not connect to the addon, but is connected to the ITHO it's self) and this rendered the MQTT connection dead (see picture). But read along to the last alinea, I might be wrong here.
When looking at the Addon webinterface, that still works, but also the controls there in the webinterface are not responding. So effectively I cannot shut down the fan now still running high speed.

By the way: the SpiderBase and ITHO and Addon are apparently not completly happy. Because normally I could control the spider base and itho-addo with firmware 2.3.5 but with 2.4.4.1 the Spider base is fixed on the auto setting. I did push the 2 button, but actually it does nothing, but it does seems to crash the addon control-part/MQTT...?!
-While typing the fan lower it's self automatically, due the humidity is dissovled and the system is stuck in auto-mode....

-Restarting the addon webinterface restores webinterface and HA controls again!
-But my humidity graph in HA stays flat now, restarting the integration and broker does not work, seems the sensor input from the itho to the addo stays down even after reboot the addon.
-Tried pushing button 2 again in the RFT-spiderbase to see if i'm correct, unfortunaly this time does not crash the addon! Might that this story of mine is a whole ghost...and i'm seeing things that are not there...and the initial error is that my automation didn't run anymore is because of the humidity sensor went down in the firsplace, and dragged all controls non responsive with it? And not cause by me pushing button 2....
So maybe something is happening there in that area...

[image]

Jasperwolsing, 2022-12-18:
Hi @arjenhiemstra,

I did some test on MQTT but can't get it working. All information is updated correctly but setting low,mid,high doesnt.
[image]

Jasperwolsing, 2022-12-29:
I fixed it using this guide:
https://github.com/arjenhiemstra/ithowifi/wiki/Home-Assistant---HRU350-integration

perfect, thanks!

closed 2022-12-29 by Jasperwolsing