arjenhiemstra/ithowifi #116: Customize button actions of remote

roblomq, 2022-11-06:
Is it possible to customize the action of the buttons?
For example I have the RFT AUTO C02 (536-0150) remote but although I've added a custom sensor the "A" button is doing not what I want. SO is it possible to set the speed to 0 when I press the "A" button on the remote?

arjenhiemstra, 2022-11-11:
The A button press (for a remote join to the add-on using the cc1101 module) currently translates in code to a speed command with value of the "medium" setting of "Speed settings (CVE only) (0-254)" found on the System settings page. 
So if you change this setting to a low value it should work.

roblomq, 2022-11-11:
It's not clear what you mean exactly. Is it correct that you mean that when I press the A button the fan goes to a medium speed?
But a speed of 0 is not equal to a low speed :-) my wife wants the fan off when she takes a bath.

arjenhiemstra, 2022-11-11:
You have your remote joined to the add-on and not to the itho directly?
If yes, then I believe settings this setting (medium):
[image]
To 0 instead of the 120 in the example, will enable you to regulate the itho to 0 (depending on your itho model you might also need to change a few settings in the itho firmware for this to work)

roblomq, 2022-11-15:
Liked the idea, tested it and can confirm it works. For me this issue can be closed.

closed 2023-01-05 by arjenhiemstra