arjenhiemstra/ithowifi #119: WPU Heatpump settings incorrect

Onl1ne1373, 2022-11-15:
The problem is that setting 133 and higher are incorrect for a WPU Heatpump, they are shifted by one. 
For example, 151 (cooling blockade) should be 152.
But the min and max values are correct, so 152 has the correct values for min max.

It seems that the label of setting 133 (Reserved) is skipped in a list/array, but not in de min/max list.

tomkooij, 2022-12-08:
@Onl1ne1373 Do you have a complete overview (table) of the fw37 settings?

From 133 it is clearly shifted down, but 178 (seconds) is NOT shifted down as 178 has values between 0-14400 (4 hours) and the next one is -10 to 10.

I'd like to submit a PR to fix this, but I need more info.

closed 2023-01-09 by arjenhiemstra

Onl1ne1373, 2023-01-18:
Can confirm this is fixed. Tested beta 6. 
Thanks!!!

PR #133 (2023-01-09): Fix settings
This tries to fix the settings pages for non-CVE devices
(Hopefully CVE devices as well, but I cannot test on CVE devices)

TODO:
 -  [x] Remove special case datatype `0x01`
 - [x] Implement "special case" datatype `0x0f` (Float divider 2).
 - [x] Fix signed floats. Cast ints to float according to datatype length (see: https://github.com/arjenhiemstra/ithowifi/issues/129#issuecomment-1368176583)
 - [x] Support unsigned raw bytes to float, make identical to #130 
 - [x] Test on WPU 5G fw37 fix #119 first
 - [x] Test on Autotemp fw10
 - [x] Test on CVE device (?)

Fixes #129 
Fixes  [... cut]
commit 6cb218c6 (2023-01-09): Merge pull request #133 from tomkooij/fix_settings

Fixes #129
Fixes #126
Fixes #119