arjenhiemstra/ithowifi #120: Set Friendly name to Demand flow Valve's

DutchMikeDutch, 2022-11-16:
Currently the Values in the Demand Flow are.

living room 1
living room 2 
washroom
bathroom1 
bath2 + toilet2
bedroom 1
bedroom 2
bedroom 3
 bedroom 4
valve extractor

These Values are used for the Co2 calc, Valve pos, these naming's don't always correspond. to reality 

Example living room 2 for me is Toilet downstairs
Bathroom + Toilet2 for me is bedroom 5
Extractor is Cooking function

Hence instead of fixing the values in for example home assistant i would love a matching table on the web interface.

Best Regards
Mike

arjenhiemstra, 2022-11-17:
The values currently reflect the values used in the itho service tool and are currently fixed. Firmware rework needed.

arjenhiemstra, 2024-12-12:
Since 2.9.0-beta5 the Home Assistant discovery has been update. Label are still defaulted to these values but can be changed if needed. The given namen will be present in the HA interface.
Unfortunately this is currently HA only.