arjenhiemstra/ithowifi #123: Negative values problem for WPU and HRU300

Onl1ne1373, 2022-11-22:
We saw the first negative temperatures outside last weekend, but they were nog shown correctly.
For the WPU it was the outside temperature. For the WTW it was the air intake.
The behaviour of both fields are the same, it goes to a 650 values:

[image]

I think the problem is that the code assumes an unsigned value, but it is a signed value.

arjenhiemstra, 2022-11-22:
Could you execute and post the output of some queries? Probably a data format label from the statusformat queiry is not handled correctly.
On the debug page execute the statusformat and status queries.
And please provide the model and itho firmware version number.

Onl1ne1373, 2022-11-22:
### WPU 65i firmware 37

**DeviceType**: 80 82 90 E0 01 07 00 01 00 0D 4C 25 00 07

**StatusFormat**: 80 82 A4 00 01 86 92 92 92 92 92 92 92 92 92 92 92 92 92 92 0C 0C 0C 0C 92 10 0C 00 00 00 00 00 0C 0C 0C 0C 0C 10 92 92 00 00 00 00 92 92 92 92 00 00 10 0C 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 12 00 00 20 00 00 0C 0C 0C 0C 0C 0C 0C 0C 0C 0C 0C 0C 0C 0C 0C 0C 0C 0C 0C 0C 0C 0C 0C 0C 0C 0C 0C 00 20 20 20 20 0C 00 92 00 90 10 00 00 92 10 00 92 10 00 00 00 00 10 10 10 10 10 10 10 00 10 10 10 10 10 10 10 AD

**Status**: 80 82 A4 01 01 D5 02 53 15 C6 15 B8 01 53 03 A8 11 9C 0A 87 02 03 03 78 0A 73 09 12 00 E2 01 74 00 00 00 00 00 00 FF B7 06 13 00 64 51 00 00 00 01 00 00 01 00 01 E2 08 5B 08 34 2E 02 07 00 01 21 0B 1F 01 21 0B 1F 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 14 00 00 00 00 00 00 00 00 00 00 00 00 0D 82 00 2D 01 F4 00 00 63 7C CD 23 2D 00 01 00 00 00 00 00 00 00 00 00 00 00 01 00 00 00 00 00 00 01 00 00 00 00 01 00 00 2E 00 00 00 09 00 00 01 7E 00 00 01 3F 00 00 00 F3 00 01 27 10 00 00 00 00 00 00 00 03 1B 00 96 00 02 03 06 40 51 00 00 FF 00 00 00 00 0E 10 00 00 00 00 05 A0 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0A

### HRU ECO 300 firmware 8

**DeviceType**:  80 82 90 E0 05 09 00 01 07 01 09 08 00 00 03 63

**StatusFormat**: 80 82 A4 00 05 20 00 00 92 92 11 92 92 92 11 0F 0F 00 00 00 12 12 0F 00 01 10 10 10 10 10 92 92 10 00 10 10 0F 10 24

**Status**: 80 82 A4 01 05 34 01 04 03 13 03 79 03 0B 07 85 08 64 04 1B 04 1E 00 31 00 00 00 26 24 33 0B C8 16 00 05 78 00 00 00 00 00 00 03 0A 09 98 09 83 00 00 00 06 FD 00 00 74 02 17 64

But currently it is not freezing

tomkooij, 2022-12-08:
Outside temp is type `0x92` which seems to mean "temperature". 

I.e. a 2s-complement 2byte value, divided by 100. As seems to be the default in Honeywell evohome/ramses II. 

There currently is no support for negative values:
https://github.com/arjenhiemstra/ithowifi/blob/858980fc79852e9f0b588138f5cbcab8a37ab927/software/NRG_itho_wifi/src/IthoSystem.cpp#L714

(I'll try to push a PR, but not high priority)

arjenhiemstra, 2022-12-08:
Thanks! Apparently i've implemented support for signed integers but not for negative floats... looking forward to your PR!

tomkooij, 2022-12-09:
@arjenhiemstra : The itho info is just "from experience" I have no source at hand. But the ramses_ii protocol seems to encode temps using signed ints divided by 100 all the time: https://github.com/zxdavb/ramses_protocol/wiki/Decoding-Data-Fields

In itho datatypes seen on the i2c bus "signed floats" seems to be 0x91 and 0x92:

0x92 =
0b10010010
bit 0, 1, 2: divisor (implemented)
bit 7: signed if set (implemented for ints, but not floats)

Itho list temperatures as type 0x91 (signed "float" divisor 10) and 0x92 (signed "float" divisor 100) in the i2c modbus manual: https://ithodaalderop.compano.com/Data/Environments/000001/Attachment/Bijlage/A02_Warmtepompen/A02_80_Accessoires/A02_80_01_Grond/A02_80_01_03_Overig/B01_06_MTE/01-04844-002%20%20Handleiding%20I%c2%b2C%20Modbus%20Module%20(nl).pdf

tomkooij, 2022-12-15:
I've fixed the status part on a local branch:
[image]

But I'll try to refactor the datatype handling of status/settings/measurements and sumbit a PR.

EdwinGH, 2022-12-17:
I have the same on my HRU300: get=ithostatus results in "Measured outside temperature (Â°C)":654.11,"Measured temperature of mixed outside air (Â°C)":652.35

Query Devicetype Result: 80 82 90 E0 05 09 00 01 07 01 06 07 00 00 00 6A
Query Status Format Result: 80 82 A4 00 05 20 00 00 92 92 11 92 92 92 11 0F 0F 00 00 00 12 12 0F 00 01 10 10 10 10 10 92 92 10 00 10 10 0F 10 24
Query Status Result: 80 82 A4 01 05 34 01 00 FF 83 FE D5 00 00 07 71 07 F6 09 D2 0C 04 51 73 00 00 18 00 00 8D A8 C8 16 05 05 78 00 00 00 00 00 00 04 51 09 DF 09 DD 00 00 00 0D 95 00 00 00 04 29 07

As I am checking the boundaries of the results (temp between -50 and 50) before I store in my database, I'm not recording any values these days :-(
@tomkooij is the fix almost ready to submit? :-)

closed 2023-01-05 by arjenhiemstra

Onl1ne1373, 2023-01-20:
Beta 6 tested, works:

[image]

Thanks!!

PR #130 (2023-01-05): Fix unsigned floats
Fixes #123 

This is a bit of a hack, as it implements a "is_signed" flag for "floats" but leaves ints and uints in place.
In reality all "itho raw bytes" are either signed/unsigned and have some divider. Ints are just floats with divider 1.

This should be refactored with `IthoDeviceStatus::type`  (int/uints/float) removed and only the `is_signed` flag in place.
 
But this works:

Cast "signed raw bytes" to signed integer first, before casting to float and applying a divider.

Example: datatype = 0x92 (1-0-010-010)
datatype & 0x80 (bit 8) => signed value
datatype & 0x38 (b [... cut]