arjenhiemstra/ithowifi #124: Summercounter shows weird values on non-CVE 2.4.3 and 2.4.4-beta1

janbredenbeek, 2022-11-22:
On a HRU 350 with non-CVE 2.4.3 and 2.4.4-beta1, Summercounter in Itho status shows as 0.000489738 rather than 0 (the actual value may vary). This didn't happen with 2.4.2.

arjenhiemstra, 2022-11-23:
That seems like a rounding error but as I'm not aware of any changes in the HRU350 allocation or how values get calculated.
Could you give some info on the firmware version of you HRU and post the output of the StatusFormat query on the debug page?

janbredenbeek, 2022-11-23:
It's getting more weird... I downgraded to 2.4.2 which fixed the error, then upgraded again to 2.4.4-beta1 to compare the output of the StatusFormat query, and the error is gone! The StatusFormat query is the same for both versions:

80 82 A4 00 01 21 91 11 10 90 10 90 92 92 00 92 92 00 00 91 00 10 10 00 90 00 00 10 00 00 90 90 00 0F 20 20 00 91 00 BD

I'll let you know if anything changes...

janbredenbeek, 2022-11-24:
After a spontaneous reboot the error is back again... Summercounter shows 0.000489939. Status format query hasn't changed though.

2022-11-24 00:40:07 I: Mem free: 108808, Mem low: 63572, Mem block: 44020
2022-11-24 00:43:15 I: System boot, last reset reason: OWDT_RESET
2022-11-24 00:43:15 I: HW rev: NON-CVE 1, FW ver.: 2.4.4-beta1
2022-11-24 00:43:23 I: WiFi: connection successful
2022-11-24 00:43:24 I: WiFi info:
2022-11-24 00:43:24 I: SSID:<deleted>
2022-11-24 00:43:24 I: RSSI:-37dBm
2022-11-24 00:43:24 I: Mode:STA
2022-11-24 00:43:24 I: Status:WL_CONNECTED
2022-11-24 00:43:24 I: IP:<deleted>
2022-11-24 00:43:25 I: Setup: Virtual remotes, start ID: 3D,85,24 - No.: 1
2022-11-24 00:43:25 I: Setup: remotes configfile loaded
2022-11-24 00:43:25 I: Setup: init of CC1101 RF module successful
2022-11-24 00:43:25 I: Webserver: started
2022-11-24 00:43:26 I: Setup: remotes configfile loaded
2022-11-24 00:43:26 I: mDNS: started
2022-11-24 00:43:26 I: Hostname: nrg-itho-8524
2022-11-24 00:43:26 I: Setup: done
2022-11-24 00:43:28 I: I2C init: QueryDevicetype - fw:4 hw:43
2022-11-24 00:43:28 I: I2C init: QueryStatusFormat - items:33
2306 I: System boot, last reset reason: OWDT_RESET
2022-11-24 00:40:39 I: HW rev: NON-CVE 1, FW ver.: 2.4.4-beta1
2022-11-24 00:40:47 I: WiFi: connection successful
2022-11-24 00:40:47 I: WiFi info:
2022-11-24 00:40:47 I: SSID:<deleted>
2022-11-24 00:40:47 I: RSSI:-38dBm
2022-11-24 00:40:48 I: Mode:STA
2022-11-24 00:40:48 I: Status:WL_CONNECTED
2022-11-24 00:40:48 I: IP:<deleted>
2022-11-24 00:40:48 I: Setup: Virtual remotes, start ID: 3D,85,24 - No.: 1
2022-11-24 00:40:36 I: Setup: remotes configfile loaded
2022-11-24 00:40:36 I: Setup: init of CC1101 RF module successful
2022-11-24 00:40:37 I: Webserver: started
2022-11-24 00:40:37 I: Setup: remotes configfile loaded
2022-11-24 00:40:37 I: mDNS: started
2022-11-24 00:40:37 I: Hostname: nrg-itho-8524
2022-11-24 00:40:37 I: Setup: done
2022-11-24 00:40:39 I: I2C init: QueryDevicetype - fw:4 hw:43
2022-11-24 00:40:40 I: I2C init: QueryStatusFormat - items:33
2022-11-24 00:40:40 I: I2C init: QueryStatus

The change in time stamp from 00:40 to 00:43 after the first reboot looks weird, maybe a NTP sync issue? (my router is configured as NTP server).
While I wrote this the Summercounter showed 0 again but after a manual reboot it shows 1.485811505e-314...

janbredenbeek, 2022-11-25:
Some more problems: after a few hours, the status froze with a 'I2C bus stuck' message. This is from the log:

2022-11-24 23:24:31 I: mDNS: started
2022-11-24 23:24:31 I: Hostname: nrg-itho-8524
2022-11-24 23:24:31 I: Setup: done
2022-11-24 23:24:33 I: I2C init: QueryDevicetype - fw:4 hw:43
2022-11-24 23:24:33 I: I2C init: QueryStatusFormat - items:33
2022-11-24 23:24:34 I: I2C init: QueryStatus
2022-11-25 05:25:28 I: Mem free: 108872, Mem low: 69244, Mem block: 44020
22569552 I: Error: I2C bus could not be cleared!
2022-11-25 11:20:51 I: Reboot requested
2022-11-25 11:21:54 I: System boot, last reset reason: SDIO_RESET
2022-11-25 11:21:54 I: HW rev: NON-CVE 1, FW ver.: 2.4.4-beta1

arjenhiemstra, 2023-01-05:
Thats an issue with that beta release, it was deliberately stopping the i2c bus on any errors to be able to capture logging.
Since 2.4.4-beta4 this happens only if the i2c debug menu is active

janbredenbeek, 2023-01-20:
Since v2.4.4-beta6 the Summercounter again shows small nonzero values (0.000489009). Possibly a relationship with Issue #142 ?

arjenhiemstra, 2023-01-20:
It a rounding thing, indeed the same as #142

closed 2023-03-28 by stale[bot]