arjenhiemstra/ithowifi #126: Autotemp - Cooling and Heating Setpoint not in Degrees Celsius

WouterJN, 2022-12-07:
**Describe the bug**

For my autotemp at Itho Settings page I expected the the parameters "cooling min setpoint", "cooling max setpoint", "heating max setpoint", "heating min setpoint" to be in degrees celsius. However me and another Tweakers.net have strange values that are not degrees celsius.
[image]
[image]

**Expected behaviour**
Current values should be:
heating min setpoint: 18 degrees celsius
heating max setpoint: 24 degrees celcius

**Device information**
Itho device type: AutoTemp 
Itho fw version: 4 
HW rev: NON-CVE 1, FW ver.: 2.4.3

tomkooij, 2022-12-13:
@WouterJN You are right. The settings page is wrong. These are the correct parameters (for a v10 device):
[image]

tomkooij, 2022-12-30:
This is indeed a datatype problem:
Autotemp fw 10 setting 42 is datatype `0x0f` which is a single byte with divisor 2. (Divide byte int by 2).
In the status page it is implemented correctly.

[image]

WouterJN, 2022-12-30:
Thank you Tom! I will try after the PR has been approved and is part of the compiled firmware.

PR #133 (2023-01-09): Fix settings
This tries to fix the settings pages for non-CVE devices
(Hopefully CVE devices as well, but I cannot test on CVE devices)

TODO:
 -  [x] Remove special case datatype `0x01`
 - [x] Implement "special case" datatype `0x0f` (Float divider 2).
 - [x] Fix signed floats. Cast ints to float according to datatype length (see: https://github.com/arjenhiemstra/ithowifi/issues/129#issuecomment-1368176583)
 - [x] Support unsigned raw bytes to float, make identical to #130 
 - [x] Test on WPU 5G fw37 fix #119 first
 - [x] Test on Autotemp fw10
 - [x] Test on CVE device (?)

Fixes #129 
Fixes  [... cut]
commit 6cb218c6 (2023-01-09): Merge pull request #133 from tomkooij/fix_settings

Fixes #129
Fixes #126
Fixes #119
closed 2023-01-09 by arjenhiemstra