arjenhiemstra/ithowifi #127: Itho WPU crashes IthoWifi module

WouterJN, 2022-12-09:
When I connect my WPU with the IthoWifi module the Ithowifi module crashes and is not longer connectible. It is also no longer possible to ping the module. The same module on my HRU and Autotemp is working fine. 
Few year back I made a I2C level shifter in combination with a RaspberryPi, that one have no problems connecting with the WPU.

Hardware:
WPU --> HardwareType: WPU, ProductVersion: 20, ListVersion: 9 (DataLabel 8)
IthoWifi --> HW rev: NON-CVE 1, FW ver.: 2.4.3

I tried with three different ethernet cables and two power supplies (incl different cable). I also tried different sequence of connecting everything but nothing helps.

The wifi signal is strong, so that can not be the problem.

What could I do to debug this?

WouterJN, 2022-12-20:
Hi @arjenhiemstra anyway I can debug this problem?

arjenhiemstra, 2022-12-21:
Hi Wouter, could you provide some more information? Please have a look at the bug report template when creating an issue, you do not have to fill in all info but a little more than what is now provided would be much appreciated.

WouterJN, 2022-12-27:
I tried both my IthoWifi modules without external power supply but unfortunately they both crash when I connect then with my WPU. With my HRU and Autotemp they work both fine.

**Debug logging (this will keep repeating)**
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
E (6085) wifi_init_default: esp_wifi_get_mac failed with 12289
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d7ae1  PS      : 0x00060930  A0      : 0x800d879d  A1      : 0x3ffcdfe0  
A2      : 0x00000000  A3      : 0x0000000d  A4      : 0x00000009  A5      : 0x00000000  
A6      : 0x80828a00  A7      : 0x00000020  A8      : 0x3f40fbb0  A9      : 0x000003e7  
A10     : 0x00000000  A11     : 0x3f40fd90  A12     : 0x00000024  A13     : 0x00000000  
A14     : 0x000003e7  A15     : 0x00000000  SAR     : 0x0000001a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400d7adc  LEND    : 0x400d7b1c  LCOUNT  : 0x000003e6  

Backtrace:0x400d7ade:0x3ffcdfe00x400d879a:0x3ffce000 0x400e19b0:0x3ffce250 0x400e1fb8:0x3ffce380 

ELF file SHA256: 0000000000000000

Rebooting...`

**Device information**
- Firmware version [e.g. 2.2.1]: 2.4.3
- Hardware revision (e.g. non-cve 1.1): NON-CVE 1 (based on system log, ordered a few weeks back)
- Type/model number Itho (on sticker outside the box): WPU-3I-S 545-0050. Based on other i2c project: ProductVersion: 20, ListVersion: 9 (DataLabel 8)
- CC1101 RF module enabled [ yes, no ]: Yes and No (I have two IthoWifi modules)
- In case of wifi issues, wifi access point make and model: Unifi, multiple accespoints. Wifi is working fine when connected with autotemp and HRU

arjenhiemstra, 2022-12-27:
Could you power the device with an USB adapter and not connect it to any device? Then go to the webpage of the add-on and have a look at the logging on the debug page. Please post the logging here.

WouterJN, 2022-12-27:
Do you mean the following?;
```
022-12-27 12:05:21 I: System boot, last reset reason: OWDT_RESET
2022-12-27 12:05:21 I: HW rev: NON-CVE 1, FW ver.: 2.4.3
2022-12-27 12:05:27 I: WiFi: connection successful
2022-12-27 12:05:27 I: WiFi info:
2022-12-27 12:05:28 I:   SSID:SECRET | BSSID[SECRET:AA:B2:4C]
2022-12-27 12:05:28 I:   RSSI:-36dBm
2022-12-27 12:05:28 I:   Mode:STA
2022-12-27 12:05:28 I:   Status:WL_CONNECTED
2022-12-27 12:05:28 I:   IP:192.168.2.147
2022-12-27 12:05:28 I: Setup: Virtual remotes, start ID: AB,5D,18 - No.: 1
2022-12-27 12:05:28 I: Setup: remotes configfile loaded
2022-12-27 12:05:29 I: MQTT: connected, System config: 1
2022-12-27 12:05:29 I: Webserver: started
2022-12-27 12:05:29 I: mDNS: started
2022-12-27 12:05:29 I: Hostname: nrg-itho-5d18
2022-12-27 12:05:29 I: Setup: done
2022-12-27 12:05:36 I: System boot, last reset reason: OWDT_RESET
2022-12-27 12:05:37 I: HW rev: NON-CVE 1, FW ver.: 2.4.3
2022-12-27 12:05:43 I: WiFi: connection successful
2022-12-27 12:05:43 I: WiFi info:
2022-12-27 12:05:43 I:   SSID:SECRET | BSSID[SECRET:AA:B2:4C]
2022-12-27 12:05:43 I:   RSSI:-35dBm
2022-12-27 12:05:43 I:   Mode:STA
2022-12-27 12:05:44 I:   Status:WL_CONNECTED
2022-12-27 12:05:44 I:   IP:192.168.2.147
2022-12-27 12:05:44 I: Setup: Virtual remotes, start ID: AB,5D,18 - No.: 1
2022-12-27 12:05:44 I: Setup: remotes configfile loaded
2022-12-27 12:05:45 I: MQTT: connected, System config: 1
2022-12-27 12:05:45 I: Webserver: started
2022-12-27 12:05:45 I: mDNS: started
2022-12-27 12:05:45 I: Hostname: nrg-itho-5d18
2022-12-27 12:05:45 I: Setup: done
2022-12-27 12:05:52 I: System boot, last reset reason: OWDT_RESET
2022-12-27 12:05:52 I: HW rev: NON-CVE 1, FW ver.: 2.4.3
2022-12-27 12:05:58 I: WiFi: connection successful
2022-12-27 12:05:59 I: WiFi info:
2022-12-27 12:05:59 I:   SSID:SECRET | BSSID[SECRET:AA:B2:4C]
2022-12-27 12:05:59 I:   RSSI:-40dBm
2022-12-27 12:05:59 I:   Mode:STA
2022-12-27 12:05:59 I:   Status:WL_CONNECTED
2022-12-27 12:05:59 I:   IP:192.168.2.147
2022-12-27 12:06:00 I: Setup: Virtual remotes, start ID: AB,5D,18 - No.: 1
2022-12-27 12:06:00 I: Setup: remotes configfile loaded
2022-12-27 12:06:01 I: MQTT: connected, System config: 1
2022-12-27 12:06:01 I: Webserver: started
2022-12-27 12:06:01 I: mDNS: started
2022-12-27 12:06:02 I: Hostname: nrg-itho-5d18
2022-12-27 12:06:02 I: Setup: done
2022-12-27 12:06:08 I: System boot, last reset reason: OWDT_RESET
2022-12-27 12:06:09 I: HW rev: NON-CVE 1, FW ver.: 2.4.3
2022-12-27 12:06:15 I: WiFi: connection successful
2022-12-27 12:06:15 I: WiFi info:
2022-12-27 12:06:15 I:   SSID:SECRET | BSSID[SECRET:AA:B2:4C]
2022-12-27 12:06:15 I:   RSSI:-34dBm
2022-12-27 12:06:15 I:   Mode:STA
2022-12-27 12:06:15 I:   Status:WL_CONNECTED
2022-12-27 12:06:16 I:   IP:192.168.2.147
2022-12-27 12:06:16 I: Setup: Virtual remotes, start ID: AB,5D,18 - No.: 1
2022-12-27 12:06:16 I: Setup: remotes configfile loaded
2022-12-27 12:06:20 I: MQTT: connected, System config: 1
2022-12-27 12:06:20 I: Webserver: started
2022-12-27 12:06:20 I: mDNS: started
2022-12-27 12:06:21 I: Hostname: nrg-itho-5d18
2022-12-27 12:06:21 I: Setup: done
```

arjenhiemstra, 2022-12-27:
> Hi Arjen, thank you for your reply. I tried both my IthoWifi modules without external power supply but unfortunately they both crash when I connect then with my WPU. With my HRU and Autotemp they work both fine.
> 
> **Debug logging** See attachment
> 
> **Serial logging**
> 
> ```
> ets Jul 29 2019 12:21:46
> 
> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:2
> load:0x3fff0030,len:1184
> load:0x40078000,len:13132
> load:0x40080400,len:3036
> entry 0x400805e4
> E (6085) wifi_init_default: esp_wifi_get_mac failed with 12289
> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
> 
> Core  1 register dump:
> PC      : 0x400d7ae1  PS      : 0x00060930  A0      : 0x800d879d  A1      
> [logfile0-2.log](https://github.com/arjenhiemstra/ithowifi/files/10308114/logfile0-2.log)
> : 0x3ffcdfe0  
> A2      : 0x00000000  A3      : 0x0000000d  A4      : 0x00000009  A5      : 0x00000000  
> A6      : 0x80828a00  A7      : 0x00000020  A8      : 0x3f40fbb0  A9      : 0x000003e7  
> A10     : 0x00000000  A11     : 0x3f40fd90  A12     : 0x00000024  A13     : 0x00000000  
> A14     : 0x000003e7  A15     : 0x00000000  SAR     : 0x0000001a  EXCCAUSE: 0x0000001c  
> EXCVADDR: 0x00000000  LBEG    : 0x400d7adc  LEND    : 0x400d7b1c  LCOUNT  : 0x000003e6  
> 
> 
> Backtrace:0x400d7ade:0x3ffcdfe00x400d879a:0x3ffce000 0x400e19b0:0x3ffce250 0x400e1fb8:0x3ffce380 
> 
> 
> 
> 
> ELF file SHA256: 0000000000000000
> 
> Rebooting...
> ```
> 
> **Device information**
> 
> * Firmware version [e.g. 2.2.1]: 2.4.3
> * Hardware revision (e.g. non-cve 1.1): NON-CVE 1 (based on system log, ordered a few weeks back)
> * Type/model number Itho (on sticker outside the box): WPU-3I-S 545-0050. Based on other i2c project: ProductVersion: 20, ListVersion: 9 (DataLabel 8)
> * CC1101 RF module enabled [ yes, no ]: Yes and No (I have two IthoWifi modules)
> * In case of wifi issues, wifi access point make and model: Unifi, multiple accespoints. Wifi is working fine when connected with autotemp and HRU

This serial logging is helpful thanks! 
Tried to decode the backtrace, this is the result:
PC: 0x400d7ae1: IthoRemote::toggleLearnLeaveMode() at /<>/software/NRG_itho_wifi/src/IthoRemote.cpp line 43
EXCVADDR: 0x00000000

Decoding stack results
0x400d7ade: IthoRemote::toggleLearnLeaveMode() at /<>/software/NRG_itho_wifi/src/IthoRemote.cpp line 43
0x400d879a: ArduinoJson6194_F1::VariantData::asIntegral () const at /<>/Arduino/libraries/ArduinoJson/src/ArduinoJson/Numbers/convertNumber.hpp line 110
0x400e19b0: failSafeBoot() at /<>/software/NRG_itho_wifi/src/task_init.cpp line 149
0x400e1fb8: ArduinoJson6194_F1::TextFormatter ::writeChar(char) at /<>/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/ConverterImpl.hpp line 15

That would indicate and issue with the CC1101 module but you stated that one model doesn't has the CC1101. 
Are there itho RF devices for the WPU? If not, could you update the moduel without CC1101 module to the latest beta release and see if you could capture another core dump like the one above?

I expected an issue with WPU firmware versions not (yet) in the add-on firmware and maybe pointing to some bogus mem. location but I'm not so sure anymore.

WouterJN, 2022-12-27:
With the Beta version on my module with CC1101 I have exactly the same problems.

**Serial log**
```

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d7fe5  PS      : 0x00060030  A0      : 0x800d85e5  A1      : 0x3ffce5e0  
A2      : 0x00000000  A3      : 0x0000000d  A4      : 0x00000009  A5      : 0x00000000  
A6      : 0x3ffce60d  A7      : 0x00000020  A8      : 0x3f41015c  A9      : 0x000003e7  
A10     : 0x00000000  A11     : 0x3f41033c  A12     : 0x00000024  A13     : 0x00000000  
A14     : 0x000003e7  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400d7fe0  LEND    : 0x400d8020  LCOUNT  : 0x000003e6  

Backtrace:0x400d7fe2:0x3ffce5e00x400d85e2:0x3ffce600 0x400e3206:0x3ffce850 0x400e34e7:0x3ffce890 0x400e3974:0x3ffce8e0 

ELF file SHA256: 0000000000000000

Rebooting...

```

**Previous log**

```
2022-12-27 20:00:08 I: Firmware update: nrgitho-noncve-v2.4.4-beta3.bin
2022-12-27 20:00:22 I: Reboot requested
2022-12-27 19:57:06 I: System boot, last reset reason: SDIO_RESET
2022-12-27 19:57:07 I: HW rev: NON-CVE 2, FW ver.: 2.4.4-beta3
2022-12-27 19:57:07 I: I2C sniffer capable hardware: yes
2022-12-27 19:57:08 E: Unable to set wifi disconnect
2022-12-27 19:57:14 I: WiFi: connection successful
2022-12-27 19:57:14 I: MQTT: connected, System config: 1
2022-12-27 19:57:14 I: Webserver: started
2022-12-27 19:57:15 I: mDNS: started
2022-12-27 19:57:15 I: Hostname: nrg-itho-5d18
2022-12-27 19:57:20 E: I2C init: QueryDevicetype - failed
      1268 I: System boot, last reset reason: POWERON_RESET
      1517 I: HW rev: NON-CVE 2, FW ver.: 2.4.4-beta3
      1682 I: I2C sniffer capable hardware: yes
      2839 E: Unable to set wifi disconnect
      8124 I: WiFi: connection successful
      8882 I: MQTT: connected, System config: 1
      9075 I: Webserver: started
      9273 I: mDNS: started
      9484 I: Hostname: nrg-itho-5d18
      1251 I: System boot, last reset reason: OWDT_RESET
      1449 I: HW rev: NON-CVE 2, FW ver.: 2.4.4-beta3
      1604 I: I2C sniffer capable hardware: yes
      2686 E: Unable to set wifi disconnect
      8274 I: WiFi: connection successful
      9015 I: MQTT: connected, System config: 1
      9234 I: Webserver: started
      9436 I: mDNS: started
      9640 I: Hostname: nrg-itho-5d18
      1456 I: System boot, last reset reason: OWDT_RESET
      1633 I: HW rev: NON-CVE 2, FW ver.: 2.4.4-beta3
      1777 I: I2C sniffer capable hardware: yes
[... 90 lines cut ...]
2022-12-27 20:54:00 I: Hostname: nrg-itho-5d18
2022-12-27 20:54:07 I: System boot, last reset reason: OWDT_RESET
2022-12-27 20:54:08 I: HW rev: NON-CVE 2, FW ver.: 2.4.4-beta3
2022-12-27 20:54:08 I: I2C sniffer capable hardware: yes
2022-12-27 20:54:09 E: Unable to set wifi disconnect
2022-12-27 20:54:15 I: WiFi: connection successful
2022-12-27 20:54:15 I: MQTT: connected, System config: 1
2022-12-27 20:54:16 I: Webserver: started
2022-12-27 20:54:16 I: mDNS: started
2022-12-27 20:54:16 I: Hostname: nrg-itho-5d18
2022-12-27 20:54:21 E: I2C init: QueryDevicetype - failed
      1256 I: System boot, last reset reason: POWERON_RESET
      1505 I: HW rev: NON-CVE 2, FW ver.: 2.4.4-beta3
      1673 I: I2C sniffer capable hardware: yes
      2982 E: Unable to set wifi disconnect
      9640 I: WiFi: connection successful
2022-12-27 20:56:10 E: I2C init: QueryDevicetype - failed
2022-12-27 20:56:14 I: MQTT: connected, System config: 1
2022-12-27 20:56:14 I: Webserver: started
2022-12-27 20:56:15 I: mDNS: started
2022-12-27 20:56:15 I: Hostname: nrg-itho-5d18
```

**other info**
For an other Itho WPU i2c project (https://github.com/pommi/python-itho-wpu) I used the table called Datalabel_V8 from the the Itho Service tool MS Access database. Is that the same as the translation table "itho_WPUstatus8_9[]" at your code?

arjenhiemstra, 2022-12-28:
Could you try this firmware?
[nrgitho-dev-v2.4.4-beta4.bin.zip](https://github.com/arjenhiemstra/ithowifi/files/10314149/nrgitho-dev-v2.4.4-beta4.bin.zip)

It will not solve the issue but will stop the i2c init after the device detection query and log a similar row in the syslog like this:
I: I2C init: QueryDevicetype - fw:xx hw:xx

This information is indeed used to map to the correct status and settings labels and I suspect something is not correct in the current add-on firmware for the WPU version you have.

WouterJN, 2022-12-28:
Thank you for creating this special firmware. 

**serial log**
```
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
Mounting FS...
System boot, last reset reason: OWDT_RESET
Hardware detected: 0x34
HW rev: NON-CVE 2, FW ver.: 2.4.4-beta4
I2C sniffer capable hardware: yes
I2C master pins - SDA: 27 SCL: 26
I2C slave pins - SDA: 27 SCL: 26
I2C sniffer pins - SDA: 14 SCL: 25
config : 1
Setup: syslog config loaded : 1
Setup: wifi config loaded successful
Connecting to wireless network...
esp_wifi_restore: ESP_OK
Unable to set wifi disconnect
WiFi.setHostname: OK
esp_wifi_set_max_tx_power: ESP_OK
esp_wifi_get_max_tx_power: ESP_OK - level:78
esp_wifi_set_storage: ESP_OK
esp_wifi_set_ps: ESP_OK
WiFi: connection successful
WiFi info:
  SSID:Secret | BSSID[Secret:29:AA:B2:4C]
  RSSI:-46dBm
  Mode:STA
  Status:WL_CONNECTED
  IP:192.168.2.147
Setup: Virtual remotes, start ID: AB,5D,18 - No.: 1
Setup: remotes configfile loaded
MQTT: connected, System config: 1
Webserver: started
mDNS: started
Hostname: nrg-itho-5d18
Setup: done
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d7fd5  PS      : 0x00060030  A0      : 0x800d870d  A1      : 0x3ffce620  
A2      : 0x00000000  A3      : 0x0000000d  A4      : 0x00000009  A5      : 0x00000000  
A6      : 0x3ffce64d  A7      : 0x00000020  A8      : 0x3f410164  A9      : 0x000003e7  
A10     : 0x00000000  A11     : 0x3f410344  A12     : 0x00000024  A13     : 0x00000000  
A14     : 0x000003e7  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400d7fd0  LEND    : 0x400d8010  LCOUNT  : 0x000003e6  

Backtrace:0x400d7fd2:0x3ffce6200x400d870a:0x3ffce640 0x400e2d73:0x3ffce890 0x400e347f:0x3ffce8b0 0x400e3908:0x3ffce900 

ELF file SHA256: 0000000000000000

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
[... 102 lines cut ...]
2022-12-28 12:41:13 E: Unable to set wifi disconnect
2022-12-28 12:41:19 I: WiFi: connection successful
2022-12-28 12:41:20 I: MQTT: connected, System config: 1
2022-12-28 12:41:20 I: Webserver: started
2022-12-28 12:41:21 I: mDNS: started
2022-12-28 12:41:21 I: Hostname: nrg-itho-5d18
2022-12-28 12:41:28 I: System boot, last reset reason: OWDT_RESET
2022-12-28 12:41:28 I: HW rev: NON-CVE 2, FW ver.: 2.4.4-beta4
2022-12-28 12:41:28 I: I2C sniffer capable hardware: yes
2022-12-28 12:41:29 E: Unable to set wifi disconnect
2022-12-28 12:41:35 I: WiFi: connection successful
2022-12-28 12:41:36 I: MQTT: connected, System config: 1
2022-12-28 12:41:36 I: Webserver: started
2022-12-28 12:41:36 I: mDNS: started
2022-12-28 12:41:36 I: Hostname: nrg-itho-5d18
2022-12-28 12:41:43 I: System boot, last reset reason: OWDT_RESET
2022-12-28 12:41:43 I: HW rev: NON-CVE 2, FW ver.: 2.4.4-beta4
2022-12-28 12:41:43 I: I2C sniffer capable hardware: yes
2022-12-28 12:41:44 E: Unable to set wifi disconnect
2022-12-28 12:41:50 I: WiFi: connection successful
2022-12-28 12:41:51 I: MQTT: connected, System config: 1
2022-12-28 12:41:51 I: Webserver: started
2022-12-28 12:41:52 I: mDNS: started
2022-12-28 12:41:52 I: Hostname: nrg-itho-5d18
```

arjenhiemstra, 2022-12-28:
It seems to crash on something that has to do with a remote index.

Could you share the content of the config.json file? (To find this file, go to the debug page and go to file editor)
You may leave out any info that you deem sensitive (passwords etc)

WouterJN, 2022-12-28:
`{"sys_username":"admin","sys_password":"admin","syssec_web":0,"syssec_api":0,"syssec_edit":0,"api_normalize":0,"syssht30":0,"itho_rf_support":0,"rfInitOK":false,"itho_fallback":20,"itho_low":20,"itho_medium":120,"itho_high":220,"itho_timer1":10,"itho_timer2":20,"itho_timer3":30,"itho_updatefreq":60,"itho_numvrem":1,"itho_sendjoin":0,"itho_pwminit_en":1,"itho_forcemedium":0,"itho_vremoteapi":0,"nonQ_cmd_clearsQ":1,"mqtt_active":1,"mqtt_serverName":"192.168.3.1","mqtt_username":"itho","mqtt_password":"Secret","mqtt_port":1883,"mqtt_version":1,"mqtt_state_topic":"autotemp/state","mqtt_ithostatus_topic":"autotemp/ithostatus","mqtt_remotesinfo_topic":"autotemp/remotesinfo","mqtt_lastcmd_topic":"autotemp/lastcmd","mqtt_state_retain":"yes","mqtt_cmd_topic":"autotemp/cmd","mqtt_lwt_topic":"autotemp/lwt","mqtt_domoticz_active":0,"mqtt_ha_active":1,"mqtt_ha_topic":"homeassistant","mqtt_idx":1,"sensor_idx":1,"version_of_program":"005"}
`

arjenhiemstra, 2022-12-30:
It's still puzzling me. I've made a few changes again, could you try this firmware?
[firmware.bin.zip](https://github.com/arjenhiemstra/ithowifi/files/10323754/firmware.bin.zip)

I'm this time only interested in the backtrace though the serial.

WouterJN, 2022-12-30:
Thank you again for creating a special firmware.

First I got the following error:
`[  9644][E][vfs_api.cpp:104] open(): /littlefs/logfile0.current.log does not exist, no permits for creation`

So I created the file with file editor, and then got to error
`[vfs_api.cpp:104] open(): /littlefs/logfile1.log does not exist, no permits for creation`

After I created that file I got the following serial output;

```
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d92d5  PS      : 0x00060030  A0      : 0x800d9999  A1      : 0x3ffce660  
A2      : 0x00000000  A3      : 0x0000000d  A4      : 0x00000009  A5      : 0x3ffd1be4  
A6      : 0x00000000  A7      : 0x00000020  A8      : 0x3f41015c  A9      : 0x000003e7  
A10     : 0x00000000  A11     : 0x3f41033c  A12     : 0x00000024  A13     : 0x00000000  
A14     : 0x000003e7  A15     : 0x00000000  SAR     : 0x00000012  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400d92d0  LEND    : 0x400d931e  LCOUNT  : 0x000003e6  

Backtrace:0x400d92d2:0x3ffce6600x400d9996:0x3ffce690 0x400e6e26:0x3ffce8e0 0x400e7107:0x3ffce920 0x400e75a4:0x3ffce980 

ELF file SHA256: 0000000000000000

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
E (5171) wifi_init_default: netstack cb reg failed with 12308
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d92d5  PS      : 0x00060230  A0      : 0x800d9999  A1      : 0x3ffce660  
A2      : 0x00000000  A3      : 0x0000000d  A4      : 0x00000009  A5      : 0x3ffd1be4  
A6      : 0x00000000  A7      : 0x00000020  A8      : 0x3f41015c  A9      : 0x000003e7  
A10     : 0x00000000  A11     : 0x3f41033c  A12     : 0x00000024  A13     : 0x00000000  
A14     : 0x000003e7  A15     : 0x00000000  SAR     : 0x00000019  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400d92d0  LEND    : 0x400d931e  LCOUNT  : 0x000003e6  

Backtrace:0x400d92d2:0x3ffce6600x400d9996:0x3ffce690 0x400e6e26:0x3ffce8e0 0x400e7107:0x3ffce920 0x400e75a4:0x3ffce980 

ELF file SHA256: 0000000000000000

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
E (5068) wifi_init_default: netstack cb reg failed with 12308
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d92d5  PS      : 0x00060030  A0      : 0x800d9999  A1      : 0x3ffce660  
A2      : 0x00000000  A3      : 0x0000000d  A4      : 0x00000009  A5      : 0x3ffd1be4  
A6      : 0x00000000  A7      : 0x00000020  A8      : 0x3f41015c  A9      : 0x000003e7  
A10     : 0x00000000  A11     : 0x3f41033c  A12     : 0x00000024  A13     : 0x00000000  
A14     : 0x000003e7  A15     : 0x00000000  SAR     : 0x0000000f  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400d92d0  LEND    : 0x400d931e  LCOUNT  : 0x000003e6  

Backtrace:0x400d92d2:0x3ffce6600x400d9996:0x3ffce690 0x400e6e26:0x3ffce8e0 0x400e7107:0x3ffce920 0x400e75a4:0x3ffce980 

ELF file SHA256: 0000000000000000

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
E (4932) wifi_init_default: netstack cb reg failed with 12308
[ 13693][E][vfs_api.cpp:104] open(): /littlefs/system.gz does not exist, no permits for creation
[ 13699][E][vfs_api.cpp:104] open(): /littlefs/system does not exist, no permits for creation
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d92d5  PS      : 0x00060030  A0      : 0x800d9999  A1      : 0x3ffce660  
A2      : 0x00000000  A3      : 0x0000000d  A4      : 0x00000009  A5      : 0x3ffd1be4  
A6      : 0x00000000  A7      : 0x00000020  A8      : 0x3f41015c  A9      : 0x000003e7  
A10     : 0x00000000  A11     : 0x3f41033c  A12     : 0x00000024  A13     : 0x00000000  
A14     : 0x000003e7  A15     : 0x00000000  SAR     : 0x00000003  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400d92d0  LEND    : 0x400d931e  LCOUNT  : 0x000003e6  

Backtrace:0x400d92d2:0x3ffce6600x400d9996:0x3ffce690 0x400e6e26:0x3ffce8e0 0x400e7107:0x3ffce920 0x400e75a4:0x3ffce980 

ELF file SHA256: 0000000000000000

Rebooting...
```

After some time I got the error `[vfs_api.cpp:104] open(): /littlefs/logfile0.current.log does not exist, no permits for creation` again and the files where gone at the file editor.

arjenhiemstra, 2022-12-30:
Thanks! Now the backtrace decode works and it seems to point where I expected it to point:

```
0x400d92d2: getSettingsLength(unsigned char, unsigned char, unsigned char) at /ithowifi/software/NRG_itho_wifi/src/IthoSystem.cpp line 168
0x400d9996: sendQueryDevicetype(bool) at /ithowifi/software/NRG_itho_wifi/src/IthoSystem.cpp line 647
0x400e6e26: init_i2c_functions() at 
/ithowifi/software/NRG_itho_wifi/src/task_syscontrol.cpp line 363
```

getSettingsLength should never end up in a core panic but somehow it does, I'll need to have a look at it why.

I will probably then again adapt a firmware file to hopefully successfully log some info. I'll need some time to fix this.

WouterJN, 2022-12-30:
Thanks in advance! No hurry :-)

WouterJN, 2022-12-30:
In the getSettingsLength code I saw the following?
`const struct ihtoDeviceType *ithoDevicesendPtr = ithoDevices + sizeof(ithoDevices) / sizeof(ithoDevices[0]);`

Can size of`(ithoDevices[0])` == 0? In that case you are trying to divide by zero.

arjenhiemstra, 2023-01-04:
> In the getSettingsLength code I saw the following; `const struct ihtoDeviceType *ithoDevicesendPtr = ithoDevices + sizeof(ithoDevices) / sizeof(ithoDevices[0]);`
> 
> Can the size of`(ithoDevices[0])` == 0? In that case you are trying to divide by zero and then the result is undefined.

That's just to determine the end of the const struct ihtoDeviceType ithoDevices[], the contents are hard coded and therefor sizeof will never be 0.

Could you try this firmware? 
[firmware.bin.zip](https://github.com/arjenhiemstra/ithowifi/files/10346218/firmware.bin.zip)

I hope this one will not crash (settings will not be working), I expect a log rule that looks something like this:
`2023-01-04 17:27:06 F: getSettingsLength dbg - deviceGroup:00, deviceID:1B, version:27
`

WouterJN, 2023-01-04:
Thnx :-)

After the update it did crash one more and after that it didn't crash anymore. The Itho status page is also working now! 👍  

Backtrace of the single crash
`Core  1 register dump:
PC      : 0x4008a407  PS      : 0x00060630  A0      : 0x800deed8  A1      : 0x3ffd02f0  
A2      : 0x8000733a  A3      : 0x8000733a  A4      : 0x00000fa0  A5      : 0x0000ff00  
A6      : 0x00000010  A7      : 0x3ffd0334  A8      : 0x80084250  A9      : 0x3ffd02e0  
A10     : 0x3ffee4e0  A11     : 0x00001800  A12     : 0x0000180c  A13     : 0x00000001  
A14     : 0x00000004  A15     : 0x3ffb6cb0  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x8000733a  LBEG    : 0x4008a475  LEND    : 0x4008a499  LCOUNT  : 0x800ee143  

Backtrace:0x4008a404:0x3ffd02f00x400deed5:0x3ffd0300 0x400def6b:0x3ffd0330 0x400df88d:0x3ffd0360 0x400ee15d:0x3ffd03a0 0x400eebd1:0x3ffd05c0 0x400f9c22:0x3ffd09c0 0x400fcf9f:0x3ffd09f0 0x400f9c36:0x3ffd0a40 0x400fc479:0x3ffd0a70 0x400fce49:0x3ffd0b50 0x400ea545:0x3ffd0be0 0x400ea61f:0x3ffd0c10 
`

Syslog (incl crash and device type)
```
2023-01-04 18:47:08 I: mDNS: started
2023-01-04 18:47:08 I: Hostname: nrg-itho-5d18
2023-01-04 18:47:13 E: I2C init: QueryDevicetype - failed
2023-01-04 18:49:53 F: getSettingsLength dbg - deviceGroup:00, deviceID:0D, version:9
2023-01-04 18:49:54 I: I2C init: QueryDevicetype - fw:9 hw:13
2023-01-04 18:49:54 I: I2C init: QueryStatusFormat - items:92
2023-01-04 18:49:54 I: I2C init: QueryStatus
      1257 I: System boot, last reset reason: POWERON_RESET
      1514 I: HW rev: NON-CVE 2, FW ver.: 2.4.4-beta4-test
      1707 I: I2C sniffer capable hardware: yes
      2941 E: Unable to set wifi disconnect
     11658 I: WiFi: connection successful
     12427 I: MQTT: connected, System config: 1
     12624 I: Webserver: started
     12978 I: mDNS: started
2023-01-04 18:50:20 I: Hostname: nrg-itho-5d18
2023-01-04 18:50:22 F: getSettingsLength dbg - deviceGroup:00, deviceID:0D, version:9
2023-01-04 18:50:23 I: I2C init: QueryDevicetype - fw:9 hw:13
2023-01-04 18:50:23 I: I2C init: QueryStatusFormat - items:92
2023-01-04 18:50:23 I: I2C init: QueryStatus
2023-01-04 18:50:51 I: System boot, last reset reason: OWDT_RESET
2023-01-04 18:50:51 I: HW rev: NON-CVE 2, FW ver.: 2.4.4-beta4-test
2023-01-04 18:50:51 I: I2C sniffer capable hardware: yes
2023-01-04 18:50:52 E: Unable to set wifi disconnect
2023-01-04 18:50:57 I: WiFi: connection successful
2023-01-04 18:50:58 I: MQTT: connected, System config: 1
2023-01-04 18:50:58 I: Webserver: started
2023-01-04 18:50:58 I: mDNS: started
2023-01-04 18:50:58 I: Hostname: nrg-itho-5d18
2023-01-04 18:51:04 F: getSettingsLength dbg - deviceGroup:00, deviceID:0D, version:9
2023-01-04 18:51:04 I: I2C init: QueryDevicetype - fw:9 hw:13
2023-01-04 18:51:05 I: I2C init: QueryStatusFormat - items:92
2023-01-04 18:51:05 I: I2C init: QueryStatus
```

arjenhiemstra, 2023-01-04:
Thanks! The crash seems unrelated to the issue.

```
2023-01-04 18:50:22 F: getSettingsLength dbg - deviceGroup:00, deviceID:0D, version:9 
2023-01-04 18:50:23 I: I2C init: QueryDevicetype - fw:9 hw:13 
2023-01-04 18:50:23 I: I2C init: QueryStatusFormat - items:92 
```

This is the info I was after, thanks!

That means the issue is in this part of the code:
https://github.com/arjenhiemstra/ithowifi/blob/9d2c1863b37e516c6ad822185a22952708988bc9/software/NRG_itho_wifi/src/IthoSystem.cpp#L157-L175

I'll analyse it further, thanks.

arjenhiemstra, 2023-01-04:
I think I found it. Would you be so kind to test this firmware? This deliberately does not solve it but has an extra check so hopefully doesn't crash the add-on. Makes it a lot easier to find might it ever happen again.
 
[firmware.zip](https://github.com/arjenhiemstra/ithowifi/files/10347826/firmware.zip)

This is most probably the source of the error:

The settings map for version 9 is here:
https://github.com/arjenhiemstra/ithowifi/blob/9d2c1863b37e516c6ad822185a22952708988bc9/software/NRG_itho_wifi/src/devices/wpu.h#L11

But I simply forget to copy that map into the mapping for all versions here:
https://github.com/arjenhiemstra/ithowifi/blob/9d2c1863b37e516c6ad822185a22952708988bc9/software/NRG_itho_wifi/src/devices/wpu.h#L648

It jumps from version 8 to version 10.
`itho_WPUsetting8,  nullptr,           itho_WPUsetting10`

arjenhiemstra, 2023-01-06:
2.4.4-beta4 should be the working version for your WPU incl. settings. (Still interested if above version works without crashes)

WouterJN, 2023-01-06:
> I think I found it. Would you be so kind to test this firmware? This deliberately does not solve it but has an extra check so hopefully doesn't crash the add-on. Makes it a lot easier to find might [...]

Thank you @arjenhiemstra. This version is working.

After testing this version I tested the official released v2.4.4-beta4 firmware. With that one I can read the settings!

Thank you for solving this.

closed 2023-01-06 by WouterJN