arjenhiemstra/ithowifi #128: Feature request

KetsOne, 2022-12-09:
Is it possible to add to the status page values 'calculated' by the addon using values from WPU?
For example: COP (delta_T over bron (K) * flow (liter/h) / 3600 * 4180 J/kg/K.

arjenhiemstra, 2023-01-05:
Are the values delta_T over bron (K) and flow (liter/h) available as status values on a WPU?

WouterJN, 2023-01-06:
@arjenhiemstra For FW 9;

delta_T= "Temp to source (°C)"-"Temp from source (°C)"
Flow = "Flow sensor (lt_hr)"

@KetsOne you can also calculate this yourself with HomeAssistant, Node-red or many more Home Automations platforms.

WouterJN, 2023-01-08:
@KetsOne I think you formula is wrong as some electric power is also being converted in thermal power, due that reason COP can only be calculated by outgoing delta_T * flow.

If that information is not available then a more accurate estimation is;
COP Heat pump = elektrisch vermogen/ (bronvermogen + elektrisch vermogen)

As per the first law of thermodynamics energy will not get lost, all the electric energy is being converted in heat. The heat pump it self is not heating up that much, so most of the heat will go to the condensor.

KetsOne, 2023-01-08:
@arjenhiemstra Thanks. As @WouterJN already mentioned, the values are already available in the status page of the add-on module.
@WouterJN Unfortunately i dont have HA. I would like to avoid having to install even more IT systems in the house :)
You are correct Wouter.

arjenhiemstra, 2023-01-09:
I'm afraid the firmware will get overly complex implementing these kind of features. There are lots of different itho devices to support with all their own firmware revisions. 
Better to implement this logic on the receiving end based on the data already available.

WouterJN, 2023-01-14:
Information for people who want to program this on the receiving end; I did some testing with this and this works completely unreliable as the temperature sensors are effected by the heat in the housing of the WPU. Especially the temperature sensor 'to source' can get affected by a few degrees. Only way to make this working is by adding extra external temp sensors.