arjenhiemstra/ithowifi #129: (WPU) setting (2410) type 0x01 seems wrong

tomkooij, 2022-12-10:
I am fixing #119. I'm compairing the info (from servicetool parameters, I guess) from: 
https://github.com/arjenhiemstra/ithowifi/issues/119#issuecomment-1342883561 with the settingspage:

Top: Settingpage non-CVE module. Bottom: info from servicetool.
[image]

All settings which have type `0x01` are off by a factor 10.
(`0x01` is treated as a special case -> uint8)
https://github.com/arjenhiemstra/ithowifi/blob/a5745c22f43495f5b4ff3ceb6dbc355727ae1a07/software/NRG_itho_wifi/src/IthoSystem.cpp#L1476-L1480

I believe it should be just a float constructed from an unsigned int divided by 10. 

I sniffed all types (hex) per setting number (decimal) for WPU 5G v37:
```
'0x1': [6, 7, 13, 41, 42, 51, 52, 63,74,  98, 137, 141, 142, 147, 148, 150, 153, 155, 167, 168, 169, 208, 209, 210, 221, 234],
```

tomkooij, 2022-12-10:
Datatype definities zijn overigens wel de vinden in de servicetool: https://pastebin.com/Ese3TP94

arjenhiemstra, 2022-12-10:
> Datatype definities zijn overigens wel de vinden in de servicetool: https://pastebin.com/Ese3TP94

Hmm, how much of this is source code that can actually be used? (ie. is it copy rights protected or open source, I think....)

arjenhiemstra, 2022-12-10:
https://github.com/rustyx/itho-esp/blob/main/Specs.md#message-a4-00-query-status-format

This is where most of the formatting info came from, but I cannot find the 0x01 case in there.

tomkooij, 2022-12-11:
> > Datatype definities zijn overigens wel de vinden in de servicetool: https://pastebin.com/Ese3TP94
> 
> Hmm, how much of this is source code that can actually be used? (ie. is it copy rights protected or open source, I think....)

This is certainly not open source. I’ll remove the link.

tomkooij, 2022-12-11:
> https://github.com/rustyx/itho-esp/blob/main/Specs.md#message-a4-00-query-status-format
> 
> This is where most of the formatting info came from, but I cannot find the 0x01 case in there. I might have added it later based on observations, it might also be a mistake.

I'm pretty sure `0x01` means:  "unsigned 1 byte value, divided by 10". It's probably an easy fix. If so, I'll implement and see what breaks.

arjenhiemstra, 2022-12-11:
You are probably right and also these value definitions are used not only for settings but also for the system status query as well (and maybe other places):
https://github.com/arjenhiemstra/ithowifi/blob/a5745c22f43495f5b4ff3ceb6dbc355727ae1a07/software/NRG_itho_wifi/src/IthoSystem.cpp#L696-L739

It might be best to refactor both parts of the bode to call 1 function that handles all status format cases, but any change improving the current implementation is also fine of course!

WouterJN, 2022-12-11:
@tomkooij Pls see #126 could be be that some other datatype's are also wrong? Could you please compare with the ServiceTool?

tomkooij, 2022-12-31:
WPU 5G fw37 setting 54 (datatype `0x91`) is also wrong:
[image]
Default should be -10. Clearly another signed/unsigned 2s-complement error. 
(Documenting this here so I can fix it)

PR #133 (2023-01-09): Fix settings
This tries to fix the settings pages for non-CVE devices
(Hopefully CVE devices as well, but I cannot test on CVE devices)

TODO:
 -  [x] Remove special case datatype `0x01`
 - [x] Implement "special case" datatype `0x0f` (Float divider 2).
 - [x] Fix signed floats. Cast ints to float according to datatype length (see: https://github.com/arjenhiemstra/ithowifi/issues/129#issuecomment-1368176583)
 - [x] Support unsigned raw bytes to float, make identical to #130 
 - [x] Test on WPU 5G fw37 fix #119 first
 - [x] Test on Autotemp fw10
 - [x] Test on CVE device (?)

Fixes #129 
Fixes  [... cut]
commit 6cb218c6 (2023-01-09): Merge pull request #133 from tomkooij/fix_settings

Fixes #129
Fixes #126
Fixes #119
closed 2023-01-09 by arjenhiemstra