arjenhiemstra/ithowifi #134: No label for device error (HRU 250-300)

desertspider, 2023-01-06:
**Describe the bug**
When updating from 2.4.3 to 2.4.4-beta4, on the "Itho status" page, i get no labels but the error "No label for device error". When migrating back to 2.4.3 the issue is resolved.

On the "itho settings" page the following information is shown:
Itho device type: HRU 250-300
Itho fw version: 8

The settings can be retrieved.

- Firmware version 2.4.4-beta4
- Hardware revision non-cve v2
- Type/model number HRU 300
- CC1101 RF module enabled [ no ]

arjenhiemstra, 2023-01-06:
That's most interesting, thanks for reporting. Can't find an immediate reason for this to happen but I've found a suspicious piece of code.  Only thing is that this part didn't change since 2.4.3. 
I'll look into it.

dennisvbussel, 2023-01-06:
I have a similar issue where the device status remains empty.
Query via the "Low Level Itho I2C commands does work".

EdwinGH, 2023-01-06:
I have the same issue on my HRU300, and had this with all the 2.4.4 betas (and re-installing 2.4.3 fixes the problem). 
Itho device type: HRU 250-300, Itho fw version: 7.
System log provides:
```
2023-01-06 16:33:16 I: I2C init: QueryStatusFormat - items:32
2023-01-06 16:33:16 I: I2C init: QueryDevicetype - fw:7 hw:1
...
2023-01-06 16:33:02 I: I2C sniffer capable hardware: yes
2023-01-06 16:33:02 I: HW rev: NON-CVE 2, FW ver.: 2.4.4-beta4
```
My Query Device type still provides 
```Result: 80 82 90 E0 05 09 00 01 07 01 06 07 00 00 00 6A```
And the Query Status Format and Query Status both fail...
Query 2410 provides:
```Result: 80 82 A4 10 05 13 00 00 02 BC 00 00 02 BC 00 00 04 4C 00 00 00 32 10 00 37 ED
Current: 700
Minimum value: 700
Maximum value: 1100```

arjenhiemstra, 2023-01-10:
please test if beta5 solves this issue, if not please reopen

closed 2023-01-10 by arjenhiemstra

arjenhiemstra, 2023-01-10:
> I have a similar issue where the device status remains empty. Query via the "Low Level Itho I2C commands does work".
> 
> 2.4.3 returns hex string: `8082A40005200000929211929292110F0F00000012120F000110101010109292100010100F1024` 2.4.4-beta4 returns a much longer hex string: `8082A401053400050534056206BF077E079405EA0A2AC7C700000052F87BF3C81400057800000000000007670A910A960000002B3E0000FF01FEC4`
> 
> Is the status actually queried or is it sniffed for the HR300?

These are results of 2 different requests (2400 and 2401), that explain the difference in length

desertspider, 2023-01-10:
> please test if beta5 solves this issue, if not please reopen

Beta 5 fixed the issue. Thank you!

arjenhiemstra, 2023-01-10:
Thanks for the feedback!
The funny thing is that I don't understand how it ever could have worked before, this code issue was there from the start....
Sometimes it's magic :)

dennisvbussel, 2023-01-11:
Beta 5 fixed this issue for me too :) Thanks a lot.

commit 03646862 (2023-01-10): Version 2.4.4-beta5

For testing purposes only!

Changes since 2.4.4-beta4:

- fix: Update status label retrieval to fix #134
- fix: (WPU) setting (2410) type 0x01 seems wrong #129
- fix: Autotemp - Cooling and Heating Setpoint not in Degrees Celsius #126
- fix: Negative values problem for WPU and HRU300 #123
- fix: setting labels for WPU v33 v34 v37 - WPU Heatpump settings incorrect #119
- fix: Syslog page keeps reloading: Websocket error #136

Firmware binary (CVE HW rev.2 and NON-CVE):
https://github.com/arjenhiemstra/ithowifi/raw/master/compiled_firmware_files/hardware_rev_2/nrgitho-hw2-v2 [... cut]
commit e8680439 (2023-01-10): Update status label retrieval to fix #134 Make syslog request async