arjenhiemstra/ithowifi #136: Syslog page keeps reloading: Websocket error

tomkooij, 2023-01-08:
**Describe the bug**
The syslog page keeps reloading over and over again. Other pages work fine.

**To Reproduce**
Steps to reproduce the behaviour:
I'm not sure it happens right after hard power cycle, but...
1. Connect to esp32 webserver
2. Click on 'Syslog' menu.
3. Wait
4. The syslog window will be black / empty at first
5. The page will reload when a (RANDOM) part of the syslog is shown in the window.
The page will reload over and over again.
[image]

The error in the Chrome debug console is either:
`WebSocket connection to 'ws://192.168.2.105:8000/ws' failed: Could not decode a text frame as UTF-8.`
Or:
`WebSocket connection to 'ws://192.168.2.105:8000/ws' failed: Invalid frame header`

In the console I see the error first, and only after the error the debug messages appear.
[image]

Syslog: https://pastebin.com/neYqQRVK
(it reloads on different parts of the syslog, so I does not seem be a particular syslog entry that causes this)

This is on `noncve 2.4.4-beta4`.
It appears on both WPU and Autotemp

arjenhiemstra, 2023-01-08:
I noticed the issue as well, if you delete the logfile the issue is probably gone (for a while)
I think the add-on is sending the log rules at a too high rate choking the websocket of overflowing javascript handling, not completely sure what happens.
Possible solutions to prevent this from happening, send per row (lots of overhead) or send bigger chucks (multiple rows) but I rather understand what is happening "under the hood"

tomkooij, 2023-01-09:
The fact that it keeps sending / reloading over and over again is also a problem, I guess.

It seems `$(document).ready( )` is called over and over again. A failure/error triggers another `ready()`. When I have time I'll see if it helps to use a `hasRun` variable or something to keep it from running more than once.

arjenhiemstra, 2023-01-09:
I've been poking around a bit with different solutions and I think for now the best course of action is to use something else than websocket to load the log. I've implemented a XMLHttpRequest to get the file content by URL and place it in the div.
The reason for moving away from this was instability in the ESPAsynWebserver lib but this is a bit leaner than before so I hope prevents the original issue.

arjenhiemstra, 2023-01-10:
Should be fixed with [2.4.4-beta5](https://github.com/arjenhiemstra/ithowifi/commit/0364686266e33c3d456d50c22d36ca0e9b24d1fa)

closed 2023-01-10 by arjenhiemstra