arjenhiemstra/ithowifi #137: Repeating stretches ("blank spots") of na status data

tomkooij, 2023-01-08:
**Describe the bug**
non-cve module lose i2c connection and/or has short periods where no status info is sent.
This can be a few seconds upto a few minutes.

**To Reproduce**
Steps to reproduce the behaviour:
1. Install 2.4.4-beta4.
(not seens on 2.3.5, but seen on every 2.4 beta so far).

All the white lines in this HA graphs are "lost connection" / no data:
[image]

**Expected behaviour**
[image]

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Device information**
- Firmware version 2.4.4.-beta4 
- Hardware revision non-cve 2
- Type/model number Itho  WPU 5G 
- CC1101 RF module enabled  no
- i2c debug menu turned off
- 
**Debug logging**
Nothing in logs, but rsyslog turned off.

This is really an issue with the ithowifi module and not de WPU. I investigated a "blank time" when the boiler was running:
[image]
Around 12:25 there is a blank spot with "Itho compressor active" going to 0. 
But the compressor is not really turned off: There is still electril power in the graph.

The boiler temp also shows a pause in data:
[image]
(There is also a "blank spot" around 12:15).

Is there a way to "log" / investigate these errors?

tomkooij, 2023-01-09:
I'll try and investigate this in the next few days. I just reverted back to `2.3.5` to be really sure it doesn't happen on that branch.

tomkooij, 2023-01-09:
I investigate a hiccup (documenting here, so I can have a look later)
[image]
Hiccup starting at  `17:30:52` until `17:30:57` and again at `17:31:19` until `17:31:44`

At `17:30:52` there is an MQTT message, which is zero after about 10 values, starting from  "Compressor Current"
```
2023-01-09T17:30:52+0100 : ithowpu/ithostatus : {"Outside temp (°C)":-1,
"Boiler temp down (°C)":51.49,"Boiler temp up (°C)":52.17,
[...]
"CV pressure (Bar)":1.71,
"Compressor current (A)":0,"Current e-element (A)":0
All zero from here.
```
(After that a normal MQTT message at `17:30:57`, so a single hiccup.)

I investigated the i2c sniff logs:
First hiccup: no A401 between 17:30:52 and :57
Second hiccup: several "normal" A401 replies in the interval (but i2c logs are cropped...)

tomkooij, 2023-01-11:
I added some `D_LOG()` statements to log the `queryStatus...` function: No more blank spots/hiccups... !??
I upgraded to `2.4.4-beta5`: no hiccups:
I downgraded to an "known bad firmware": no hiccups !?!??!!??

I don't know what is going on... The problem is magically gone.

The only config change I made was deleting most of the syslog, so the syslog menu would not keep reloading.
I did not even hard power cycle the ithowifi unit and/or touch the connector...

arjenhiemstra, 2023-01-11:
Who knowns, maybe it does return...
with beta5 I fixed a bug for HRU devices that was in there since the start, somehow with some versions it did work.... can't explain why, I used the word magic too :)

tomkooij, 2023-01-11:
This is probably not an i2c issue, since I saw no error when logging the i2c reply in the `sendQueryStatus...` function. 
Perhaps something in either MQTT or json is running out of memory.

dennisvbussel, 2023-01-11:
The HRU300 also suffers blank spots in the data received by HA:
[image]

arjenhiemstra, 2023-01-11:
> The HRU300 also suffers blank spots in the data received by HA while my WPU5G does not. They run the same firmware 2.4.4-beta5: [image]

I think you can then safely conclude that it is not a firmware issue. 
Normally I would think memory could be an issue. But the amount of memory a WPU takes for its status values and settings is bigger than the HRU300. So I would expect the WPU to have issues before the HRU300 has issues.

I would look into your cabling and possible interference

closed 2023-03-19 by stale[bot]