arjenhiemstra/ithowifi #143: Support for the ITHO HRU-3 GBL-001

galadril, 2023-01-19:
Would it be possible to add support for the ITHO HRU-3 GBL-001?

**Screenshots**
[image]
[image]

**Device information**
- Firmware version - 2.4.3
- Hardware revision - non-cve
- Type/model number Itho - ITHO HRU-3 GBL-001
- CC1101 RF module enabled - no

**Debug logging**
```
2023-01-19 09:02:38 I: HW rev: NON-CVE 1, FW ver.: 2.4.3
2023-01-19 09:02:44 I: WiFi: connection successful
2023-01-19 09:02:44 I: WiFi info:
2023-01-19 09:02:44 I: SSID:******| BSSID[*******]
2023-01-19 09:02:44 I: RSSI:-69dBm
2023-01-19 09:02:44 I: Mode:STA
2023-01-19 09:02:45 I: Status:WL_CONNECTED
2023-01-19 09:02:45 I: IP:*******
2023-01-19 09:02:45 I: Setup: Virtual remotes, start ID: DF,ED,E4 - No.: 1
2023-01-19 09:02:45 I: Setup: remotes configfile loaded
2023-01-19 09:02:25 I: MQTT: connected, System config: 1
2023-01-19 09:02:25 I: Webserver: started
2023-01-19 09:02:25 I: mDNS: started
2023-01-19 09:02:25 I: Hostname: nrg-itho-ede4
2023-01-19 09:02:26 I: Setup: done
2023-01-19 09:02:30 I: I2C init: QueryDevicetype - failed
```

arjenhiemstra, 2023-01-19:
Could you go to the debug page and click the button "Query devicetype" and post the result?

galadril, 2023-01-19:
Result: failed

galadril, 2023-01-19:
yes i tried with 2 cables

arjenhiemstra, 2023-01-19:
What is the length of the cable? 
And would it be possible to post some pictures? I'm not sure how your HRU is build but the one I have has a daughterboard inside that houses the connectors. Is that board accessible? If so, could you post a picture of that board (I might be able to trace some PCB tracks to find out if something is physically different on your HRU.

Power comes through the UTP cable (so you have no USB adapter connected I assume)?

galadril, 2023-01-19:
mmm no lights are coming on with only the UTP cable connected.. so i assumed it needed power as well
Ill make some pictures

galadril, 2023-01-19:
[image]
[image]
[image]

arjenhiemstra, 2023-01-19:
> mmm no lights are coming on with only the UTP cable connected.. so i assumed it needed power as well
> Ill make some pictures

No power should come through the RJ45 connector. I2C cannot travel long distances, power can. So if that is not available the issue is somewhere in your HRU. That could mean this RJ45 connector does not support the itho service tool (in that case the add-on won't work as well) or there is something inside the HRU that disables the port.

arjenhiemstra, 2023-01-19:
You model looks almost the same from the outside but maybe it's older from the inside and not yet working with the I2C protocol? Not sure at what year tho started adopting the Honeywell ramses-ii based I2C protocol.

galadril, 2023-01-19:
>  That could mean this RJ45 connector does not support the itho service tool (in that case the add-on won't work as well) or there is something inside the HRU that disables the port.

ai that sucks.. :(

dennisvbussel, 2023-01-19:
@galadril 
You could take a look in the manual [Eco Fan HRU 3](https://www.vansteenonline.nl/wp-content/uploads/ECO_FAN_HRU_3_installatie_handleding.pdf). Section 3-7 mentions that `Poort 2` is a connector that can be used to connect the HRU to a computer. Not sure if it is the i2c protocol required.

arjenhiemstra, 2023-01-19:
> @galadril You could take a look in the manual [Eco Fan HRU 3](https://www.vansteenonline.nl/wp-content/uploads/ECO_FAN_HRU_3_installatie_handleding.pdf). Section 3-7 mentions that `Poort 2` is a connector that can be used to connect the HRU to a computer. Not sure if it is the i2c protocol required.
> 
> Some information can be found regarding connector/protocol on the [circuitsonline](https://www.circuitsonline.net/forum/view/65868/1) forum.

Thanks for the manual and the link. I know the circuitsonline topic (look at the last post ;-)) but it could be that there are different versions of the ecofan 3 just like there are different versions of the CVE ECO 2. Some CVE ECO 2 versions have a 9 pin in a row header, some have to 2x4 pin header. It seems that the models with the 9 pins in a row use a different communication standard. That could also be the case here with the older HRU devices.

galadril, 2023-01-19:
What can i check in the device? Do your rever to making a custom cable?

arjenhiemstra, 2023-01-19:
First of all a photo of the internals of the HRU (PCB) is needed, best to have a picture of both sides of the PCB. That will help to trace the PCB lines to hopefully get a bit of insight into what lines are connected to where. It would also be nice to have a picture of chips on the board with the designators, this will help to find datasheets that describe the chip.
After that I hope to be able to share wether it is possible to solve things be an adapter cable or that a totally different solution is needed (in which case you have to find a newer HRU or a different solution to control this HRU).
Btw, no worries, you can always sent the add-on back might we not get it working.

galadril, 2023-01-19:
ok, i need to find how to get to the PCB then, maybe the screws behind the power/utp cables.
Thanks

galadril, 2023-01-22:
I don't really get how to come to the pcb for pictures. Any tips, or does anyone have the same type?

galadril, 2023-01-31:
Btw i asked Itho for their help/input about the PCD schematics, but they we're not helping and only awnsering:

_Dit is niet mogelijk met de HRU 3, de RJ45 aansluiting is alleen voor het uitlezen met een servicetool, wij hebben hier verder geen ondersteuning voor._

galadril, 2023-04-02:
Anyone with the same device?

closed 2023-06-15 by stale[bot]

arjenhiemstra, 2023-09-29:
In later models (hru ecofan) the connector block is secured with just 1 screw. If you remove the screw you can easily slide out the connector module. There is even a handle like shape for that.
Is something like this not present on your itho?

arjenhiemstra, 2023-09-29:
[image]

Like in the picture

galadril, 2023-09-30:
ok i could get it out.. (didnt use force at all).. but pushing the unit 2 cm back probably broke my wtw :(
After putting in the power it didnt do anything anymore :(

arjenhiemstra, 2023-09-30:
Oh no! That should not happen :( Maybe just a connector dislodged inside? 
I tried googling a bit and found this manual:
https://www.vansteenonline.nl/category/geen-categorie/itho/hru3/

If that type is the same, this might be of help

galadril, 2023-10-01:
I guess it's time for a new one. Probably the eco hru 3 which would then
work with your chip.

galadril, 2023-10-02:
@arjenhiemstra  the 'Itho WTW HRU 350 ECO' WtW unit should work out of the box without adjustments right?

arjenhiemstra, 2023-10-02:
Yes that unit works fur sure, out of the box

You weren't successful in reviving your HRU?

galadril, 2023-10-03:
nope the chip seems fried :(