arjenhiemstra/ithowifi #148: Frequest crashes/restarts along with I2C failures

YorickPeterse, 2023-01-30:
**Describe the bug**

Over the course of the last week or two I've been building some software to control our Demandflow/HRU ventilation setup, based on the changes introduced in https://github.com/arjenhiemstra/ithowifi/pull/144. During this time I've been observing two recurring issues:

1. I2C errors in the logs
2. The add-on crashing/restarting quite regularly 

Here's an example of some recent logs:

<details>

<summary>Logs</summary>

```
2023-01-30 13:15:01 I: I2C init: QueryDevicetype - fw:21 hw:11
2023-01-30 13:15:01 I: I2C init: QueryStatusFormat - items:75
2023-01-30 13:15:01 I: I2C init: QueryStatus
2023-01-30 13:00:01 I: System boot, last reset reason: OWDT_RESET
2023-01-30 13:00:01 I: HW rev: NON-CVE 2, FW ver.: 2.4.4-beta6
2023-01-30 13:00:01 I: I2C sniffer capable hardware: yes
2023-01-30 13:00:02 E: Unable to set wifi disconnect
2023-01-30 13:00:08 I: WiFi: connection successful
2023-01-30 13:00:09 I: Webserver: started
2023-01-30 13:00:09 I: mDNS: started
2023-01-30 13:00:09 I: Hostname: itho-wifi
2023-01-30 13:00:15 I: I2C init: QueryDevicetype - fw:21 hw:11
2023-01-30 13:00:15 I: I2C init: QueryStatusFormat - items:75
2023-01-30 13:00:15 I: I2C init: QueryStatus
2023-01-30 13:34:24 I: System boot, last reset reason: OWDT_RESET
2023-01-30 13:34:24 I: HW rev: NON-CVE 2, FW ver.: 2.4.4-beta6
2023-01-30 13:34:24 I: I2C sniffer capable hardware: yes
2023-01-30 13:34:25 E: Unable to set wifi disconnect
2023-01-30 13:34:31 I: WiFi: connection successful
2023-01-30 13:34:32 I: Webserver: started
2023-01-30 13:34:32 I: mDNS: started
2023-01-30 13:34:32 I: Hostname: itho-wifi
2023-01-30 13:34:38 I: I2C init: QueryDevicetype - fw:21 hw:11
2023-01-30 13:34:38 I: I2C init: QueryStatusFormat - items:75
2023-01-30 13:34:38 I: I2C init: QueryStatus
2023-01-30 13:21:28 E: i2c_master_init error: ESP_FAIL
2023-01-30 13:21:28 E: i2c_master_init error: ESP_FAIL
2023-01-30 13:21:28 E: i2c_master_init error: ESP_FAIL
2023-01-30 13:27:09 E: i2c_master_init error: ESP_FAIL
2023-01-30 13:27:09 E: i2c_master_init error: ESP_FAIL
2023-01-30 13:27:09 E: i2c_master_init error: ESP_FAIL
2023-01-30 13:20:29 I: System boot, last reset reason: OWDT_RESET
2023-01-30 13:20:29 I: HW rev: NON-CVE 2, FW ver.: 2.4.4-beta6
2023-01-30 13:20:30 I: I2C sniffer capable hardware: yes
2023-01-30 13:20:31 E: Unable to set wifi disconnect
2023-01-30 13:20:36 I: WiFi: connection successful
2023-01-30 13:20:37 I: Webserver: started
2023-01-30 13:20:37 I: mDNS: started
2023-01-30 13:20:38 I: Hostname: itho-wifi
2023-01-30 13:20:43 I: I2C init: QueryDevicetype - fw:21 hw:11
2023-01-30 13:20:43 I: I2C init: QueryStatusFormat - items:75
2023-01-30 13:20:44 I: I2C init: QueryStatus
2023-01-30 14:03:59 E: i2c_master_init error: ESP_FAIL
2023-01-30 14:03:59 E: i2c_master_init error: ESP_FAIL
2023-01-30 14:03:59 E: i2c_master_init error: ESP_FAIL
2023-01-30 14:07:39 E: i2c_master_init error: ESP_FAIL
2023-01-30 14:07:39 E: i2c_master_init error: ESP_FAIL
2023-01-30 14:07:40 E: i2c_master_init error: ESP_FAIL
2023-01-30 14:12:20 E: i2c_master_init error: ESP_FAIL
2023-01-30 14:18:00 E: i2c_master_init error: ESP_FAIL
2023-01-30 14:18:00 E: i2c_master_init error: ESP_FAIL
2023-01-30 14:18:01 E: i2c_master_init error: ESP_FAIL
2023-01-30 14:35:29 I: System boot, last reset reason: OWDT_RESET
2023-01-30 14:35:30 I: HW rev: NON-CVE 2, FW ver.: 2.4.4-beta6
2023-01-30 14:35:30 I: I2C sniffer capable hardware: yes
2023-01-30 14:35:31 E: Unable to set wifi disconnect
[... 116 lines cut ...]

I'll continue investigating this to see if I can find the cause.

**To Reproduce**

I'm not sure yet how to reproduce this, as I'm still trying to figure out what exactly causes these errors in the first place.

**Expected behaviour**

The add-on doesn't crash, and there are no I2C errors.

**Device information**
- Firmware version: 2.4.4-beta6
- Hardware revision: NON-CVE 2
- Type/model number Itho: HRU ECO 350 HR, paired with a DemandFlow/QualityFlow 
- CC1101 RF module enabled: no

**Debug logging**

See above.

**Desktop (please complete the following information):**
 - OS: Arch Linux
 - Browser: Firefox
 - Version: 108.0.2

arjenhiemstra, 2023-01-30:
If you have the ability to log serial content from the add-on that could give insight to the issues you experience.

When a crash occurs, a core dump is send through the serial port of the add-on. There is a ESP32 exception decoder that helps to identify the piece of code that is responsible for the crash (poor mans debugger but it's better than nothing)

Visual Studie Code + PlatformIO have the ability to monitor the serial and in case of a core dump immediately decode the message (you have to build the software version you are using before this works) 
If you want to do this asynchronous, you could install the Arduino IDE, build the software using this IDE and use the ESP32 exception decoder that you can install as an add-on to the IDE to achieve the same result.

arjenhiemstra, 2023-01-30:
And btw, with API, do you mean WebAPI or MQTTAPI?

YorickPeterse, 2023-01-31:
@arjenhiemstra Do I need any special hardware (e.g. a special cable) for that, or can I just connect the add-on to a laptop using the USB connection?

> And btw, with API, do you mean WebAPI or MQTTAPI?

The web API.

arjenhiemstra, 2023-02-01:
Just connect it with a regular usb micro cable to a laptop, that will suffice

> The web API.

That's what I thought. The ESPAsyncWebServer lib I use has some issues, especially with high flow. I already moved the websocket part to the Mongoose based webserver, the website itself poses some more challenges to port (but a lot has already been done and is lingering inactive in the current code base to be developed further after the I2C issues are finally done).

This is probably not the only reason things hang up, since the I2C sniffer code and queueing code has been added I've seen more reboots as well. Most have been solved but it might be that there are still issues.

arjenhiemstra, 2023-02-03:
Interesting, I found in de SDK documentation that there is a way to write the core dump to the flash partition so that it can be downloaded and analysed later! I have a PoC working and the core dump can be downloaded through the webinterface. 

Unfortunately this needs a change to the flash partition table, not sure if it is possible to do that with OTA only. Needs more testing but this might help a lot in finding bugs!

arjenhiemstra, 2023-02-03:
example output:

```
===============================================================
==================== ESP32 CORE DUMP START ====================

Crashed task handle: 0x3ffe9af4, name: 'async_tcp', GDB name: 'process 1073650420'

================== CURRENT THREAD REGISTERS ===================
exccause       0x1d (StoreProhibitedCause)
excvaddr       0x0
epc1           0x40109fcd
epc2           0x0
epc3           0x0
epc4           0x0
epc5           0x40082fb6
epc6           0x0
eps2           0x0
eps3           0x0
eps4           0x0
eps5           0x60120
eps6           0x0
pc             0x40083ee4          0x40083ee4 <panic_abort+24>
lbeg           0x40089edc          1074306780
lend           0x40089ee6          1074306790
lcount         0x0                 0
sar            0x10                16
ps             0x60e20             396832
threadptr      <unavailable>
br             <unavailable>
scompare1      <unavailable>
acclo          <unavailable>
acchi          <unavailable>
m0             <unavailable>
m1             <unavailable>
m2             <unavailable>
m3             <unavailable>
expstate       <unavailable>
f64r_lo        <unavailable>
f64r_hi        <unavailable>
f64s           <unavailable>
fcr            <unavailable>
fsr            <unavailable>
a0             0x8008dba0          -2146903136
a1             0x3ffe9740          1073649472
a2             0x3ffe97c4          1073649604
a3             0x2                 2
a4             0xa                 10
a5             0x0                 0
a6             0x4                 4
a7             0x3ffb6c8c          1073441932
a8             0x0                 0
a9             0x1                 1
a10            0x3ffe978f          1073649551
a11            0x3ffe978f          1073649551
a12            0x3ffe978f          1073649551
a13            0xff00              65280
a14            0xff0000            16711680
a15            0xff000000          -16777216

==================== CURRENT THREAD STACK =====================
#0  0x40083ee4 in panic_abort (details=0x3ffe97c4 "assert failed: void handleCoreCrash(AsyncWebServerRequest*) task_web.cpp:743 (0)") at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:402
#1  0x4008dba0 in esp_system_abort (details=0x3ffe97c4 "assert failed: void handleCoreCrash(AsyncWebServerRequest*) task_web.cpp:743 (0)") at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:128
#2  0x40093740 in __assert_func (file=<optimized out>, line=<optimized out>, func=<optimized out>, expr=<optimized out>) at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/assert.c:85
#3  0x400e452e in handleCoreCrash (request=0x3ffe9e4c) at src/task_web.cpp:743
#4  0x40189f92 in std::_Function_handler<void (AsyncWebServerRequest*), void (*)(AsyncWebServerRequest*)>::_M_invoke(std::_Any_data const&, AsyncWebServerRequest*&&) (__functor=..., __args#0=@0x3ffe98f8: 0x3ffe9e4c) at /Users/arjen/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/xtensa-esp32-elf/include/c++/8.4.0/bits/std_function.h:88
#5  0x400f35e8 in std::function<void (AsyncWebServerRequest*)>::operator()(AsyncWebServerRequest*) const (__args#0=<optimized out>, this=0x3ffe4ec8) at /Users/arjen/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/xtensa-esp32-elf/include/c++/8.4.0/bits/std_function.h:687
#6  AsyncCallbackWebHandler::handleRequest (this=0x3ffe4e80, request=<optimized out>) at .pio/libdeps/dev/ESP Async WebServer/src/WebHandlerImpl.h:132
#7  0x400f0ed8 in AsyncWebServerRequest::_parseLine (this=0x3ffe9e4c) at .pio/libdeps/dev/ESP Async WebServer/src/WebRequest.cpp:581
#8  0x400f0fe8 in AsyncWebServerRequest::_onData (this=0x3ffe9e4c, buf=0x3ffef98c, len=2) at .pio/libdeps/dev/ESP Async WebServer/src/WebRequest.cpp:123
[... 201 lines cut ...]
.coredump.tasks.data 0x3ffb50c4 0x158 RW 
.coredump.tasks.data 0x3ffb4ed0 0x1e0 RW 
.coredump.tasks.data 0x3ffd1798 0x158 RW 
.coredump.tasks.data 0x3ffd1400 0x394 RW 
.coredump.tasks.data 0x3ffb91dc 0x158 RW 
.coredump.tasks.data 0x3ffb36a0 0x214 RW 
.coredump.tasks.data 0x3ffb60d8 0x158 RW 
.coredump.tasks.data 0x3ffb5ed0 0x1f4 RW 
.coredump.tasks.data 0x3ffaef88 0x158 RW 
.coredump.tasks.data 0x3ffd8950 0x1b0 RW 
.coredump.tasks.data 0x3ffb7ce0 0x158 RW 
.coredump.tasks.data 0x3ffba4f0 0x1a0 RW 
.coredump.tasks.data 0x3ffb7a4c 0x158 RW 
.coredump.tasks.data 0x3ffb7870 0x1c8 RW 
.coredump.tasks.data 0x3ffe3820 0x158 RW 
.coredump.tasks.data 0x3ffe3620 0x1ec RW 
.coredump.tasks.data 0x3ffbcc5c 0x158 RW 
.coredump.tasks.data 0x3ffbcaa0 0x1a8 RW 
.coredump.tasks.data 0x3ffd8b14 0x158 RW 
.coredump.tasks.data 0x3ffdc720 0x1d8 RW 

===================== ESP32 CORE DUMP END =====================
===============================================================

```

arjenhiemstra, 2023-02-10:
Ok, I found a way to repartition and create the coredump partition during runtime and also backup the current config in a temporary location to be restored after reboot.
Upgrading should be quite smooth to the user with that. After that the last core dump is always downloadable for further analysis. I'll change the build in such a way that with every release also the debug file will be stored to make later core dump analysis possible. Downgrading involves an extra step but that is usually not needed and can be guided with some extra instructions.

I'm in active search for guinea pigs ;-)

YorickPeterse, 2023-02-10:
@arjenhiemstra I'd be happy to test those changes, as that sounds a lot easier compared to the alternative :smiley:

sanderkob, 2023-02-10:
Ik wil wel testen, mijn installatie met beta6 is redelijk stabiel maar
crasht veelvuldig wanneer ik de RFT auto remote gebruik. Lijkt op iets met
de communicatie tussen CC1101 en add-on.

Vriendelijke groet,
Sander Kobussen

GelsemaP, 2023-02-12:
guinea pig reporting for duty, sir

arjenhiemstra, 2023-02-12:
I'm testing the last few thing before I dare to post it here :) 
If things go wrong the device can be bricked with these changes. But I think I've tested all possible scenarios by now...

arjenhiemstra, 2023-02-12:
Here is the firmware file, this is the changelog compared to beta6:
[nrgitho-hw2-v2.4.4-beta7.bin.zip](https://github.com/arjenhiemstra/ithowifi/files/10717234/nrgitho-hw2-v2.4.4-beta7.bin.zip)

fix: link to previous log did not show
fix: OTA-update on modules without CC1101 module could lead to crash
fix: changes ESP restart logic to do a more graceful reboot
fix: simplified CC1101 ISR routine to one boolean
feat: crash information is now stored on flash and can be downloaded for further analysis later
feat: changes build to Arduino + ESP-IDF unlocking bare ESP-IDF features
feat: save ELF file during build to enable later crash analysis
feat: add reboot button to wifi config page

What to test:

What I'm most interested in for the short term is if the update went smooth and if the config was retained.
What happens is the following:
1. Firmware is updated
2. On first boot a backup of the current config is made to a different area flash
3. partition is rewritten
4. filesystem is cleared
5. reboot and restore config

The module should if everything goes well reconnect to your wifi network and on the debug page the file system size is now 128k instead of 192k (those 64k are now the area for the core dump). The core dump can also be downloaden through the debug page.

GelsemaP, 2023-02-12:
Worked like a charm. Didnt even notice the 2nd reboot. Well done.

Config version: 005
Itho I2C connection status: connected
File system: 20480 bytes used / 131072 bytes total
Remotes are still there. And can control from HA.

Should I already start testing the issue with the remote control causing an OWDT initiated reset?

2023-02-12 20:42:59 I: I2C init: Safe guard enabled
2023-02-12 20:42:59 I: I2C init: QueryStatus
2023-02-12 20:42:59 I: I2C init: QueryStatusFormat - items:12
2023-02-12 20:42:58 I: I2C init: QueryDevicetype - fw:27 hw:27
2023-02-12 20:42:56 I: Hostname: nrg-itho-741c
2023-02-12 20:42:55 I: mDNS: started
2023-02-12 20:42:55 I: Webserver: started
2023-02-12 20:42:55 I: MQTT: connected, System config: 1
2023-02-12 20:42:55 I: Setup: init of CC1101 RF module successful
2023-02-12 20:42:54 I: WiFi: connection successful
2023-02-12 20:42:46 E: Unable to set wifi disconnect
2023-02-12 20:42:45 I: I2C sniffer capable hardware: yes
2023-02-12 20:42:45 I: HW rev: 2, FW ver.: 2.4.4-beta7
2023-02-12 20:42:45 I: System boot, last reset reason: SW_RESET

arjenhiemstra, 2023-02-12:
Great, glad to hear! Yes you can start testing, I wasn’t able to reproduce the issue with the remote so if you can, please do. 
I checked in all the files in version control so the needed files are saved and usable for crash analysis.

sanderkob, 2023-02-12:
The update to nrgitho-hw2-v2.4.4-beta7 went smoothly, no problems. 
With the new firmware I did not see any crashes (OWDT or other) when using the RFT remote.
It seems the problem was solved, maybe the CC1101 ISR routine?

YorickPeterse, 2023-02-13:
@arjenhiemstra Are the mentioned changes available on GitHub somewhere (e.g. as a Git branch)? That way I can test them combined with those from https://github.com/arjenhiemstra/ithowifi/pull/144, as I can only reliably crash the system through usage of the API :smiley:

arjenhiemstra, 2023-02-13:
Not yet, I was waiting on some feedback here on the update proces

arjenhiemstra, 2023-02-13:
Commits have been pushed

arjenhiemstra, 2023-02-13:
btw, you need to build with PlatformIO (or CMake) to make this work, the Arduino IDE won't produce code that can update the partition table.

The PlatformIO build script creates and copies the ELF file you need for coredump analysis later. Binary and ELF files are tightly coupled together for this to work.
Using ESP-IDF tooling you can easily analyse the coredump using the following command:

`python ~/esp/esp-idf/components/espcoredump/espcoredump.py info_corefile --core coredump.bin --core-format raw firmware.elf`

GelsemaP, 2023-02-13:
Tested and all is working. No errors or crashes or whatsoever. Thanks. To me it sounds like we have a release candidate

sanderkob, 2023-02-14:
Just a question (not sure if this is the right place). The interrupt service routinie ITHOinterrupt() is clean now, only setting boolean ithoCheck. Is there a mechanism to temporarily disable a CC1101 interrupt while handling? Or is this not needed?

arjenhiemstra, 2023-02-14:
Yep that is possible. Just add:
`detachInterrupt(itho_irq_pin);`

before the code block where you want to have the ISR disabled

and:
`attachInterrupt(itho_irq_pin, ITHOinterrupt, RISING);`

to the end of the code block to reenable the interrupt again

sanderkob, 2023-02-14:
Should the `detachInterrupt` be included in the handling of ithoCheck, in task_cc1101.cpp in the function `void TaskCC1101(void*pvParameters)` after `if (ithoCheck)`?

arjenhiemstra, 2023-02-14:
> 

Depends on what you try to achieve with disabling the ISR.

sanderkob, 2023-02-14:
To prevent re-entry of the `void TaskCC1101(void*pvParameters)` function while it is still handling the interrupt. This may occur because of other CC1101 messages that arrive from other RF remotes. The boolean ithoCheck is cleared directly at the beginning of the handler function, so the handler function can be called while still processing. Or is there some mechanism preventing re-entry?

arjenhiemstra, 2023-02-14:
How will the handler function be called again? I think that cannot be the case, the function is first executed before the boolean is evaluated again.

sanderkob, 2023-02-14:
Correct, my oversight.

arjenhiemstra, 2023-02-14:
only thing I'm not sure about is if the buffer of the CC1101 can be overwritten by a newly received packet during readout by the ESP32 and if that is the case, what will happen then (probably a malformed packed)

arjenhiemstra, 2023-02-14:
Could very well be the case. ISR routines can be tricky on the ESP32. They need to be as short as possible and certain actions are not possible during an ISR.
In prior versions not only the boolean was set but also the buffer was readout. This should be very short but it is an action on the SPI bus so maybe this caused the issues.

YorickPeterse, 2023-02-14:
@arjenhiemstra I've been trying to build the new firmware using PlatformIO, but sadly without any luck. After pulling the latest changes, I ran this:

```
git submodule init
git submodule update
pio init
pio pkg install # pio init didn't seem to install all dependencies at first? Not sure
pio run
```

The end result is output such as this:

```
Compiling .pio/build/beta_noncve/components/esp_littlefs/src/littlefs_api.o
components/esp_littlefs/src/esp_littlefs.c:10:10: fatal error: littlefs/lfs.h: No such file or directory
 #include "littlefs/lfs.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
components/esp_littlefs/src/littlefs_api.c:12:10: fatal error: littlefs/lfs.h: No such file or directory
 #include "littlefs/lfs.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
Archiving .pio/build/beta_noncve/esp-idf/esp_local_ctrl/libesp_local_ctrl.a
Indexing .pio/build/beta_noncve/esp-idf/esp_local_ctrl/libesp_local_ctrl.a
Compiling .pio/build/beta_noncve/esp_netif/esp_netif_handlers.o
Compiling .pio/build/beta_noncve/esp_netif/esp_netif_objects.o
Compiling .pio/build/beta_noncve/esp_netif/esp_netif_defaults.o
Compiling .pio/build/beta_noncve/esp_netif/lwip/esp_netif_lwip.o
Compiling .pio/build/beta_noncve/esp_netif/lwip/esp_netif_lwip_defaults.o
Compiling .pio/build/beta_noncve/esp_netif/lwip/esp_netif_sta_list.o
*** [.pio/build/beta_noncve/components/esp_littlefs/src/esp_littlefs.o] Error 1
*** [.pio/build/beta_noncve/components/esp_littlefs/src/littlefs_api.o] Error 1

```

I have the following platforms installed:

```
NRG_itho_wifi $ pio platform list

WARNING: This command is deprecated and will be removed in the next releases. 
Please use `pio pkg list` instead.

Platforms
├── atmelavr @ 4.0.1 (required: platformio/atmelavr)
├── espressif32 @ 6.0.1 (required: platformio/espressif32)
└── espressif32 @ 5.3.0 (required: platformio/espressif32)
```

I tried uncommenting the LITTLEFS dependency in `platformio.ini`, but it makes no difference. I also tried to uncomment the line `platform                    = https://github.com/Jason2866/platform-espressif32.git#5a264890b306d569aabdd577cfa1a629c817ebdf #version of platform-espressif32 that links to ESP-IDF with LittleFS component included`, but that too made no difference.

Am I perhaps missing something (probably obvious)?

arjenhiemstra, 2023-02-14:
Hmm, I spend a whole day fixing this issue.... so easy and obvious is not what I would call it ;-)
Jason2866 his repo does include the needed littlefs component but it is in a different location, tried that, I could work but is not portable enough imho.

After a lot of trial and error I ended up with a solution that, at least I hoped, stays as close as possible to mainline repo's and is completely portable but clearly it needs some fixing.

The littlefs component is needed for the build to succeed, you cannot do without. In platformio.ini it is passed as extra build flag:
`-I components/esp_littlefs
` 

In this directory the submodule esp_littlefs should reside, maybe a recursive submodule update will solve it?

So esp_littlefs should have this path:

[]/software/NRG_itho_wifi/components/esp_littlefs

Also depicted here:
[image]

arjenhiemstra, 2023-02-14:
And otherwise, if you have ESP-IDF installed, the install description here could also help:
https://github.com/joltwallet/esp_littlefs

In the project directory (where platformio.ini resides) go the the pio terminal and execute:
`idf.py add-dependency joltwallet/littlefs==1.5.1`

YorickPeterse, 2023-02-15:
@arjenhiemstra The submodule is already there, and the commit it uses is `f2a949fbb303b93a0ae1cd47f078c49de96a59a7` ([this one](https://github.com/joltwallet/esp_littlefs/commit/f2a949fbb303b93a0ae1cd47f078c49de96a59a7)).

After setting up esp-idf and running said `idf.py add-dependency` command, I sadly still get the same errors. If I instead run `idf.py build` I'm greeted with the following error:

```
In file included from /home/yorickpeterse/Projects/c/ithowifi/software/NRG_itho_wifi/main/IthoRemote.cpp:1:
/home/yorickpeterse/Projects/c/ithowifi/software/NRG_itho_wifi/main/IthoRemote.h:9:10: fatal error: Arduino.h: No such file or directory
    9 | #include <Arduino.h>
      |          ^~~~~~~~~~~
compilation terminated.
[847/884] Building C object esp-idf/joltwallet__littlefs/CMakeFiles/__idf_joltwallet__littlefs.dir/src/littlefs/lfs.c.objninja: build stopped: subcommand failed.
HINT: Please make sure that the header name is correct. Also please check if you've specified all component dependencies with 'idf_component_register(REQUIRES ...)'. If the component is not present then it should be added by the IDF Component Manager. For more information run 'idf.py docs -sp api-guides/build-system.html'.
Also, please check if the header file has been removed, renamed or relocated - refer to the migration guide for more information.
ninja failed with exit code 1, output of the command is in the /home/yorickpeterse/Projects/c/ithowifi/software/NRG_itho_wifi/build/log/idf_py_stderr_output_246489 and /home/yorickpeterse/Projects/c/ithowifi/software/NRG_itho_wifi/build/log/idf_py_stdout_output_246489
```

Looking at the project, the header file `lfs.h` indeed is nowhere to be found. I tried cloning https://github.com/lorol/LITTLEFS into `components/littlefs` and adding `-I components/littlefs` to `platformio.ini` (similar to the other component), but sadly this didn't do the trick either :confused:

YorickPeterse, 2023-02-15:
Taking a closer look at `components/esp_littlefs`, for some reason the submodule at `src/littlefs` isn't checked out. After running `git submodule update --recursive` I'm now able to build the firmware without errors. :facepalm:

YorickPeterse, 2023-02-15:
I flashed the firmware on the WiFi module. I'll keep an eye on it in the coming days to see if/how often it crashes, and if so what the log file may contain to explain the crashes.

YorickPeterse, 2023-02-15:
Since flashing the new firmware I've had at least two crashes. I managed to get and decode the most recent core dump, the output is as follows:

```
===============================================================
==================== ESP32 CORE DUMP START ====================

Crashed task handle: 0x3ffdc2fc, name: 'async_tcp', GDB name: 'process 1073595132'

================== CURRENT THREAD REGISTERS ===================
exccause       0x1c (LoadProhibitedCause)
excvaddr       0x0
epc1           0x401179b9
epc2           0x0
epc3           0x0
epc4           0x0
epc5           0x40083e29
epc6           0x0
eps2           0x0
eps3           0x0
eps4           0x0
eps5           0x60120
eps6           0x0
pc             0x400d97ee          0x400d97ee <decodeQuery2410(int*, ithoSettings*, double*, double*, double*)+6>
lbeg           0x400014fd          1073747197
lend           0x4000150d          1073747213
lcount         0xfffffff2          4294967282
sar            0xe                 14
ps             0x60420             394272
threadptr      <unavailable>
br             <unavailable>
scompare1      <unavailable>
acclo          <unavailable>
acchi          <unavailable>
m0             <unavailable>
m1             <unavailable>
m2             <unavailable>
m3             <unavailable>
expstate       <unavailable>
f64r_lo        <unavailable>
f64r_hi        <unavailable>
f64s           <unavailable>
fcr            <unavailable>
fsr            <unavailable>
a0             0x800e83c4          -2146532412
a1             0x3ffe82a0          1073644192
a2             0x0                 0
a3             0x3ffde548          1073603912
a4             0x3ffe8308          1073644296
a5             0x3ffe8310          1073644304
a6             0x3ffe8318          1073644312
a7             0x5555aaaa          1431677610
a8             0x0                 0
a9             0x3ffe7b20          1073642272
a10            0x3f401d03          1061166339
a11            0x3f419d35          1061264693
a12            0x1f00984f          520132687
a13            0x3ffc3654          1073493588
a14            0x1f00984f          520132687
a15            0xff00ff00          -16711936

==================== CURRENT THREAD STACK =====================
#0  0x400d97ee in decodeQuery2410 (ptr=0x0, setting=0x3ffde548, cur=0x3ffe8308, min=0x3ffe8310, max=0x3ffe8318) at main/IthoSystem.cpp:1533
#1  0x400e83c4 in handleAPI (request=0x3ffdece8) at main/task_web.cpp:682
#2  0x40182e72 in std::_Function_handler<void (AsyncWebServerRequest*), void (*)(AsyncWebServerRequest*)>::_M_invoke(std::_Any_data const&, AsyncWebServerRequest*&&) (__functor=..., __args#0=@0x3ffe87c8: 0x3ffdece8) at /home/yorickpeterse/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/std_function.h:88
#3  0x400f1e14 in AsyncCallbackWebHandler::handleRequest(AsyncWebServerRequest*) ()
#4  0x400ef6b8 in AsyncWebServerRequest::_parseLine() ()
#5  0x400ef7c8 in AsyncWebServerRequest::_onData(void*, unsigned int) ()
#6  0x400ef9b0 in std::_Function_handler<void (void*, AsyncClient*, void*, unsigned int), AsyncWebServerRequest::AsyncWebServerRequest(AsyncWebServer*, AsyncClient*)::{lambda(void*, AsyncClient*, void*, unsigned int)#8}>::_M_invoke(std::_Any_data const&, void*&&, AsyncClient*&&, void*&&, unsigned int&&) ()
#7  0x400ec9c4 in AsyncClient::_recv(tcp_pcb*, pbuf*, signed char) ()
#8  0x400eca68 in AsyncClient::_s_recv(void*, tcp_pcb*, pbuf*, signed char) ()
[... 213 lines cut ...]
.coredump.tasks.data 0x3ffd4750 0x1bc RW 

===================== ESP32 CORE DUMP END =====================
===============================================================
```

Looking at the current thread stack it seems `decodeQuery2410()` is given a NULL pointer as its first argument:

```
==================== CURRENT THREAD STACK =====================
#0  0x400d97ee in decodeQuery2410 (ptr=0x0, setting=0x3ffde548, cur=0x3ffe8308, min=0x3ffe8310, max=0x3ffe8318) at main/IthoSystem.cpp:1533
#1  0x400e83c4 in handleAPI (request=0x3ffdece8) at main/task_web.cpp:682
#2  0x40182e72 in std::_Function_handler<void (AsyncWebServerRequest*), void (*)(AsyncWebServerRequest*)>::_M_invoke(std::_Any_data const&, AsyncWebServerRequest*&&) (__functor=..., __args#0=@0x3ffe87c8: 0x3ffdece8) at /home/yorickpeterse/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/std_function.h:88
#3  0x400f1e14 in AsyncCallbackWebHandler::handleRequest(AsyncWebServerRequest*) ()
#4  0x400ef6b8 in AsyncWebServerRequest::_parseLine() ()
#5  0x400ef7c8 in AsyncWebServerRequest::_onData(void*, unsigned int) ()
#6  0x400ef9b0 in std::_Function_handler<void (void*, AsyncClient*, void*, unsigned int), AsyncWebServerRequest::AsyncWebServerRequest(AsyncWebServer*, AsyncClient*)::{lambda(void*, AsyncClient*, void*, unsigned int)#8}>::_M_invoke(std::_Any_data const&, void*&&, AsyncClient*&&, void*&&, unsigned int&&) ()
#7  0x400ec9c4 in AsyncClient::_recv(tcp_pcb*, pbuf*, signed char) ()
#8  0x400eca68 in AsyncClient::_s_recv(void*, tcp_pcb*, pbuf*, signed char) ()
#9  0x400ed079 in _async_service_task(void*) ()
```

In this particular case the argument is the `resultPtr2410` global, assigned using `resultPtr2410 = sendQuery2410(idx, true);`.

Looking at the implementation of `sendQuery2410`, it returns `NULL` if the I2C command failed, or if no settings are present. These cases aren't handled in the API code, so I'll fix that and push the changes to my PR.

arjenhiemstra, 2023-02-15:
Nice work! I'll try to find out why the submodule isn't handled during clone or project setup

arjenhiemstra, 2023-02-15:
> In this particular case the argument is the `resultPtr2410` global, assigned using `resultPtr2410 = sendQuery2410(idx, true);`.

I see it! It should probably be something like:

```C++

bool decodeQuery2410(int32_t *ptr, ithoSettings *setting, double *cur, double *min, double *max)
{
    if(ptr == nullptr)
        return false;
//rest of code
}
```

YorickPeterse, 2023-02-15:
@arjenhiemstra The fix for the API that I'm currently testing is to just return an HTTP 500 if `resultPtr2410` is NULL, then in my ventilation code I simply retry such requests (after waiting a few seconds).

I haven't had any crashes over night with the new changes. I do notice a decent amount of I2C failures though, but I'm not sure if there's anything that can be done about that (might be an issue with the DemandFlow system?):

```
2023-02-15 11:55:39 E: i2c_master_init error: ESP_FAIL
2023-02-15 11:55:39 E: i2c_master_init error: ESP_FAIL
2023-02-15 11:55:38 E: i2c_master_init error: ESP_FAIL
2023-02-15 11:23:35 E: i2c_master_init error: ESP_FAIL
2023-02-15 11:23:35 E: i2c_master_init error: ESP_FAIL
2023-02-15 11:23:35 E: i2c_master_init error: ESP_FAIL
2023-02-15 07:21:25 E: i2c_master_init error: ESP_FAIL
2023-02-15 06:48:11 E: i2c_master_init error: ESP_FAIL
2023-02-15 06:39:30 E: i2c_master_init error: ESP_FAIL
2023-02-15 06:36:40 E: i2c_master_init error: ESP_FAIL
2023-02-15 06:13:37 E: i2c_master_init error: ESP_FAIL
2023-02-15 06:13:27 E: i2c_master_init error: ESP_FAIL
2023-02-15 06:12:07 E: i2c_master_init error: ESP_FAIL
2023-02-15 06:10:37 E: i2c_master_init error: ESP_FAIL
2023-02-15 06:10:36 E: i2c_master_init error: ESP_FAIL
2023-02-15 06:10:36 E: i2c_master_init error: ESP_FAIL
2023-02-15 06:06:27 E: i2c_master_init error: ESP_FAIL
2023-02-15 06:06:26 E: i2c_master_init error: ESP_FAIL
2023-02-15 06:06:26 E: i2c_master_init error: ESP_FAIL
2023-02-15 06:04:06 E: i2c_master_init error: ESP_FAIL
2023-02-15 06:04:06 E: i2c_master_init error: ESP_FAIL
2023-02-15 06:04:06 E: i2c_master_init error: ESP_FAIL
2023-02-15 04:21:03 E: i2c_master_init error: ESP_FAIL
2023-02-15 04:20:53 E: i2c_master_init error: ESP_FAIL
2023-02-15 03:59:01 E: i2c_master_init error: ESP_FAIL
2023-02-15 03:59:01 E: i2c_master_init error: ESP_FAIL
2023-02-15 03:59:00 E: i2c_master_init error: ESP_FAIL
2023-02-15 03:56:30 E: i2c_master_init error: ESP_FAIL
2023-02-15 03:56:30 E: i2c_master_init error: ESP_FAIL
2023-02-15 03:56:30 E: i2c_master_init error: ESP_FAIL
2023-02-15 02:37:04 I: I2C init: QueryStatus
2023-02-15 02:37:03 I: I2C init: QueryStatusFormat - items:75
2023-02-15 02:37:03 I: I2C init: QueryDevicetype - fw:21 hw:11
```

arjenhiemstra, 2023-02-16:
The add-on is switching between i2c master and slave all the time. The error indicated that the master init was not possible at that time. There can be multiple reasons for that looking at the master driver install code of ESP IDF:
https://github.com/espressif/esp-idf/blob/8ba96ab87389dd5400b7670a14356ea862022309/components/driver/i2c/i2c.c#L251-L437

arjenhiemstra, 2023-02-16:
I found a way to pipe the ESP IDF error logging to a different log handler.
I'll include an option to also log ESP IDF logging to the syslog, that way these kind of errors can be analysed easier.

arjenhiemstra, 2023-02-17:
Interesting; I had a OWDT resets here and there, not many but it happend. Now, all my test devices, completely stable for days in a row...
Seemingly those last changes (to the CC1101 isr routine?) made a noticeable difference!

For the build proces; I included a small python pre-build script that triggers on every PIO build updating the esp_littlefe submodule recursive from git. Hope that solves the issue you experienced.

GelsemaP, 2023-02-17:
Just to confirm. Haven’t had any issues. No resets. Everything working regardless if using virtual remote, a real remote paired to the addon or through home assistant. It really did seem to solve any issue I was experiencing. 
Only observation: there are no preset modes advertised through mqqt so can’t use the fan.preset service call. No biggie.

arjenhiemstra, 2023-02-17:
>Only observation: there are no preset modes advertised through mqqt so can’t use the fan.preset service call. No biggie.

Not sure if this is possible at all using HA Auto Discovery, I understood it is not.

arjenhiemstra, 2023-02-17:
2.5.0 has been released

https://github.com/arjenhiemstra/ithowifi/releases/tag/2.5.0

YorickPeterse, 2023-02-17:
Given my findings in https://github.com/arjenhiemstra/ithowifi/issues/148#issuecomment-1431254049, and me not having experienced any crashes since, I think this issue can be closed? :smiley:

closed 2023-02-17 by arjenhiemstra