arjenhiemstra/ithowifi #149: Is it possible to set humidity sensor thresholds

ymatsiuk, 2023-02-03:
**Describe the bug**
I noticed that humidity level is about 10% lower on the unit sensor than inside the bathroom.
So I wonder is it possible to lower the thresholds of the unit so it kicks in automatically earlier and stays up slightly longer?

**Device information**
Itho device type: CVE-Silent
Itho fw version: 27

arjenhiemstra, 2023-02-03:
Yes that is possible, have a look at the itho settings page and these settings:
`Rise in absolute humidity (X) at which it goes to high (mg_kg)`
`Minimum time for fan to go to high in RH control (min)`

btw, the actual humidity level is not important, it is the change in humidity that triggers the CVE to go to high. If you lower `Rise in absolute humidity (X) at which it goes to high (mg_kg)` it will turn on sooner, if you increase `Minimum time for fan to go to high in RH control (min)` it will stay on high for longer

ymatsiuk, 2023-02-03:
Thanks for clarification, just to recap, correct me if I'm wrong:

>Minimum time for fan to go to high in RH control (min)

increase if I want to run high for longer no matter what sensor reads (?)

>Rise in absolute humidity (X) at which it goes to high (mg_kg)

decrease if I want to start earlier, helps detecting lower humidity spikes (needs testing though since it can potentially cause unit go to high with now real reason 🤔 )

>Hysteresis X before falling back to normal operation (mg_kg)

decrease if I want it to switch from high earlier than minimum time (?)

>Maximum time it is allowed by the RH control to remain in high (min)

not sure why would I use this

Also why would I use these:
>Sampling time of the RH and temperature measurement (sec)
>Hysteresis RH for return to normal operation (%)

PS: I see these default values but I wonder where are they coming from? I'm asking because I noticed that "medium" setting defaulted to `120` and it was set to `~60` on my unit, so I assume that it was not read from the unit, but came out of the box with the add-on.

arjenhiemstra, 2023-02-03:
>decrease if I want it to switch from high earlier than minimum time (?)

this is the opposite of 'Rise in absolute humidity (X) at which it goes to high (mg_kg)'

>Maximum time it is allowed by the RH control to remain in high (min)

I think this ensures the fan will not stay in high mode longer than this setting

>Sampling time of the RH and temperature measurement (sec)

This is the time between sensor readouts, I would not change that, not sure what the impact is

>Hysteresis RH for return to normal operation (%)

same as 'Hysteresis X before falling back to normal operation (mg_kg)' I think but then based om relative humidity (%) instead of absolute humidity (mg_kg)

>PS: I see these default values but I wonder where are they coming from? I'm asking because I noticed that "medium" setting defaulted to 120 and it was set to ~60 on my unit, so I assume that it was not read from the unit, but came out of the box with the add-on.

Default values on the itho settings page come from the itho firmware. All other settings are settings of the add-on firmware itself.

ymatsiuk, 2023-04-06:
I just noticed the following behavior and re-opening to confirm if that's expected one:
My settings remain the same as described in this issue. I used a physical switch to switch to "high" mode. After few minutes I switched it back to "medium/auto" using the same physical switch. But now the unit doesn't kick-in automatically when the humidity raises. It seems like the add-on just lost control over the unit.
What can I do to prevent this from happening?

closed 2023-06-13 by ymatsiuk