arjenhiemstra/ithowifi #150: After a while I can no longer control the Itho unit.

ezwep, 2023-02-03:
**Describe the bug**
After a while I can no longer control the Itho unit. 

**To Reproduce**
That's tricky, I can't reproduce it very well yet. After I unplugged it, it started working again.

**Device information**
- Firmware version: 2.4.3 (2.4.4-beta6 immediately causes problems, do I have to create a separate issues for this?)
- Hardware revision: Itho CVE RFT WiFi add-on
- Type/model number Itho: CVE-S ECO SP

- CC1101 RF module enabled: yes

**Debug logging**
[logfile1.current.log](https://github.com/arjenhiemstra/ithowifi/files/10583367/logfile1.current.log)
[logfile0.log](https://github.com/arjenhiemstra/ithowifi/files/10583368/logfile0.log)
[Config.txt](https://github.com/arjenhiemstra/ithowifi/files/10583369/Config.txt)

arjenhiemstra, 2023-02-07:
First of all it is probably a good idea to lower the update frequency from 5 seconds to a higher value (whatever gives you high enough resolution of data to use), but at least 10 seconds is advisable.

What kind of issues do you experience with 2.4.4-beta6?

ezwep, 2023-02-08:
Hi Arjen,

I have adjusted the value to 10 seconds. Thanks for this tip.

This morning with version 2.4.3 I had the problem that the Itho did not respond to the mode set via the add-on.
I have set via my homey that as soon as the sensor in the bathroom measures a high humidity, it sets the Itho to high. This went well. Once the humidity is at a certain level, he should set the Itho to medium or low. But this did not happen. Even if the position was adjusted via the web interface, the Itho did not react to this. It was only after a while that the Itho went back to low and I was able to control it again.
I didn't see any error message in the add-on.
What could this be? Would the itho itself raise or lower the Itho based on the built-in sensor?
This situation has occurred 2x this morning after using the shower.

Just installed version 2.4.4-beta6. I2C safe guard enabled. Power off for a minute.
I can operate the Itho via the add-on. What doesn't work is:
- Temperature and humidity are not displayed
- Itho I2C connection status: init failed - please power cycle the itho unit -
Itho device type: Unknown device type
Itho fw version: 0

If there is anything I can do or need to provide additional information, please let me know. I'm happy to help.
I'm running the add-on on 2.4.4-beta6 for now.

And thanks in advance for your response.

ezwep, 2023-02-10:
Hi Arjen,

I saw your last comment: https://github.com/arjenhiemstra/ithowifi/issues/111#issuecomment-1424933528
Does this also apply to me?

arjenhiemstra, 2023-02-10:
Sorry, I missed the reaction of 2 day ago.

>What could this be? Would the itho itself raise or lower the Itho based on the built-in sensor?
Yes, you have itho with build in humidity sensor, that sensor will override other inputs (also from the add-on). As long as the humidity sensor is controlling the itho, the only way to influence the speed of the itho from the add-on is by using a virtual remote (just like a normal physical remote paired with the itho). Standard this can last up to an hour.
(there is a way to influence this changing some settings in the itho firmware)

>I can operate the Itho via the add-on. What doesn't work is:
So you were able to operate the itho? for how long?

>Does this also apply to me?
Not sure, the logging you provided is from version 2.4.3, in that version that function was not available yet. In the beta6 logging there should be a row stating if your add-on is sniffer capable or not.
ie.:
`2023-02-09 10:59:31 I: I2C sniffer capable hardware: yes`

ezwep, 2023-02-10:
> Sorry, I missed the reaction of 2 day ago.
>
>
> Yes, you have itho with build in humidity sensor, that sensor will override other inputs (also from the add-on). As long as the humidity sensor is [...]

I just reinstalled 2.4.4-beta6:
2023-02-10 13:53:31 E: I2C init: QueryDevicetype - failed
2023-02-10 13:53:26 I: Hostname: nrg-itho-68a4
2023-02-10 13:53:25 I: mDNS: started
2023-02-10 13:53:25 I: Web server: started
2023-02-10 13:53:25 I: MQTT: connected, System config: 1
2023-02-10 13:53:25 I: Setup: init of CC1101 RF module successful
2023-02-10 13:53:24 I: WiFi: connection successful
2023-02-10 13:53:19 E: Unable to set wifi disconnect
2023-02-10 13:53:18 I: I2C sniffer capable hardware: no
2023-02-10 13:53:17 I: HW rev: 2, FW ver: 2.4.4-beta6
2023-02-10 13:53:17 I: System boot, last reset reason: SW_RESET
2023-02-10 13:54:29 I: Setup: init of CC1101 RF module successful
2023-02-10 13:54:29 I: WiFi: connection successful
2023-02-10 13:54:23 E: Unable to set wifi disconnect
2023-02-10 13:54:21 I: I2C sniffer capable hardware: no

arjenhiemstra, 2023-02-10:
Thanks, that explains it I think.

Was 2.3.5 working stable?

ezwep, 2023-02-10:
With 2.3.5, right after the update, I started having the problem that it wouldn't automatically connect to my Wi-Fi. I've never had this happen once with newer versions.

I just don't understand what's going on with the humidity sensor yet. Do I understand correctly that I can pass the humidity sensor with a virtual remote? is it better to control my module via MQTT commands that then give the virtual remote commands?

arjenhiemstra, 2023-02-10:
Do you happen to have multiple access points with the same SSID?

>Do I understand correctly that I can pass the humidity sensor with a virtual remote? is it better to control my module via MQTT commands that then give the virtual remote commands?

If that works with a regular remote paired to your itho it will also work with the virtual remote.

'Better' strongly depends on what you want to achieve. I made my temp sensor of the itho very insensitive by altering settings in the itho firmware. That way I can control what happens with the measured hum value through my home domotica software. So I only the PWM2I2C commands through the MQTT API. There are also users that explicitly want keep the humidity control as it is and they use the virtual remote through the same MQTT API. It strongly depends on what you want to achieve. About all the possible ways to control the fan are in the add-on firmware nowadays. Thats great but also makes that it might be difficult to chose what to use.

About the i2c issues; if you have wifi issues with 2.3.5, have the add-on revision that doesn't support the i2c sniffer function and a fan that gives issues with newer firmware versions... it might be best to swap the add-on. I still hope to solve the i2c issues in a different way but I'm not sure this will work or how long this will take. The function 'i2c safe guard' which is based in the i2c sniffer seem to work very good in these situations so that may be the best solution for now.
Then you at least have a stable platform to try out what works best in your situation.

ezwep, 2023-02-15:
> Do you happen to have multiple access points with the same SSID?
No, 1 AP.

Thank you for your explanation.

Tonight I had the problem again that Itho was not responding to the wifi add-on. After a power reset it worked again.
The strange thing was that the Itho did not react by itself to the built-in sensor when showering. The position did not go to high.

I am thinking about ordering 2 new modules, but which type do I need? the standard or the one with CVE Optima with built-in sensor?

Here's the log:
     13017 I: MQTT: connection failed, System config: 1
     13163 I: Webserver: started
     13306 I: mDNS: started
     13611 I: Hostname: nrg-itho-68a4
     13729 I: Setup: done
     15287 I: I2C init: QueryDevicetype - fw:27 hw:27
     15617 I: I2C init: QueryStatusFormat - items:12
     15982 I: I2C init: QueryStatus
      2371 I: System boot, last reset reason: TG0WDT_SYS_RESET
      2671 I: HW rev: 2, FW ver.: 2.4.3
      8499 I: WiFi: connection successful
      8684 I: WiFi info:
      8900 I:   SSID:XXXXX | BSSID[B4:79:C8:0F:B2:A8]
      9182 I:   RSSI:-55dBm
      9457 I:   Mode:STA
      9582 I:   Status:WL_CONNECTED
      9722 I:   IP:192.168.178.40
      9878 I: Setup: Virtual remotes, start ID: 1A,68,A4 - No.: 1
     10082 I: Setup: remotes configfile loaded
     10268 I: Setup: init of CC1101 RF module successful
     10534 I: Setup: remotes configfile loaded
     15643 I: I2C init: QueryDevicetype - fw:27 hw:27
     16624 I: I2C init: QueryStatusFormat - items:12
     18524 I: I2C init: QueryStatus

arjenhiemstra, 2023-02-17:
You do not have a built-in CO2 sensor? If you don't have that, the regular CVE add-on will fit.

If you want me to change to ad-on so that it supports the safe guard functionality, that is also possible. It involves just soldering 2 extra wires.

mkruiver, 2023-04-19:
I think I have the same / similar issue:

ijn Itho add-on loopt soms vast op een gekke manier.
Symptomen:
- Itho reageert niet op speedsettings via de add-on webinterface
- Itho reageert soms wel & soms niet op high humidity van de interne sensor
- Itho reageert niet op externe RF remote
- Itho reageert wel op de virtual remote
- webinterface van de add-on blijft bereikbaar
- add-on reboot helpt niet
- powercycle helpt wel

Dit treedt op na een aantal dagen. Ik heb geen duidelijke aanleiding kunnen ontdekken.

Itho CVE (grijze deksel)
Add-on Rev2 met inmiddels firmware 2.5.6. Maar hetzelfde effect treedt op in 2.4.x en 2.5.x firmwares.

[logfile0.current.log.txt](https://github.com/arjenhiemstra/ithowifi/files/11270447/logfile0.current.log.txt)
[logfile1.log.txt](https://github.com/arjenhiemstra/ithowifi/files/11270449/logfile1.log.txt)
[coredump.zip](https://github.com/arjenhiemstra/ithowifi/files/11270456/coredump.zip)

0crap, 2023-04-22:
Few weeks ago I bought myself a wifi add-on and run it on the same box.

Device information:
- Firmware version: 2.5.6
- Hardware revision: Itho CVE WiFi add-on
- Type/model number: Itho CVE-S ECO SP (white lid, factory build-in humid sensor)
- CC1101 RF module enabled: yes

Until now it seems to run fine with the remotes connected to the add-on and only one virtual remote connected to the Itho box.
Only difference is I have the `Update frequency` still on the default value of 60.

arjenhiemstra, 2023-06-13:
@mkruiver and @0crap looking at your fan type (itho with built-in humidity sensor) you need to activate the I2C safe guard feature. This should prevent further lock-ups.

0crap, 2023-06-19:
> @mkruiver and @0crap looking at your fan type (itho with built-in humidity sensor) you need to activate the I2C safe guard feature. This should prevent further lock-ups.

In my case it was set to "auto". (Probably the default, which uses "on" I assume.)
I just set it to "on" to be sure.
Thx.

[image]

arjenhiemstra, 2023-07-13:
auto should equal on if your hardware supports it indeed, but better safe than sorry in this case. The log should also mention a confirmation.

closed 2023-09-20 by stale[bot]