arjenhiemstra/ithowifi #151: Every first command through remote causes OWDT_Reset, after reset remote it works

GelsemaP, 2023-02-11:
**Describe the bug**
I'm running beta 2.4.4-beta6 and have my remotes paired with the add-pm. When I want to control the fan (CVE S ECO SE) from the remote I notice that nothing happens with the fan. In the syslog I notice a OWDT_Reset. After the reset has happened the remote functions, one press and it works.
In Home Assistant nothing shows, everything remains working.

**To Reproduce**
Steps to reproduce the behaviour:
1. On a remote that is paired with the add-on press Timer (any button will do the same)
2. Wait a couple of minutes
3. Press again on a remote that is paired with the add-on any button
4. See error

**Expected behaviour**
I expect that when I press any button on the rf remote that the fan responds.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Device information**
- Firmware version 2.4.4-beta6
- Hardware revision (e.g. non-cve 1.1) 2
- Type/model number Itho (on sticker outside the box): cve-s eco se
- CC1101 RF module enabled [ yes, no ]: yes
- In case of wifi issues, wifi access point make and model:

**Debug logging**
Please provide debug logging from the debug page if possible:
2023-02-11 14:37:27 I: I2C init: Safe guard enabled
2023-02-11 14:37:26 I: I2C init: QueryStatus
2023-02-11 14:37:26 I: I2C init: QueryStatusFormat - items:12
2023-02-11 14:37:26 I: I2C init: QueryDevicetype - fw:27 hw:27
2023-02-11 14:35:19 I: Hostname: nrg-itho-741c
2023-02-11 14:35:19 I: mDNS: started
2023-02-11 14:35:19 I: Webserver: started
2023-02-11 14:35:18 I: MQTT: connected, System config: 1
2023-02-11 14:35:18 I: Setup: init of CC1101 RF module successful
2023-02-11 14:35:17 I: WiFi: connection successful
2023-02-11 14:35:12 E: Unable to set wifi disconnect
2023-02-11 14:35:11 I: I2C sniffer capable hardware: yes
2023-02-11 14:35:10 I: HW rev: 2, FW ver.: 2.4.4-beta6
2023-02-11 14:35:10 I: System boot, last reset reason: OWDT_RESET
2023-02-11 14:35:52 I: Hostname: nrg-itho-741c
2023-02-11 14:35:52 I: mDNS: started
2023-02-11 14:35:52 I: Webserver: started
2023-02-11 14:35:52 I: MQTT: connected, System config: 1
2023-02-11 14:35:52 I: Setup: init of CC1101 RF module successful
2023-02-11 14:35:51 I: WiFi: connection successful
2023-02-11 14:35:43 E: Unable to set wifi disconnect
2023-02-11 14:35:42 I: I2C sniffer capable hardware: yes
2023-02-11 14:35:42 I: HW rev: 2, FW ver.: 2.4.4-beta6
2023-02-11 14:35:41 I: System boot, last reset reason: OWDT_RESET
2023-02-11 14:34:46 I: I2C init: Safe guard enabled
2023-02-11 14:34:46 I: I2C init: QueryStatus
2023-02-11 14:34:46 I: I2C init: QueryStatusFormat - items:12
2023-02-11 14:34:46 I: I2C init: QueryDevicetype - fw:27 hw:27
2023-02-11 14:34:45 I: Hostname: nrg-itho-741c
2023-02-11 14:34:45 I: mDNS: started
2023-02-11 14:34:44 I: Webserver: started
2023-02-11 14:34:44 I: MQTT: connected, System config: 1
2023-02-11 14:34:44 I: Setup: init of CC1101 RF module successful
2023-02-11 14:36:20 I: WiFi: connection successful
2023-02-11 14:36:08 E: Unable to set wifi disconnect
2023-02-11 14:36:07 I: I2C sniffer capable hardware: yes
2023-02-11 14:36:06 I: HW rev: 2, FW ver.: 2.4.4-beta6
2023-02-11 14:36:06 I: System boot, last reset reason: OWDT_RESET
2023-02-11 14:37:03 I: Hostname: nrg-itho-741c
2023-02-11 14:37:03 I: mDNS: started
2023-02-11 14:37:03 I: Webserver: started
2023-02-11 14:37:03 I: MQTT: connected, System config: 1
2023-02-11 14:37:03 I: Setup: init of CC1101 RF module successful
2023-02-11 14:37:02 I: WiFi: connection successful
2023-02-11 14:36:54 E: Unable to set wifi disconnect
2023-02-11 14:36:53 I: I2C sniffer capable hardware: yes
2023-02-11 14:36:53 I: HW rev: 2, FW ver.: 2.4.4-beta6
2023-02-11 14:36:53 I: System boot, last reset reason: OWDT_RESET

Last command on rf:  lastcmd:0, lastcmdmsg:IthoUnknown

GelsemaP, 2023-02-12:
It’s easy to reproduce this. Just hit the timer button on a remote paired with the add-on. It crashes width the owdt reset. Immediately after reset the timer button works. However if you push another button afterwards the add-on resets again.

sanderkob, 2023-02-12:
Confirmed. It seems that something in the add-on happens, sometime after the reset. This makes that a CC1101 message to the add-on causes a watch-dog reset. There are no clues in the debug log or syslog.

GelsemaP, 2023-02-12:
When the remote is set to monitor only no issues occur. Tonight I will create some automagic that will monitor the lastcmd and manage it from ha. As a workaround.

GelsemaP, 2023-02-13:
Updated to latest beta (7) and issue does not appear anymore. I can use the remote again without any owdt resets. Thanks. Confirmed this can be closed.

closed 2023-02-13 by GelsemaP