arjenhiemstra/ithowifi #155: Co2 level output missing

mvaneijken, 2023-02-19:
**Describe the bug**
In version 2.3.5 there was a MQTT/API output that shows the current co2 level in ppm. I updated straight to v2.5.0 and in that release this sensor is missing. The sensor seems to be working, because "Highest CO2 concentration (ppm)" provides a value.

[image]

API request /api.html?get=ithostatus does not show this information also:

`{"temp":23.5,"hum":48.1,"ppmw":8791,"Ventilation setpoint (%)":-1,"Fan setpoint (rpm)":1182,"Fan speed (rpm)":1182,"Error":0,"Selection":7,"Startup counter":323,"Total operation (hours)":23397,"Absence (min)":0,"Highest CO2 concentration (ppm)":806,"Highest RH concentration (%)":49,"RelativeHumidity":"not available","Temperature":"not available"}`

**To Reproduce**
Steps to reproduce the behaviour:
1. Open http://<itho>/api.html?get=ithostatus, co2_le el_ppm missing. 

**Expected behaviour**
That the value that was provided before as nrg_itho_6804_co2level_ppm is available. 

**Device information**
- Firmware version: v2.5.0
- Hardware revision: 2 (modified to 2.5)
- Type/model number Itho (on sticker outside the box): Itho Daalderop CVE-S CO2 Optima Inside
- CC1101 RF module enabled: No
- In case of wifi issues, wifi access point make and model: N/A

**Debug logging**
Please provide debug logging from the debug page if possible:
`2023-02-18 15:19:17 W: I2C Safe guard disabled due to enabled i2c sniffer
      1368 I: System boot, last reset reason: TG0WDT_SYS_RESET
      1633 I: HW rev: 2, FW ver.: 2.5.0
      1902 I: I2C sniffer capable hardware: yes
      4961 E: Unable to set wifi disconnect
     10550 I: WiFi: connection successful
     12810 I: MQTT: connected, System config: 1
     12998 I: Webserver: started
     13209 I: mDNS: started
     13427 I: Hostname: nrg-itho-6804
     15338 I: I2C init: QueryDevicetype - fw:27 hw:27
     15647 I: I2C init: QueryStatusFormat - items:12
     16076 I: I2C init: QueryStatus
     16566 I: I2C init: Safe guard enabled
     17175 I: I2C init: i2c sniffer enabled
     17422 W: I2C Safe guard disabled due to enabled i2c sniffer
`

arjenhiemstra, 2023-02-19:
>I2C Safe guard disabled due to enabled i2c sniffer

Watch out for this one, the i2c safe guard function is disabled. If your i2c bus gets stuck, disable the sniffer

One to the issue. In 2.5.0 some i2c functions are disabled by default. Maybe you have to enable some of them to get back the co2 reading?

mvaneijken, 2023-02-19:
Thanks for the quick response! Based on your direction I've enabled the setting "Ventilation status (31DA)" and from that on the co2 level measurement came back. 

Ps. The IC2 sniffer is disabled now 😅.

closed 2023-02-19 by arjenhiemstra