arjenhiemstra/ithowifi #159: Version error newer than latest in firmware (HRU 250-300)

Nefkens, 2023-02-26:
Can you update the software for:
Itho device type: HRU 250-300
Itho fw version: 10

Thank you!

bokkoman, 2023-03-02:
Out of curiosity, how did you update to FW 10? I am on FW 9.

Nefkens, 2023-03-02:
> Out of curiosity, how did you update to FW 10? I am on FW 9. I have a HRU 300.

Well, i connected the itho with my laptop to the service tool and did a update.

@arjenhiemstra i can send you the latest parameters extracted from the service tool if you like.

arjenhiemstra, 2023-03-08:
Are there differences between FW 9 and FW 10 is you compare the status labels and settings labels?

bokkoman, 2023-03-11:
I'm getting this error now as well in the status page.

closed 2023-03-15 by arjenhiemstra

Nefkens, 2023-03-15:
Thank you for the update. I am now receiving messages.

On the Itho Settings page, i get this message: "Settings not available for this device or its firmware version"

arjenhiemstra, 2023-03-15:
2.5.4 is out

desertspider, 2023-03-15:
Thank you for the quick fix!

thijsdejong, 2023-05-22:
@arjenhiemstra It seems there is a problem with this: https://github.com/arjenhiemstra/ithowifi/issues/187

joppea2, 2023-08-01:
I am also on FW version 10 (HRU300 ECO duozone).
I can confirm the settings seem to be shifted starting at row 30.
29 is FrostValveOpenTimeFirstStartup (h),
30 in the current version shows NightMinVent1person (%).
30 should probably be FrostValveClosedTimeStartup (m), which is shown as index 109.
Everything that is now 30 and above should then be shifted downwards by one, so NightMinVent1person (%) should be index 31 etc.

commit 68a80a52 (2023-03-15): fix: add support for HRU250_300 fw10, fixes #159