arjenhiemstra/ithowifi #162: i2c bus stuck - please power cycle the itho unit - Firmware 2.5.2

semmtex, 2023-03-03:
**Describe the bug**
After a couple of days the i2c bus will on my WPU 5G will get stuck.
I use the MQTT server to get the stats into Home Assistant, nothing else so far...

Last I2C commands:
origin			start		duration	error				final
QUERY_STATUS	237459380	5515	I2C_ERROR_SCL_LOW	I2C_NOK
QUERY_STATUS	237399374	2		I2C_OK
QUERY_STATUS	237339365	2		I2C_OK
QUERY_STATUS	237279341	2		I2C_OK
QUERY_STATUS	237219339	2		I2C_OK
QUERY_STATUS	237159316	2		I2C_OK
QUERY_STATUS	237099304	2		I2C_OK
QUERY_STATUS	237039299	2		I2C_OK
QUERY_STATUS	236979275	2		I2C_OK
QUERY_STATUS	236919274	2		I2C_OK
QUERY_STATUS	236859260	2		I2C_OK
QUERY_STATUS	236799250	2		I2C_OK
QUERY_STATUS	236739226	2		I2C_OK
QUERY_STATUS	236679216	2		I2C_OK
QUERY_STATUS	236619216	2		I2C_OK

**To Reproduce**
Nothing. Just wait a few days.
Resetting using the TEMP/HUM Sensor says connecting.... connected .... stuck...

**Expected behaviour**
Do not get stuck.

**Device information**
- Firmware version 2.5.2.
- Hardware revision NON-CVE 1
- Type/model number Itho WPUG 5G
- No CC1101 RF module enabled

**Debug logging**
[logfile0.current.log](https://github.com/arjenhiemstra/ithowifi/files/10885450/logfile0.current.log)

arjenhiemstra, 2023-03-03:
You have i2c debug set to ON

What you describe is designed behaviour, also see the description right above the setting:
"This option enables an extra menu where more I2C options and logging is available. Enabling the I2C debug menu **will also halt I2C bus activity** in case of errors to be able to save logging!"

Are you actively debugging the I2C bus?  If not, just switch this feature off. Once in a while an error will always occur on the I2C bus, that is almost nature to how the protocol was designed. Normally this resolves itself.

semmtex, 2023-03-03:
Ah, good to know. I didn't know that it would behave like that. I thought it would simply give me the ability to debug any errors that might happen. I will try and see if this resolves the problem. Thanks...

arjenhiemstra, 2023-03-03:
I'll close the issue for now, if problems keep popping up, please reopen this issue

closed 2023-03-03 by arjenhiemstra