arjenhiemstra/ithowifi #163: RF remote function testing for HRU250-300

bokkoman, 2023-03-07:
Not a bug, but GH issue for testing develop fw for RF functions.
This fw is for HRU250-300, so that those people can send commands to the unit, since the virtual remote over i2c doesn't work.

Current fw has c1101 time outs.
Also seems to lock up/reboot, page becomes unresponsive and have to refresh page once or twice to get to the web interface again.

Is there a way to reset everything except wifi settings to default?

arjenhiemstra, 2023-03-14:
I have some difficulties reproducing the issues. Could you share your config file (you can of course clear all sensitive info), maybe I'm overlooking something or has it to do with some specific setup

bokkoman, 2023-03-15:
Can i grab this from the file editor? 

> the

arjenhiemstra, 2023-03-15:
> Can i grab this from the file editor?

Yes, those are the .json files

bokkoman, 2023-03-15:
This is my latest log file.
I tested it again on the "stable" 2.5.3. It worked, tried it twice. Then from Home assistant where I have a Restful command, i test it in the develop area. But after that the webui became unstable again and it reboots a few times.

```
2023-03-15 14:45:06 I: I2C init: QueryStatus
2023-03-15 14:45:05 I: I2C init: QueryStatusFormat - items:32
2023-03-15 14:45:05 I: I2C init: QueryDevicetype - fw:10 hw:1
2023-03-15 14:45:05 I: Hostname: nrg-itho-fc00
2023-03-15 14:45:05 I: mDNS: started
2023-03-15 14:45:04 I: Webserver: started
2023-03-15 14:45:04 I: MQTT: connected, System config: 1
2023-03-15 14:45:02 I: WiFi: connection successful
2023-03-15 14:44:56 E: Unable to set wifi disconnect
2023-03-15 14:44:53 I: I2C sniffer capable hardware: yes
2023-03-15 14:44:53 I: HW rev: NON-CVE 1, FW ver.: 2.5.4
2023-03-15 14:44:53 I: System boot, last reset reason: SW_RESET
2023-03-15 14:44:48 E: Setup: init of CC1101 RF module failed
2023-03-15 14:44:46 E: I2C init: QueryDevicetype - failed
2023-03-15 14:44:44 I: WiFi: connection successful
2023-03-15 14:44:38 E: Unable to set wifi disconnect
2023-03-15 14:44:35 I: I2C sniffer capable hardware: yes
2023-03-15 14:44:35 I: HW rev: NON-CVE 1, FW ver.: 2.5.4
2023-03-15 14:44:34 I: System boot, last reset reason: SDIO_RESET
2023-03-15 14:43:33 W: Warning: CC1101 Task timed out!
2023-03-15 14:43:24 I: I2C init: QueryStatus
2023-03-15 14:43:24 I: I2C init: QueryStatusFormat - items:32
2023-03-15 14:43:24 I: I2C init: QueryDevicetype - fw:10 hw:1
2023-03-15 14:43:24 I: Hostname: nrg-itho-fc00
2023-03-15 14:43:23 I: mDNS: started
2023-03-15 14:43:23 I: Webserver: started
2023-03-15 14:43:22 I: MQTT: connected, System config: 1
2023-03-15 14:43:19 E: I2C init: QueryDevicetype - failed
2023-03-15 14:43:18 I: Setup: init of CC1101 RF module successful
2023-03-15 14:43:15 I: WiFi: connection successful
2023-03-15 14:43:10 E: Unable to set wifi disconnect
2023-03-15 14:43:07 I: I2C sniffer capable hardware: yes
2023-03-15 14:43:07 I: HW rev: NON-CVE 1, FW ver.: 2.5.4
2023-03-15 14:43:07 I: System boot, last reset reason: SDIO_RESET
2023-03-15 14:42:05 W: Warning: CC1101 Task timed out!
2023-03-15 14:40:48 I: Hostname: nrg-itho-fc00
2023-03-15 14:40:48 I: mDNS: started
2023-03-15 14:40:48 I: Webserver: started
2023-03-15 14:40:47 I: MQTT: connected, System config: 1
2023-03-15 14:40:47 I: I2C init: QueryStatus
2023-03-15 14:40:46 I: I2C init: QueryStatusFormat - items:32
2023-03-15 14:40:46 I: I2C init: QueryDevicetype - fw:10 hw:1
2023-03-15 14:40:45 I: Setup: init of CC1101 RF module successful
2023-03-15 14:40:43 I: WiFi: connection successful
2023-03-15 14:40:37 E: Unable to set wifi disconnect
2023-03-15 14:40:35 I: I2C sniffer capable hardware: yes
2023-03-15 14:40:34 I: HW rev: NON-CVE 1, FW ver.: 2.5.4
2023-03-15 14:40:34 I: System boot, last reset reason: SDIO_RESET
2023-03-15 14:39:32 W: Warning: CC1101 Task timed out!
2023-03-15 14:39:22 I: I2C init: QueryStatus
2023-03-15 14:39:21 I: I2C init: QueryStatusFormat - items:32
2023-03-15 14:39:21 I: I2C init: QueryDevicetype - fw:10 hw:1
2023-03-15 14:39:21 I: Hostname: nrg-itho-fc00
2023-03-15 14:39:21 I: mDNS: started
2023-03-15 14:39:20 I: Webserver: started
2023-03-15 14:39:20 I: MQTT: connected, System config: 1
2023-03-15 14:39:20 I: Setup: init of CC1101 RF module successful
2023-03-15 14:39:17 I: WiFi: connection successful
2023-03-15 14:39:12 E: Unable to set wifi disconnect
2023-03-15 14:39:09 I: I2C sniffer capable hardware: yes
2023-03-15 14:39:09 I: HW rev: NON-CVE 1, FW ver.: 2.5.4
2023-03-15 14:39:09 I: System boot, last reset reason: SDIO_RESET
2023-03-15 14:38:06 W: Warning: CC1101 Task timed out!
2023-03-15 14:38:04 I: Hostname: nrg-itho-fc00
2023-03-15 14:38:03 I: mDNS: started
2023-03-15 14:38:03 I: Webserver: started
[... 57 lines cut ...]
2023-03-15 12:14:56 I: WiFi: connection successful
2023-03-15 12:14:51 E: Unable to set wifi disconnect
2023-03-15 12:14:48 I: I2C sniffer capable hardware: yes
2023-03-15 12:14:48 I: HW rev: NON-CVE 1, FW ver.: 2.5.3
2023-03-15 12:14:48 I: System boot, last reset reason: SDIO_RESET
2023-03-15 12:13:45 W: Warning: CC1101 Task timed out!
2023-03-15 12:13:42 I: Hostname: nrg-itho-fc00
2023-03-15 12:13:42 I: mDNS: started
2023-03-15 12:13:41 I: Webserver: started
2023-03-15 12:13:41 I: MQTT: connected, System config: 1
2023-03-15 12:13:38 I: I2C init: QueryStatus
2023-03-15 12:13:38 I: I2C init: QueryStatusFormat - items:32
2023-03-15 12:13:37 I: I2C init: QueryDevicetype - fw:10 hw:1
2023-03-15 12:13:37 I: Setup: init of CC1101 RF module successful
2023-03-15 12:13:34 I: WiFi: connection successful
2023-03-15 12:13:28 E: Unable to set wifi disconnect
2023-03-15 12:13:25 I: I2C sniffer capable hardware: yes
2023-03-15 12:13:25 I: HW rev: NON-CVE 1, FW ver.: 2.5.3
2023-03-15 12:13:25 I: System boot, last reset reason: SDIO_RESET
2023-03-15 12:12:23 W: Warning: CC1101 Task timed out!
2023-03-15 12:12:21 I: Hostname: nrg-itho-fc00
2023-03-15 12:12:21 I: mDNS: started
2023-03-15 12:12:21 I: Webserver: started
2023-03-15 12:12:20 E: MQTT: connection failed, System config: 1
```

bokkoman, 2023-03-15:
[configs.zip](https://github.com/arjenhiemstra/ithowifi/files/10983554/configs.zip)
These are my config files (and all the other files).

bokkoman, 2023-03-15:
Just did two RF api calls again, this is the log result. It reboots so many times.

```
2023-03-15 18:49:07 I: MQTT: connected, System config: 1
2023-03-15 18:49:07 I: Webserver: started
2023-03-15 18:49:07 I: mDNS: started
2023-03-15 18:49:07 I: Hostname: nrg-itho-fc00
2023-03-15 18:49:39 I: I2C init: QueryDevicetype - fw:10 hw:1
2023-03-15 18:49:39 I: I2C init: QueryStatusFormat - items:32
2023-03-15 18:49:39 I: I2C init: QueryStatus
2023-03-15 18:50:05 W: Warning: CC1101 Task timed out!
2023-03-15 18:51:06 I: System boot, last reset reason: SDIO_RESET
2023-03-15 18:51:06 I: HW rev: NON-CVE 1, FW ver.: 2.5.4
2023-03-15 18:51:06 I: I2C sniffer capable hardware: yes
2023-03-15 18:51:09 E: Unable to set wifi disconnect
2023-03-15 18:51:15 I: WiFi: connection successful
2023-03-15 18:51:17 I: Setup: init of CC1101 RF module successful
2023-03-15 18:51:18 E: I2C init: QueryDevicetype - failed
2023-03-15 18:51:22 I: MQTT: connected, System config: 1
2023-03-15 18:51:22 I: Webserver: started
2023-03-15 18:51:22 I: mDNS: started
2023-03-15 18:51:23 I: Hostname: nrg-itho-fc00
2023-03-15 18:51:23 W: Warning: CC1101 Task timed out!
2023-03-15 18:51:27 I: I2C init: QueryDevicetype - fw:10 hw:1
2023-03-15 18:51:28 I: I2C init: QueryStatusFormat - items:32
2023-03-15 18:51:28 I: I2C init: QueryStatus
2023-03-15 18:52:26 I: System boot, last reset reason: SDIO_RESET
2023-03-15 18:52:26 I: HW rev: NON-CVE 1, FW ver.: 2.5.4
2023-03-15 18:52:27 I: I2C sniffer capable hardware: yes
2023-03-15 18:52:29 E: Unable to set wifi disconnect
2023-03-15 18:52:35 I: WiFi: connection successful
2023-03-15 18:52:38 I: Setup: init of CC1101 RF module successful
2023-03-15 18:52:38 I: I2C init: QueryDevicetype - fw:10 hw:1
2023-03-15 18:52:38 I: I2C init: QueryStatusFormat - items:32
2023-03-15 18:52:38 I: I2C init: QueryStatus
2023-03-15 18:52:42 I: MQTT: connected, System config: 1
2023-03-15 18:52:43 I: Webserver: started
2023-03-15 18:52:43 I: mDNS: started
2023-03-15 18:52:43 I: Hostname: nrg-itho-fc00
2023-03-15 18:52:48 W: Warning: CC1101 Task timed out!
2023-03-15 18:53:49 I: System boot, last reset reason: SDIO_RESET
2023-03-15 18:53:49 I: HW rev: NON-CVE 1, FW ver.: 2.5.4
2023-03-15 18:53:50 I: I2C sniffer capable hardware: yes
2023-03-15 18:53:52 E: Unable to set wifi disconnect
2023-03-15 18:53:58 I: WiFi: connection successful
2023-03-15 18:54:01 I: Setup: init of CC1101 RF module successful
2023-03-15 18:54:03 E: I2C init: QueryDevicetype - failed
2023-03-15 18:54:05 I: MQTT: connected, System config: 1
2023-03-15 18:54:05 I: Webserver: started
2023-03-15 18:54:05 I: mDNS: started
2023-03-15 18:54:06 I: Hostname: nrg-itho-fc00
2023-03-15 18:54:32 W: Warning: CC1101 Task timed out!
2023-03-15 18:54:46 I: I2C init: QueryDevicetype - fw:10 hw:1
2023-03-15 18:54:47 I: I2C init: QueryStatusFormat - items:32
2023-03-15 18:54:47 I: I2C init: QueryStatus
2023-03-15 18:54:52 I: System boot, last reset reason: OWDT_RESET
2023-03-15 18:54:52 I: HW rev: NON-CVE 1, FW ver.: 2.5.4
2023-03-15 18:54:52 I: I2C sniffer capable hardware: yes
2023-03-15 18:54:55 E: Unable to set wifi disconnect
2023-03-15 18:55:01 I: WiFi: connection successful
2023-03-15 18:55:03 I: Setup: init of CC1101 RF module successful
2023-03-15 18:55:04 I: I2C init: QueryDevicetype - fw:10 hw:1
2023-03-15 18:55:04 I: I2C init: QueryStatusFormat - items:32
2023-03-15 18:55:04 I: I2C init: QueryStatus
2023-03-15 18:55:06 E: MQTT: connection failed, System config: 1
2023-03-15 18:55:06 I: Webserver: started
2023-03-15 18:55:06 I: mDNS: started
2023-03-15 18:55:07 I: Hostname: nrg-itho-fc00
2023-03-15 18:55:40 W: Warning: CC1101 Task timed out!
2023-03-15 18:56:41 I: System boot, last reset reason: SDIO_RESET
[... 31 lines cut ...]
2023-03-15 19:00:34 E: Unable to set wifi disconnect
2023-03-15 19:00:39 I: WiFi: connection successful
2023-03-15 19:00:42 I: Setup: init of CC1101 RF module successful
2023-03-15 19:00:42 I: I2C init: QueryDevicetype - fw:10 hw:1
2023-03-15 19:00:42 I: I2C init: QueryStatusFormat - items:32
2023-03-15 19:00:43 I: I2C init: QueryStatus
2023-03-15 19:00:44 E: MQTT: connection failed, System config: 1
2023-03-15 19:00:45 I: Webserver: started
2023-03-15 19:00:45 I: mDNS: started
2023-03-15 19:00:45 I: Hostname: nrg-itho-fc00
2023-03-15 19:02:24 W: Warning: CC1101 Task timed out!
2023-03-15 19:03:25 I: System boot, last reset reason: SDIO_RESET
2023-03-15 19:03:25 I: HW rev: NON-CVE 1, FW ver.: 2.5.4
2023-03-15 19:03:25 I: I2C sniffer capable hardware: yes
2023-03-15 19:03:28 E: Unable to set wifi disconnect
2023-03-15 19:03:34 I: WiFi: connection successful
2023-03-15 19:03:37 I: Setup: init of CC1101 RF module successful
2023-03-15 19:03:39 I: I2C init: QueryDevicetype - fw:10 hw:1
2023-03-15 19:03:40 I: I2C init: QueryStatusFormat - items:32
2023-03-15 19:03:40 I: I2C init: QueryStatus
2023-03-15 19:03:41 I: MQTT: connected, System config: 1
2023-03-15 19:03:41 I: Webserver: started
2023-03-15 19:03:42 I: mDNS: started
2023-03-15 19:03:42 I: Hostname: nrg-itho-fc00
```

arjenhiemstra, 2023-03-15:
Could you share the call you make from home assistant?

bokkoman, 2023-03-15:
> 

rest_command:
  afzuiging_30min:
    url: "http://nrg-itho-fc00.local/api.html?rfcommand=timer3"
  afzuiging_autonight:
    url: "http://nrg-itho-fc00.local/api.html?rfcommand=autonight"
  afzuiging_auto:
    url: "http://nrg-itho-fc00.local/api.html?rfcommand=auto"
  afzuiging_high:
    url: "http://nrg-itho-fc00.local/api.html?rfcommand=high"

These are the commands. I usually test it with rubbing the high first, but that already failed. It did work once.
But with that latest log i posted, i didn't use HA, those were just api calls in chrome.

arjenhiemstra, 2023-03-15:
Thanks! I'll setup a setup with the same settings and try to debug this

desertspider, 2023-03-20:
Maybe this not the place for this discussion but here it goes:

When decompiling the source code of the service tool. There is a whole section "hand bediening"  for manual controlling the Itho. Also in the access database there is a table "handbed" with also the entry "rpm". I conclude there must be a way in the service tool to manual control the Itho. If so, the speed can also controlled by i2c. The thing i dont know: What the address is. It must be somewhere in the code. Also what i don't know: I handbed is enabled, will the whole unit stop working automatically and must alle parameters be controlled externally?

I am not sure this is know knowledge or not. 

Does someone knows more on this topic?

I used "JustDecompile" to look into the code of the service tool.

bokkoman, 2023-04-16:
Since the latest update, it looks much more stable.

A new problem/bug i find, is that the `timer3` command only activated the 10min timer.

arjenhiemstra, 2023-04-16:
> A new problem/bug i find, is that the `timer3` command only activated the 10min timer.

That's strange... I've stepped through the code because I thought I might made a copy/past error somewhere but all checks out. The command being send look good and contains 30 als time value. (the only difference between command timer1, timer2 and timer3 is 1 byte that indicated the timer duration in minutes)
Does the timer2 command correctly run for 20 minutes?

bokkoman, 2023-04-16:
I'll try that now and report back.

bokkoman, 2023-04-16:
> That's strange... I've stepped through the code because I thought I might made a copy/past error somewhere but all checks out. The command being send look good and contains 30 als time value. (the [...]

Timer2 is ook 10min.

bokkoman, 2023-05-02:
Is there any update on this?
Just curious if the functionality of RF is going to be implemented in the MQTT as well. 
Or maybe a new HA script for controlling the HRU300 with API calls?

closed 2023-07-09 by stale[bot]