arjenhiemstra/ithowifi #164: non-CVE add on doesn't accept vremoteindex and/or vremotename via MQTT

blb4github, 2023-03-12:
**Describe the bug**
I have created 12 vremotes (index 0..11) in combination with my QF system. vremote commands work as expected via add-on website (page virtual remotes)and as well via web API but not via MQTT. Providing vremoteindex and or vremotename doesn't have the expected effect, the lastcmd MQTT message shows always "MQTT API-vremote-0"

**To Reproduce**
Steps to reproduce the behaviour (assuming vremote 3 is setup):
1. send MQTT message itho/cmd with JSON { "vremote": "high", "vremoteindex": "3"}
2. lastcmd gives as result {"source":"MQTT API-**vremote-0**","command":"high","timestamp":xxxxxxxxxx}
3. on itho status (page) the flow of room connected to vremoteindex 3 is not changed (expected 75 m3/h)
4. if this command is give via web-API or website the command is executed and the flow goes to 75 m3/h

**Expected behaviour**
the vremote should be recognised, with "vremoteindex": "3", the lastcmd output should be {"source":"MQTT API-**vremote-3**"

**Device information**
- Firmware version: 2.5.2
- Hardware revision: non-cve 1, batch 10-623
- Type/model number Itho: demandflow
- CC1101 RF module enabled: yes

**Debug logging**
Please provide debug logging from the debug page if possible:
I don't see debug option of MQTT messages.

arjenhiemstra, 2023-03-14:
I think I found the (stupid) error:
https://github.com/arjenhiemstra/ithowifi/blob/30542a76d9cb07e4ac196de45dde4ae5e34f0214/software/NRG_itho_wifi/main/tasks/task_mqtt.cpp#L368

0 must be index here

blb4github, 2023-03-14:
If I underderstand the code well it's always putting 0 in and not "index"? So that line should be ithoI2CCommand(**index**, command, MQTTAPI);? Will you generate a new version?

blb4github, 2023-03-15:
Thanks Arjen, it works now!
cmd: { "vremote": "high", "vremoteindex": "3"}
lastcmd: {"source":"MQTT API-vremote-3","command":"high","timestamp":1678872290}

closed 2023-03-15 by blb4github