arjenhiemstra/ithowifi #165: Feature request: Support PIR remotes

blb4github, 2023-03-13:
It would be great if also PIR remotes can be emulated. PIR remote can be used to ventilate the connected room (e.g. toilet) for a shorter time and with less flow (25 m3/h) than the default timer flow of 75 m3/h. A central PIR sensor can trigger away/at home status
[image]

bokkoman, 2023-03-17:
Can't you use a PIR sensor and just send a command for medium (as example) when said PIR sensor detects motion in toilet?
I don't really see the value of adding a remote that only send a medium command.

blb4github, 2023-03-17:
Maybe. I don't know how PIR is exactly implemented. There is some additional logic in the demandflow unit I guess as there is different behavior depending on the room type:
[image]

With the normal Timer30 function the flow goes always to 75 m³/h, with the PIR the flow goes to 25 m³/h.

The PIR sensor can aslo be connected as a central sensor and will function as "at home" or "away" sensor and put the system in the associated state. I don't know if this can be done on another way. If so it is not needed.

blb4github, 2023-03-17:
Currently I'm sending "Low" which gives 25 m³/h and after 15 minutes I send "Auto". 

And indeed, I'm trying to get a (second hand) PIR sensor.

bokkoman, 2023-03-17:
> Currently I'm sending "Low" which gives 25 m³/h and after 15 minutes I send "Auto".
> 
> And indeed, I'm trying to get a (second hand) PIR sensor.

Tip, buy it online. Grab the i2c commands. Send these to Arjen. Return the product within 14days. Profit.

arjenhiemstra, 2023-03-17:
The PIR sensor is not sold anymore unfortunately, so second hand seems the only option

bokkoman, 2023-03-17:
In that case, i guess your current workflow does exactly what you want. With that table, you can copy the workflow of the RF-PIR in HA or whatever home automation you use.

arjenhiemstra, 2024-01-11:
RFT-PIR device support should be working in upcoming 2.7.0 release

blb4github, 2024-01-11:
> RFT-PIR device support should be working in upcoming 2.7.0 release

Should it work in the current beta already? I did have upgrade to the beta and did have a quick look but didn't see this remote.

arjenhiemstra, 2024-01-11:
No sorry not yet. I forgot to check the stash in and include it in the beta. The commit can be found here:
https://github.com/arjenhiemstra/ithowifi/commit/5fa8d8ec12a398cbda59dd155e6b4b2ca1d2daea
and will be included in the next build.
With the commit you can checkout the code yourself and built a firmware file if you like.

arjenhiemstra, 2024-01-17:
Should be available with https://github.com/arjenhiemstra/ithowifi/releases/tag/Version-2.7.0

closed 2024-01-17 by arjenhiemstra