arjenhiemstra/ithowifi #169: Question about minimum speed (30%?)

rpardini, 2023-03-31:
Hello @arjenhiemstra, I got the addon last week, setup MQTT for my Home Assistant and all is working great, including remote. Thanks!

One difference I notice it that in the lowest speed possible (which seems to be "30%") the fan is now a bit stronger/louder than it was before the addon -- not by much, but surely is noticeable. It was a lot quieter before...

I've taped my (kitchen) speed selector to the left ("two-dots") so it can't be changed by accident. 

Here's some screenshots from the web interface -- I have _not_ changed anything in the Itho settings (yet).

Can you orient me on how to lower the minimum? Should I do it via the addon in some setting, or best to fiddle with potentiometer?

Thanks!

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

arjenhiemstra, 2023-03-31:
Please have a look at this wiki article:
https://github.com/arjenhiemstra/ithowifi/wiki/Controlling-the-speed-of-a-fan#help-fan-wont-go-lower-than-ie-30

rpardini, 2023-03-31:
Damn, guess I should have read more before posting. Super thanks!

closed 2023-03-31 by rpardini