arjenhiemstra/ithowifi #173: Auto Temp device type 30: Unkown device type

Reomishu, 2023-04-14:
Itho device type: Unkown device type
Itho fw version: 10
Retrieve settings result:
Message: Settings not available for this device or its firmware version

Itho status:
Itho I2C connection status: connected

Label Value
Nullptr error 0

Debug:
Device type Result: 80 82 90 E0 01 07 00 01 00 30 2E 0A 00 1D

Auto Temp Regelunit RF plus
Type 8 Groeps
01-04251
H04V46 2104

Laho19, 2023-10-18:
Exact same here with device type: Result: 80 82 90 E0 01 07 00 01 00 30 2E 0A 00 1D

tomkooij, 2023-10-29:
My guess is that Devicetype 0x30 is a regular autotemp. Perhaps the "Autotemp basic" set.  From the itho webpage the autotemp basic looks like just another autotemp unit and even the firmware byte (0x10) is the same as a regular recent autotemp.

I'll see if @Laho19 or @Reomishu can test/verify.

arjenhiemstra, 2023-10-30:
> My guess is that Devicetype 0x30 is a regular autotemp. Perhaps the "Autotemp basic" set. From the itho webpage the autotemp basic looks like just another autotemp unit just packaged with spider and even the firmware byte (0x10) is the same as a regular recent autotemp.
> 
> I'll see if @Laho19 or @Reomishu can test/verify.

if this is the case and the settings / status labels are the same across firmware versions, then the change is simply adding 1 row of code.... I can compile a version to try

Laho19, 2023-10-30:
> Is the device (kinda) new?

2022, so kinda

arjenhiemstra, 2023-10-30:
[nrgitho-dev-v2.6.0-autotemptest.bin.zip](https://github.com/arjenhiemstra/ithowifi/files/13210511/nrgitho-dev-v2.6.0-autotemptest.bin.zip)
here a version you could test that includes
```C
{0x00, 0x30, "AutoTemp Basic", ithoAutoTempSettingsMap, sizeof(ithoAutoTempSettingsMap) / sizeof(ithoAutoTempSettingsMap[0]), ithoAutoTempSettingsLabels, ithoAutoTempStatusMap, sizeof(ithoAutoTempStatusMap) / sizeof(ithoAutoTempStatusMap[0]), ithoAutoTempStatusLabels},

```

Laho19, 2023-10-31:
Works like a charm. Thanks!
[image]

arjenhiemstra, 2023-10-31:
Great! I'll include this change is the next version.
Strange that itho chose to have a different hardware number for basically the same device (at least as far as we know now), that's a first.

tomkooij, 2023-10-31:
Haha, I added the same fix and someone else (not on github) tested this as well. Confirmed to work as well.

I still think that this is the Autotemp basic (which is a preprogrammed autotemp that is sold packaged with an underflow heating valve unit and prepaired spiders), either sold as basic or sold stand-alone.

I'd be very suprised it this is new hardware.

Reomishu, 2023-11-01:
Thank you Arjen, I can also confirm it works.

closed 2023-11-01 by Reomishu

Laho19, 2023-11-02:
For me it works to fetch I2C info on (re)boot. But than the info is static. After rebooting the info is refreshed. Any idea? 

No new log entries after (re)boot. So cant debug. I2C status is connected.

[image]

arjenhiemstra, 2023-11-02:
what do you want to debug using the i2c debug function?

Laho19, 2023-11-02:
> what do you want to debug using the i2c debug function?

I just want to know why all the info is static and it issnt updating. Thought it could be the I2C is stuck?

arjenhiemstra, 2023-11-02:
One of the side effects of the debug function is halting the i2c bus. So I was wondering whether it was a cause or effect.

arjenhiemstra, 2023-11-02:
do the low level itho I2C commands work?

Laho19, 2023-11-02:
> do the low level itho I2C commands work? (on the debug page)

[image]

arjenhiemstra, 2023-11-02:
???
Almost looks like querying status is not allowed / blocked!?
@Reomishu are you getting a similar result?

Reomishu, 2023-11-02:
No problems here, values are getting updated and the low level itho I2C commands work.

arjenhiemstra, 2023-11-02:
Is there a setting in the list of itho settings that could explain this?

Laho19, 2023-11-02:
> Is there a setting in the list of itho settings that could explain this?

Cannot find any related one. Only valve, temp settings. Nothing comm related.