arjenhiemstra/ithowifi #174: Can't overrule setting with add-on

0crap, 2023-04-14:
**Describe the bug**
From the `System settings` page I read the following:
A virtual remote can be used to force the itho unit in medium mode before sending a command from the add-on. This way the add-on can overrule the current speed settings of the itho (ie. due to active input from a built in humidity sensor or another remote)

When I use the physical remote and set it to Low or High I can't overrule the setting with the add-on.

The physical remote is un-learned from the Itho box and added to the Add-on.
The System setting `Force medium/auto mode` is set to `on`.

**To Reproduce**
Steps to reproduce the behaviour:
1. Click Low on Physical remote
2. Go to WiFi controller webpage and use slider or buttons to change speed
3. Go to Domoticz and use Fan slider to change speed
4. nothing happens

**Expected behaviour**
Be able to change speed when physical remote is set to Low or High

**Screenshots**
not applicable

**Device information**
- Firmware version: v2.5.6
- Hardware revision: Itho CVE WiFi add-on v2
- Type/model number: Itho CVE-S ECO SP (white lid)
- CC1101 RF module enabled: yes
- In case of wifi issues, wifi access point make and model: -

**Debug logging**
Click for logfile [logfile1.current.log](https://pastebin.com/raw/KMb00S8q)

**Desktop (please complete the following information):**
 - OS: Windows 11 Pro
 - Browser, chrome
 - Version, 112.0.5615.86

**Smartphone (please complete the following information):**
 - Device: Samsung
 - OS: Android
 - Browser Stock
 - Version ?

0crap, 2023-04-14:
This is what the `Itho status` page shows when using the Physical remote on Low, Mid, High.

Low:
[image]

Mid:
[image]

High:
[image]

0crap, 2023-04-14:
And Debug Level2 pressing Low, Mid, High once on the Physical remote.

[image]

sanderkob, 2023-04-14:
Please make sure that I2C commands for
Ventilation status and Fan system status are enabled.
These settings are on the System settings page.

0crap, 2023-04-14:
Thx!
Those are off by default.
I changed them to `on` but unfortunately no change in behavior.
Still not able to override when Physical remote is set to Low or High.

[image]

arjenhiemstra, 2023-04-14:
If you go to the virtual remotes page of the add-on. Are you able to control the fan using the buttons of the first virtual remote?

0crap, 2023-04-14:
No, that is not possible.
There is by default already one Virtual Remote visible on index 0.
I have not done anything to it, assuming it is properly working by default.

arjenhiemstra, 2023-04-14:
Then that first remote is not joined correctly to the itho which is a prerequisite for this:
>The System setting Force medium/auto mode is set to on

To have the desired effect,  the first virtual remote must be joined correctly to your itho.

closed 2023-04-14 by 0crap

0crap, 2023-04-14:
@arjenhiemstra One more thing I would like to ask.
Before I had the add-on installed I was able to turn the Itho down, by pressing the Low button on the remote, when it kicked in automagically because of humidity.
With the add-on this is not possible.
Is this as intended?

When the humidity sensor gets dirty it often happens the Itho is running way too long after taking a shower in the bathroom.
We always solved it by pressing the Low button until I have the urge to clean the Itho and its humidity sensor.
Missing out on the Low button is a bit of a downside for us.
I'm using the remote as per wiki: [RFT Remote W (RFT bediening stand 1,2,3,T) (536-0124)](https://github.com/arjenhiemstra/ithowifi/wiki/Remotes#use-cases)

arjenhiemstra, 2023-04-14:
The low command can be send using the add-on in multiple ways, it uses the virtual remote function for this (which works just like a normal remote).

Do you use a home domotica system (like Home Assistant)? Using that you can completely control the behaviour of the itho, it is even possible to tune down the control of the humidity sensor so that the itho will not react by itself but only using the input from your domotica system (based on the sensor readout)
Another option is to have a simple push buttom somewhere that will online send the low command etc.

0crap, 2023-04-14:
Walks upstairs, turns on the hot tap to test some more. :-)
I have Domoticz connected via MQTT to the Add-on.
This gives only a fan slider and a temp/hum sensor.

After the humidity sensor puts the Itho full power I pressed the Low button on the Virtual remote webpage.
That works!
However pressing Low on the physical remote, mounted on the bathroom wall, keeps the Itho full speed. (When humidity still kicks the Itho full beans.)
When pressing the Physical remote Low, Medium, High, I do see the slider on the Add-on webpage and Domoticz follow perfectly.
It's just pressing the physical remote Low button does not slow down the Itho.
Only the webpage Virtual remote Low button does.

It would help indeed if I can send the Virtual button Low from Domoticz.
But that does not help with the wife and kids.
Would be perfect if the physical remote Low always works. (Like it did before the Add-on installation.)

roblomq, 2023-04-15:
When I read this, the main cause is in my opinion that the sensor gets "dirty". What do you mean with that, and how do you solve this?

0crap, 2023-04-15:
> When I read this, the main cause is in my opinion that the sensor gets "dirty". What do you mean with that, and how do you solve this?

Yep, just normal dust. Vacuum it out and use a paint brush to clean the sensor.

arjenhiemstra, 2023-04-15:
> Walks upstairs, turns on the hot tap to test some more. :-) I have Domoticz connected via MQTT to the Add-on. This gives only a fan slider and a temp/hum sensor.
> 
> After the humidity sensor puts the Itho full power I pressed the Low button on the Virtual remote webpage. That works! However pressing Low on the physical remote, mounted on the bathroom wall, keeps the Itho full speed. (When humidity still kicks the Itho full beans.) When pressing the physical remote Low, Medium, High, I do see the slider on the Add-on webpage and Domoticz follow perfectly. It's just pressing the physical remote Low button does not slow down the Itho. (When still in humid condition.) Only the webpage Virtual remote Low button does.
> 
> It would help indeed if I can send the Virtual button Low from Domoticz. But that does not help with the wife and kids. Would be perfect if the physical remote Low always works. (Like it did before the Add-on installation.)

There are a few ways to approach this, mine is to control the itho only by 0-100 pwm2i2c protocol (so no (virtual) remotes commands at all). As said, logic of the humidity (and in my case CO2 sensor) is done in the home domotica software space.

The other option is to control the itho using (virtual remote) commands, this can be done through the add-on's CC1101 module or having remotes joined to the itho, what you prefer. The down side, the speed status is not available in your home domotica software by default. Domoticz is even more limited because of its MQTT topic structure but this can be overcome using the WebAPI.
On the MQTT topic itho/ithostatus or through the WebAPI there is a topic FanInfo (you might need to activate some i2c functions for this to be available) which shows the remote status (ie. low, medium high etc), you can use this in you own integration. There is most of the times also a topic "Fan setpoint (%)" which shows the speed of the fan (independant wether this speed was set using the pwm2i2c protocol (slider on the web page of the add-on) or a (virtual) remote. You can use this to make a status slider in ie. domoticz to show the current speed.

These are just a few examples what you could do, I have seen other examples of creative solutions as well. Just keep in mind which control options you have and what the limitations of those are.
Some extra info can be found here:
https://github.com/arjenhiemstra/ithowifi/wiki/Controlling-the-speed-of-a-fan

0crap, 2023-04-15:
Thx, I appreciate your reply.

For other family members it's most convenient if the physical remote just works as it did before.
Any explanation why it is not possible to make the Low button not override when the Itho is in humid condition?
It seems/feels to me the Add-on is technically capable of doing it...

(Of course I can learn them to the Itho box again if not possible.)