arjenhiemstra/ithowifi #178: [feature request] give correct slider position when auto humid is active

0crap, 2023-04-22:
On my Itho CVE-S ECO SP, with factory build-in humidity sensor, the fan goes automagically to High when it sees a jump in humidity.
Perfectly fine.
However, the Add-on does not update the slider position on the main webpage.
(And also does not update MQTT info.)

However, I do notice the Itho status page shows the correct `Ventilation setpoint (%)`.
[image]
The webAPI `/api.html?get=ithostatus` is also able to provide the correct info for `"Fan setpoint (rpm)"` and `"Fan speed (rpm)"`.

It seems to me it should be possible to use this info to correctly adjust the slider position and update MQTT when the humid status is active and controlling the fan.
Can that be done?

0crap, 2023-06-22:
I don't make the time and effort of creating an issue to let it be closed by a bot.
At least a remark from development is appreciated.

svrooij, 2023-07-11:
I'm also wondering this. We have the build-in humidity version and it ramps up when we shower, but it does not show up correctly in home assistant.

arjenhiemstra, 2023-07-13:
The difficulty is that the slider is input for the PWM protocol that itho uses (values between 0-255) and the ventilation setpoint is the result of that command (0-100%). 
I can have a look if the slider can be updated based on the ventilation setpoint info but I'm not sure if that info is available for all itho firmware versions.

In the mean time, the ventilation setpoint is available on MQTT, you could use that information to update the status of the slider in ie. Home Assistant (you probably have to add the config manually then but you can use the auto discovery config as a template)

svrooij, 2023-07-20:
If you receive a new value for the setpoint you can do:

```cpp
slider = round(setpoint * 2.55);
```

The auto-detected fan control in home assistant would also benefit from this. Now the level is stuck at the level that was set, not the level it changed itself to because it's humid (in our situation with a humidity sensor built-in).

remcom, 2024-02-26:
I would also love to see the slider correctly update when it is on auto.