arjenhiemstra/ithowifi #179: feature request: power consumption of fan (W)

menno-smit, 2023-04-22:
Currently the itho status shows:  "Current consumption of fan (mA)".
It would be great to also show:  "Power consumption of fan (W)"

This should be easy to calculate (mA / 1000 * fan_voltage), however does anyone know what the fan voltage is? Does it vary between models? I have a HRU ECO 300.

arjenhiemstra, 2023-04-25:
If there is no voltage status shown it is unknown. The only thing you then can do is read the voltage yourself (ie. with p1 info out of your smart meter) and calculate the watt value with that figure.

bokkoman, 2023-05-02:
If you have one of those plugs that show the current usage on a wall socket, than you can probably calculate it as well?
But my guess is it runs at 24V.
The max power is 126W at full usage, source:
https://ithodaalderop.compano.com/Data/Environments/000001/Attachment/Bijlage/A04_Ventilatie/A04_02_Gebalanceerd/A04_02_01_HRU/A04_02_01_02_250_300/B01_06_MTE/01-04423-004%20%20Handleiding%20HRU%20ECO%20300.pdf

closed 2023-07-09 by stale[bot]