arjenhiemstra/ithowifi #185: No response for percentages below 6%

M66B, 2023-05-11:
**Describe the bug**
Going from 100% to 5% results in still running at 100%.
Going from 100% to 6% results in running at 6%.

**To Reproduce**
Using the default slider in home assistant (MQTT), see screenshots.

**Expected behaviour**
Either applying the low percentage or turning off if this is not possible.

**Screenshots**
100 to 5%: https://imgur.com/a/fXHNT4r
100 to 6%: https://imgur.com/a/ZAATR3W

**Device information**
- Firmware version: 2.5.6
- Hardware revision: 2
- Type/model number Itho CVE ECO RFT SP versie: 545-5036-001
- CC1101 RF module enabled: no

**Debug logging**
https://pastebin.com/Hf4DhVT0

M66B, 2023-05-11:
In addition: the ventilation setpoint doesn't match the slider.

arjenhiemstra, 2023-06-13:
This depends on your itho model. It seems that most models do not change the motor speed anymore below a certain value.
The protocol supports values between 0 and 255 to control the fan speed. The itho firmware decides what to do with this input. This can be influenced to a certain extend by changing itho firmware settings but with my fan anything below 9 will not change the fan speed anymore.

Without complex per fan type / settings combination logic in the add-on firmware this is a itho firmware limitation that we have to life with I'm afraid.

M66B, 2023-06-13:
@arjenhiemstra Isn't it an idea to add an option for this? "*Minimum fan speed*" or something?

closed 2023-08-21 by stale[bot]