arjenhiemstra/ithowifi #188: Unit does no longer react on changes in fan speed

mjpblaauw, 2023-05-25:
Since a few weeks i noticed that my unit does no longer react on changes when i use the slider in the UI to change the Fan speed. It also does not react on this when i send MQTT commands.

It only reacts in the UI when i push the buttons Low, Medium or High.

Is this a known issue? I run the latest firmware release 2.5.6

arjenhiemstra, 2023-06-13:
Almost always this has to do with the active mode of you fan. I would expect that slider to work again after you hit the Medium button. Could you test this?
See also here for more information:
https://github.com/arjenhiemstra/ithowifi/wiki/Controlling-the-speed-of-a-fan#help-a-speed-command-does-not-work

mjpblaauw, 2023-06-18:
Hi Arjen,

 

Ik heb de test gedaan en hij doet het weer. Echter niet door de medium button te activeren maar meer die op physieke zender.

 

Dank voor je antwoord nogmaals.

 

Groeten,

 

Marcel

 

https://github.com/arjenhiemstra/ithowifi/wiki/Controlling-the-speed-of-a-fan#help-a-speed-command-does-not-work

Bodmoor, 2023-07-09:
Hello,

I have the same issue. I tried the physical button and the medium button in the interface, to no avail. I need to reset the unit by disconnecting the power for ~10 seconds before turning it back on.

I am using version 2.5.6 and it is installed in a: 
Itho device type: CVE-Silent
Itho fw version: 25

The status of the i2c bus: Itho I2C connection status: connected

The last part of my syslog:

```
2023-06-27 19:58:53 I: Reboot requested
     16441 I: I2C init: Safe guard enabled
     16249 I: I2C init: QueryStatus
     16062 I: I2C init: QueryStatusFormat - items:12
     15890 I: I2C init: QueryDevicetype - fw:25 hw:27
     15460 I: Hostname: nrg-itho-9580
     15178 I: mDNS: started
     14924 I: Webserver: started
     13009 I: WiFi: connection successful
     12789 I: Timezone: Europe/Amsterdam, specifier CET-1CEST,M3.5.0,M10.5.0/3 
      6602 E: Unable to set wifi disconnect
      4028 I: I2C sniffer capable hardware: yes
      3864 I: HW rev: 2, FW ver.: 2.5.6
      3699 I: System boot, last reset reason: SW_RESET
    102585 I: Reboot requested
     16204 I: I2C init: Safe guard enabled
     15950 I: I2C init: QueryStatus
     15745 I: I2C init: QueryStatusFormat - items:12
     15404 I: I2C init: QueryDevicetype - fw:25 hw:27
     10256 I: Hostname: nrg-itho-9580
     10011 I: mDNS: started
      9804 I: Webserver: started
      8379 I: Setup: AP mode active
      7982 I: Timezone: Europe/Amsterdam, specifier CET-1CEST,M3.5.0,M10.5.0/3 
      7764 I: wifi AP mode started
      3547 I: I2C sniffer capable hardware: yes
      3448 I: HW rev: 2, FW ver.: 2.5.6
      3377 I: System boot, last reset reason: SW_RESET
```

mjpblaauw, 2023-07-09:
Morning,Just to let you know that the issue was solved on my end.Thanks again.MarcelVerstuurd vanaf mijn iPhone

arjenhiemstra, 2023-07-13:
@Bodmoor almost always this is due to the fan not being in the correct mode, more info can be found here:
https://github.com/arjenhiemstra/ithowifi/wiki/Controlling-the-speed-of-a-fan#help-a-speed-command-does-not-work

closed 2023-09-20 by stale[bot]