arjenhiemstra/ithowifi #190: Itho CVE WiFi add-on wifi disconnect errors - WL_DISCONNECTED

fgerrits, 2023-06-21:
**Describe the bug**
WiFi problems with the Itho CVE WiFi add-on from the start, I put a kaku in between to restart the add-on.
Sometimes it goes well for a few days, then I sometimes have to restart it 6-7 times a day.
Is there a possibility to connect an rj-45 connector to it?
I have several esp controller boards at home and this is the only one that constantly gives problems.
hw-v2.4.3 is the only one that stays online reasonably well, the hw-v2.5.* does not solve the problem.
see attached pictures.
**To Reproduce**
Steps to reproduce the behaviour: cannot
1. in syslog window

**Expected behaviour**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Device information**
- Firmware version [e.g. 2.4.3]
- Hardware revision (e.g. cve 2.0)
- Type/model number Itho (on sticker outside the box)
- CC1101 RF module enabled [ yes ]
- In case of wifi issues, wifi access point make and model: Asus RT-AX86U Pro

**Smartphone (please complete the following information):**
 - Device: [e.g. samsung S20 FE]
 - OS: [e.g. Android 13]
 - Browser [Chrome]
 - Version [e.g. 114.0
[image]
[image]
]

arjenhiemstra, 2023-07-13:
could it be that the signal strength is an issue? 
If you do a wifi scan from the add-on, are all the bars of your wifi network filled?
ie.:
[image]

It could also be a faulty module of course, in that case I'll replace the module.

Unfortunately there is only a wifi option, including wired networking support would mean a complete redesign of the module.

fgerrits, 2023-07-19:
Hi, the signal strength with scan from the box is good, even has my asus routers adjusted in the wlan setting for esp use.
All the other esp devices (i have 8 of them) are working fine, accept the 1 in the itho CVE.
[image]

fgerrits, 2023-08-31:
Still no answer .

arjenhiemstra, 2023-08-31:
There seems to be no other users experiencing these kind of issues. Maybe your add-on is faulty. Could you send me an e-mail with order details, I'll replace your module just to be sure.

sanderkob, 2023-08-31:
I see multiple instances of AP MAIN. Could this be an issue?

arjenhiemstra, 2023-08-31:
Shouldn't be an issue I expect. Some versions ago I've changed the behaviour; ,if there are multiple APs with the same SSID the firmware selects the AP with the strongest signal on boot

fgerrits, 2023-10-12:
Finally had the time to replace the new add-on, but because i had old firmware in the old add-on v2.4.3 (this works the best) i upgraded it to the latest firmware.
After the update the add-on never showed up in my wlan, but it was set in first-setup mode with AP enabled.
So i connect to the AP to setup it again but could never get in the webpage on http://192.168.4.1
I scanned the AP network and i could see the port 80 open with a very high ping rate 800-900ms ping.
Disconnected the old add-on with the temp sensor and placed the new add-on with temp sensor, after that started up, connect to the new AP and also no webpage on http://192.168.4.1

After investigation and trying the original itho add-on (this one worked as aspected) and removed the temp sensor the webpage finally showed up.

So is it possible that the faulty temp sensor was the cause off all this mess ?
The temp sensor is 6 months old and worked until i upgraded to v2.5.6.

closed 2023-12-19 by stale[bot]