arjenhiemstra/ithowifi #191: Setting custom hostname suffix makes the board respond very slow (~60s per request)

grasmanek94, 2023-07-03:
**Describe the bug**
Setting custom hostname suffix makes the board respond very slow (~60s per request)
**To Reproduce**
Steps to reproduce the behaviour:
1. Go to the Web UI `http://nrg-itho-****.local/`
2. Click on `Wifi setup`
3. Change `Hostname` by appendig a suffix, e.g. `.fritz.box`, so `nrg-itho-A1B2` will become `nrg-itho-A1B2.fritz.box`
4. Save
5. Reboot

**Expected behaviour**
I expect the board to be as responsive as when the hostname doesn't contain a suffix.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Device information**
- Firmware version `2.5.2` (`3874 I: HW rev: NON-CVE 1, FW ver.: 2.5.2`)
- Hardware revision `NON-CVE 1` (`3874 I: HW rev: NON-CVE 1, FW ver.: 2.5.2`)
- Type/model number Itho: `HRU350`
- CC1101 RF module enabled `yes` but not using it for anything (yet)
- In case of wifi issues, wifi access point make and model: `Fritz!Box 7530` with disabled WiFi, WiFi APs: `Ubiquiti UniFi AP-AC LITE`

**Debug logging**
```
2023-07-03 21:37:57 I: I2C init: QueryStatus
2023-07-03 21:37:57 I: I2C init: QueryStatusFormat - items:33
2023-07-03 21:37:57 I: I2C init: QueryDevicetype - fw:4 hw:43
2023-07-03 21:37:56 I: Hostname: nrg-itho-b450
2023-07-03 21:37:56 I: mDNS: started
2023-07-03 21:37:56 I: Webserver: started
2023-07-03 21:37:55 I: Setup: init of CC1101 RF module successful
2023-07-03 21:37:53 I: WiFi: connection successful
2023-07-03 21:37:46 E: Unable to set wifi disconnect
2023-07-03 21:37:44 I: I2C sniffer capable hardware: yes
      3874 I: HW rev: NON-CVE 1, FW ver.: 2.5.2
2023-07-03 21:37:44 I: System boot, last reset reason: SW_RESET
2023-07-03 21:37:38 I: Reboot requested
     21103 I: I2C init: QueryStatus
     20886 I: I2C init: QueryStatusFormat - items:33
     20695 I: I2C init: QueryDevicetype - fw:4 hw:43
     20511 I: Hostname: nrg-itho-b450.fritz.box
     20341 I: mDNS: started
     19945 I: Webserver: started
     19665 I: Setup: init of CC1101 RF module successful
     17407 I: WiFi: connection successful
      6531 E: Unable to set wifi disconnect
      3854 I: I2C sniffer capable hardware: yes
      3655 I: HW rev: NON-CVE 1, FW ver.: 2.5.2
      3447 I: System boot, last reset reason: POWERON_RESET
2023-07-03 17:05:02 I: I2C init: QueryStatus
2023-07-03 17:05:02 I: I2C init: QueryStatusFormat - items:33
2023-07-03 17:05:01 I: I2C init: QueryDevicetype - fw:4 hw:43
2023-07-03 17:05:01 I: Hostname: nrg-itho-b450.fritz.box
2023-07-03 17:05:01 I: mDNS: started
2023-07-03 17:05:00 I: Webserver: started
2023-07-03 17:05:00 I: Setup: init of CC1101 RF module successful
2023-07-03 17:04:58 I: WiFi: connection successful
2023-07-03 17:04:51 E: Unable to set wifi disconnect
2023-07-03 17:04:49 I: I2C sniffer capable hardware: yes
2023-07-03 17:04:48 I: HW rev: NON-CVE 1, FW ver.: 2.5.2
2023-07-03 17:04:48 I: System boot, last reset reason: SW_RESET
2023-07-03 17:20:56 I: Reboot requested
2023-07-02 21:06:56 I: I2C init: QueryStatus
2023-07-02 21:06:56 I: I2C init: QueryStatusFormat - items:33
2023-07-02 21:06:56 I: I2C init: QueryDevicetype - fw:4 hw:43
2023-07-02 21:06:56 I: Hostname: nrg-itho-b450
2023-07-02 21:06:55 I: mDNS: started
2023-07-02 21:06:55 I: Webserver: started
2023-07-02 21:06:55 I: Setup: init of CC1101 RF module successful
2023-07-02 21:06:53 I: WiFi: connection successful
2023-07-02 21:06:42 E: Unable to set wifi disconnect
2023-07-02 21:06:39 I: I2C sniffer capable hardware: yes
2023-07-02 21:06:39 I: HW rev: NON-CVE 1, FW ver.: 2.5.2
2023-07-02 21:06:39 I: System boot, last reset reason: OWDT_RESET
2023-07-02 21:00:14 I: Firmware update: sketch
2023-06-29 22:27:18 I: I2C init: QueryStatus
2023-06-29 22:27:17 I: I2C init: QueryStatusFormat - items:33
2023-06-29 22:27:17 I: I2C init: QueryDevicetype - fw:4 hw:43
2023-06-29 22:27:17 I: Hostname: nrg-itho-b450
2023-06-29 22:27:17 I: mDNS: started
2023-06-29 22:27:16 I: Webserver: started
2023-06-29 22:27:16 I: Setup: init of CC1101 RF module successful
2023-06-29 22:27:15 I: WiFi: connection successful
2023-06-29 22:27:08 E: Unable to set wifi disconnect
2023-06-29 22:27:05 I: I2C sniffer capable hardware: yes
2023-06-29 22:27:05 I: HW rev: NON-CVE 1, FW ver.: 2.5.2
2023-06-29 22:27:04 I: System boot, last reset reason: SW_RESET
2023-06-29 22:26:59 I: Reboot requested
2023-06-29 19:21:47 I: I2C init: QueryStatus
2023-06-29 19:21:46 I: I2C init: QueryStatusFormat - items:33
2023-06-29 19:21:46 I: I2C init: QueryDevicetype - fw:4 hw:43
2023-06-29 19:21:46 I: Hostname: nrg-itho-b450
2023-06-29 19:21:45 I: mDNS: started
2023-06-29 19:21:45 I: Webserver: started
2023-06-29 19:21:45 I: Setup: init of CC1101 RF module successful
     17584 I: WiFi: connection successful
      6701 E: Unable to set wifi disconnect
      4135 I: I2C sniffer capable hardware: yes
      3965 I: HW rev: NON-CVE 1, FW ver.: 2.5.2
      3749 I: System boot, last reset reason: POWERON_RESET
2023-06-29 19:21:08 I: I2C init: QueryStatus
2023-06-29 19:21:07 I: I2C init: QueryStatusFormat - items:33
2023-06-29 19:21:07 I: I2C init: QueryDevicetype - fw:4 hw:43
2023-06-29 19:21:07 I: Hostname: nrg-itho-b450
2023-06-29 19:21:07 I: mDNS: started
2023-06-29 19:21:06 I: Webserver: started
2023-06-29 19:21:06 I: Setup: init of CC1101 RF module successful
2023-06-29 19:21:04 I: WiFi: connection successful
2023-06-29 19:20:57 E: Unable to set wifi disconnect
2023-06-29 19:20:55 I: I2C sniffer capable hardware: yes
2023-06-29 19:20:54 I: HW rev: NON-CVE 1, FW ver.: 2.5.2
2023-06-29 19:20:54 I: System boot, last reset reason: SW_RESET
2023-06-29 19:20:44 I: Reboot requested
     17069 I: I2C init: QueryStatus
     16654 I: I2C init: QueryStatusFormat - items:33
     16379 I: I2C init: QueryDevicetype - fw:4 hw:43
     16127 I: Hostname: nrg-itho-b450
     15891 I: mDNS: started
     15666 I: Webserver: started
     13559 I: WiFi: connection successful
      6594 E: Unable to set wifi disconnect
      4080 I: I2C sniffer capable hardware: yes
      3923 I: HW rev: NON-CVE 1, FW ver.: 2.5.2
      3492 I: System boot, last reset reason: SW_RESET
    217424 I: Reboot requested
     15659 I: I2C init: QueryStatus
     15457 I: I2C init: QueryStatusFormat - items:33
     15272 I: I2C init: QueryDevicetype - fw:4 hw:43
     10910 I: Hostname: nrg-itho-b450
     10698 I: mDNS: started
     10210 I: Webserver: started
      8281 I: Setup: AP mode active
      8043 I: wifi AP mode started
      3805 I: I2C sniffer capable hardware: yes
      3595 I: HW rev: NON-CVE 1, FW ver.: 2.5.2
      3383 I: System boot, last reset reason: POWERON_RESET
     15831 I: I2C init: QueryStatus
     15670 I: I2C init: QueryStatusFormat - items:75
     15326 I: I2C init: QueryDevicetype - fw:20 hw:11
     10952 I: Hostname: nrg-itho-b450
     10666 I: mDNS: started
     10391 I: Webserver: started
      8503 I: Setup: AP mode active
      8255 I: wifi AP mode started
      4112 I: I2C sniffer capable hardware: yes
      3789 I: HW rev: NON-CVE 1, FW ver.: 2.5.2
      3496 I: System boot, last reset reason: POWERON_RESET
     10814 I: Hostname: nrg-itho-b450
     10605 I: mDNS: started
     10435 I: Webserver: started
      8441 I: Setup: AP mode active
      8223 I: wifi AP mode started
      4017 I: I2C sniffer capable hardware: yes
      3918 I: HW rev: NON-CVE 1, FW ver.: 2.5.2
      3847 I: System boot, last reset reason: POWERON_RESET
```

```
System memory usage:

CC1101 task memory: 1568 bytes free
MQTT task memory: 4374 bytes free
Web task memory: 2600 bytes free
Config and Log task memory: 3040 bytes free
SysControl task memory: 3540 bytes free
Loop task memory: 1472 bytes free
```

**Desktop (please complete the following information):**
 - OS: Windows 10 x64 22H2 19045.3086
 - Browser `Chromium`
 - Version `Version 113.0.5672.127 (Official Build) (64-bit)`
- Also tried on Firefox

**Smartphone (please complete the following information):**
 - Device: `OnePlus 5 A5000`
 - OS: `OxygenOS 10.0.1`
 - Browser `Chrome`
 - Version `114.0.5735.196`
 - Also tried on Firefox Mobile

grasmanek94, 2023-07-03:
I think this might be related to espressif because the same is happening for my ESP32 devices.

arjenhiemstra, 2023-07-13:
Most ESP32 devices will be using the ESP-IDF framework for development of functionality. mDNS is a standard framework functionality. Therefor I would expect all devices having the same issue.
I'm also not entirely sure of mDNS was designed to support naming structures that are different than [hostname].local.
mDNS standard serves on the .local TLD, if you configure an extra .fritz.box it will become [hostname].fritz.box.local which may result in resolving issues (as ie. nrg-itho-A1B2.fritz.box is not a hostname DNS wise)

closed 2023-09-20 by stale[bot]