arjenhiemstra/ithowifi #192: Power cord

WvanWaas, 2023-07-05:
Just received the module for my HRU350 but I do see a USB Micro connector. I expected it to be powered over the ethernet cable is this true or should I add an old phone adapter with USB Micro cable for power?

arjenhiemstra, 2023-07-05:
> Just received the module for my HRU350 but I do see a USB Micro connector. I expected it to be powered over the ethernet cable is this true or should I add an old phone adapter with USB Micro cable for power? 

only an Ethernet cable is needed for the device to function in conjunction with the HRU350

WvanWaas, 2023-07-05:
Thank you it wasn't clear from the manual but I figured I couldn't fry the device by not overpowering it so tried ethernet only. It worked like a charm. Thanks what a great device!

closed 2023-07-05 by WvanWaas

arjenhiemstra, 2023-07-05:
> Thank you it wasn't clear from the manual but I figured I couldn't fry the device by not overpowering it so tried ethernet only. It worked like a charm. Thanks what a great device!

No prob, better safe than sorry, appreciated!