arjenhiemstra/ithowifi #193: Migrate to ESPHome

svrooij, 2023-07-11:
This addon is great, let me start with that!

In ESPHome they are actively working on all sort of communications (API/Web/MQTT), they also have a lot in place for logging and stuff like that. If you would migrate this code to ESPHome you would no longer have to maintain all that code because someone else does that for you.

Then the device would just be another `component` that people can use, in combination with this board off course.

https://esphome.io/components/fan/index.html?highlight=fan

roblomq, 2023-07-11:
For you just a reminder that not everybody is using Home Asstant so moving to ESPhome can cause integration problems to other domotica software.

svrooij, 2023-07-11:
I’m using home assistant myself, but that is not the main reason. ESPHome still has support for mqtt, and other home automation systems probably already support the messages that used to be just for home assistant.

using ESPhome with home assistant is super easy, using ESPhome with another home automation system will not be more or less difficult then it is now. It might even increase the support because other systems already build support for home assistant based mqtt messages.

arjenhiemstra, 2023-07-13:
There are pro's and con's to ESPHome, the platform certainly developed to a very mature and capable base. 
A user already created a port to ESPHome which can be found here and might be a good start:
https://github.com/hencou/esphome_components/tree/main/components/itho

In the future I might try to migrate the whole project to ESPHome but for now time wise it a difficult one. The biggest issue might be if current users are able to upgrade (I use a not standard partition setup that might make it difficult, not sure if ESPHome supports it)

marcelmah, 2023-08-25:
I would just like to add that for me being ESPHome based is a big +
This way end users can easily update the firmware and you only have to provide some config, the rest is managed by the ESPhome platform.

closed 2023-11-04 by stale[bot]